espressif/arduino-esp32 #4316: WiFiEventInfo_t IP address is wrong - does not matches real one

shafr, 2020-09-01:
### Hardware:
Board:							esp32dev
Core Installation version:			framework-arduinoespressif32 3.10004.200129 (1.0.4)
IDE name:						Platform.io
Flash Frequency:					40Mhz (default one)
PSRAM enabled:                                     not sure
Upload Speed:					460800
Computer OS:                                        Windows 10

### Description:
There is an issue with value of IP addresses returned inside `Wifi event loops`, instantiated from `WiFiEventInfo_t`.
While **WiFi.softAPIP()** returns correct value 192.168.4.1, any attempts to parse IP to String return **209.235.16.128**.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <Arduino.h>
#include <WiFi.h>

void WiFiGotIP(WiFiEvent_t event, WiFiEventInfo_t info)
{
    Serial.println(IPAddress(info.got_ip.ip_info.ip.addr));
    Serial.println(ip4addr_ntoa(&info.got_ip.ip_info.ip));
}

void setup()
{
  Serial.begin(115200);
  Serial.println();

  WiFi.onEvent(WiFiGotIP, SYSTEM_EVENT_AP_STAIPASSIGNED);

  WiFi.mode(WIFI_AP); 
  WiFi.softAP("WFT2", "1234567890" );
  Serial.println("MY IP address is" + WiFi.softAPIP());
}

void loop()
{
}
```

Output:
```
MY IP address is
192.168.4.1     
245.235.16.128
245.235.16.128
```

### Debug Messages:
I don't have arduino.

lbernstone, 2020-09-01:
system_event_ap_staconnected_t has a .mac and a .aid.  You will need to enumerate the list of connected stations to then find the mac that matches.
https://techtutorialsx.com/2019/09/22/esp32-arduino-soft-ap-obtaining-ip-address-of-connected-stations/

lbernstone, 2020-09-01:
It should just be info.ap_staipassigned.ip

shafr, 2020-09-02:
@lbernstone thank you for reply - this variable seems to be part of `esp_event_legacy.h` file, which i'm not able to call. 

[image]

Should I include legacy file in order to use this method ?

shafr, 2020-09-02:
From the other side, when I itterate over connected clients, i'm able to see correct IP:

```
wifi_sta_list_t wifi_sta_list;
  tcpip_adapter_sta_list_t adapter_sta_list;
 
  memset(&wifi_sta_list, 0, sizeof(wifi_sta_list));
  memset(&adapter_sta_list, 0, sizeof(adapter_sta_list));
 
  esp_wifi_ap_get_sta_list(&wifi_sta_list);
  tcpip_adapter_get_sta_list(&wifi_sta_list, &adapter_sta_list);
 
  for (int z = 0; z < adapter_sta_list.num; z++) {
 
    tcpip_adapter_sta_info_t station = adapter_sta_list.sta[z];
 
    ips[*size] = ip4addr_ntoa(&(station.ip));
    Serial.println(ip4addr_ntoa(&(station.ip)));
    *size = *size + 1;
  }
```

lbernstone, 2020-09-02:
It compiles for me in Arduino IDE just changing your original code with `Serial.println(IPAddress(info.ap_staipassigned.ip.addr));`.  No includes needed beyond WiFi.h.

closed 2020-09-11 by shafr