espressif/arduino-esp32 #4318: After upgrading to 1.0.4 UART problem

tkralj, 2020-09-02:
Hello, after upgrading to version 1.0.4 UART seems to skip some bytes sometimes. I can reproduce this behaviour every time i compile code with 1.0.4 or 1.0.3 versions. When downgrade to 1.0.2 code works as expected.

lbernstone, 2020-09-02:
See if https://github.com/espressif/arduino-esp32/commit/49b76649f13acdb8abaf795015d2740bc5f6b340 fixes it.

rljonesau, 2020-11-05:
I've had a related problem, likewise after updating.

I was getting random -1 (0xff) added to the received data when using this construct:

`if (Serial2.available()) {  
    data = Serial2.read();   
}`

closed 2021-01-20 by stale[bot]

Miq1, 2021-03-22:
@rljonesau - thanks a lot, your post saved me hours of chasing a bug. To not have to patch the core over and over (the maintainers seem to not have any troubles, else it would have been changed since), I now am using this construction:
```
if (serial.available()) {
  int c = serial.read();
  if (c >= 0) {
    ... (use c) ...
  }
}
```