espressif/arduino-esp32 #4320: [E][WiFiGeneric.cpp:654] hostByName(): DNS Failed for server_domain.com - still continues on ESP32

davidkjackson54, 2020-09-02:
The issue reported and subsequently closed in 
https://github.com/espressif/arduino-esp32/issues/3203
is still present.

I have just moved code from an ESP8266 to ESP32.
I have MQTT running on a Raspberry p[i which is also running avahi with the default , so Avahi expects .local.
I can ping the RPi from my Windows 10 machine using:
raspberrypi
raspberrypi.local
raspberroypi.localdomain
These are all successful.

My code:
if (!MDNS.begin("esp32")) { // Start the mDNS responder for esp32.local
Serial.println("Error setting up MDNS responder!");
while(1) {
delay(1000);
}
}
Serial.println("mDNS responder started");

client.setServer("raspberrypi.localdomain", 1883); // set ip address of Raspberry pi and default mqtt port number
client.setCallback(callback); // set callback for messages that arrive on a topic

This works but if I change it to **raspberrypi.local** , I get
**[E][WiFiGeneric.cpp:654] hostByName(): DNS Failed**

Using raspberrypi.local  worked fine on ESP8266 but fails on ESP32.

There is nothing in the Pi Mosquitto log to show any error. so it appears to be failing before that point  trying to resolve the raspberrypi.local  name. 

I would have expected a MQTT connect ot raspberrypi.local to have worked from the esp32. 
As I understand it, .local is the assumed suffix for mDNS.  
If I have to hardcode a suffix other than .local , then it is no better than hard coding an IP address.

I am using Unifi router and wifi .

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

closed 2020-09-03 by davidkjackson54

markg85, 2020-12-12:
I too have this issue. I wonder why you closed your own issue and ask for help...
Probably by accident i suppose?

I'm using the exact same stack as you it seems. Also with WifiManager.

ollie1400, 2021-02-21:
I ran into this as well, and google took me to https://github.com/espressif/arduino-esp32/issues/3822 and https://www.tutorialfor.com/questions-324359.htm

Seems like on the ESP32 you have to do the MDNS lookup explicitly.

markg85, 2021-02-21:
> I ran into this as well, and google took me to #3822 and https://www.tutorialfor.com/questions-324359.htm
> 
> Seems like on the ESP32 you have to do the MDNS lookup explicitly.

I took a deep dive into this issue and documented my findings here: https://github.com/espressif/esp-idf/issues/2507#issuecomment-761836300

The actual issue is that for whatever reason the ESP32 SDK blindly adds ".local" to mdns queries. That's downright wrong as ".local" is something a user can change (.local is the default) and doing this makes it a lot more difficult to lookup names.

I'll add this to #3822 too.

Timmmy001, 2021-11-09:
it seems to be a tool chain problem:
https://github.com/espressif/esp-idf/issues/6590#issuecomment-827086670