espressif/arduino-esp32 #4321: touchAttachInterrupt() userfunc parameter should not be void

ericgu, 2020-09-02:
I'm working on some code that uses the touch subsystem and the arduino HAL. 

I can call touchAttachInterrupt() multiple times for different pins. The underlying code in __touchISR() knows which routine to call for a specific pin, but it doesn't pass any pin information on. 

This forces me to write something like:
touchAttachInterrupt(4, Touch0Interrupt, 20);
...
touchAttachment(32, Touch9Interrupt, 20);

touchAttachInterrupt() should store the pin number away at the same time it stores the function pointer away and then pass it on the callback. That would let me write a routine like:

void MyTouchHandler(int pinNumber)
{
  
}

and handle all the touch pins with a single chunk of generic code.

sansillusion, 2020-09-03:
Not sure as I'm not much of a programmer but would something like that work ? :
    touchAttachInterrupt(4, TouchInterrupt(0), 20);
    touchAttachment(32, TouchInterrupt(9), 20);

ericgu, 2020-09-03:
Unfortunately, that does not work, but you did get me thinking that I can probably do something with the C++ lambda support. Something like:

int pin = 32;
touchAttachInterrupt(pin, [pin](){MyInterruptHandler(pin);});

assuming I have my lambda syntax correct. That's not great but it's better than what I had planned from a syntax perspective, though the overhead might be worse...

ericgu, 2020-09-03:
Lambdas do work but not in the format I used; you can't pass lambdas that do captures to function pointers, so they aren't really any better than my initial implementation.

Leo-C, 2021-02-24:
Hi,
I believe that function [`esp_sleep_get_touchpad_wakeup_status()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#_CPPv436esp_sleep_get_touchpad_wakeup_statusv) can help you.

Interrupt handler can not get a parameter indicating pin touched, but calling function above as first instruction, you'll have same information

Leonardo

SuGlider, 2022-01-24:
New Touch API will have this function ```touchAttachInterruptArg(...)```

``` dart
/*
 * Set cycles that measurement operation takes
 * The result from touchRead, threshold and detection
 * accuracy depend on these values. Defaults are
 * 0x1000 for measure and 0x1000 for sleep.
 * With default values touchRead takes 0.5ms
 * */
void touchSetCycles(uint16_t measure, uint16_t sleep);

/*
 * Read touch pad (values close to 0 mean touch detected)
 * You can use this method to chose a good threshold value
 * to use as value for touchAttachInterrupt
 * */
uint32_t touchRead(uint8_t pin);

/*
 * Set function to be called if touch pad value falls (ESP32)
 * below the given threshold / rises (ESP32-S2/S3) by given increment (threshold). 
 * Use touchRead to determine a proper threshold between touched and untouched state
 * */
void touchAttachInterrupt(uint8_t pin, void (*userFunc)(void), uint32_t threshold);
void touchAttachInterruptArg(uint8_t pin, void (*userFunc)(void*), void *arg, uint32_t threshold);
void touchDetachInterrupt(uint8_t pin);

/*
 * Specific functions to ESP32 
 * Tells the dirver if it shall activate the ISR if the sensor is Lower or Higher than the Threshold
 * Default if Lower.
 **/
void touchInterruptSetThresholdDirection(bool mustbeLower);
```

closed 2022-02-03 by me-no-dev

jdeltoft, 2024-03-11:
Does this new touchAttachInterruptArg() api support wake up from deep sleep?  Or is that only supported with touchAttachInterrupt().

I have this for setup:

```
void setup() {
  touch_pad_init();

  for(int i=0; i<TOUCH_PIN_COUNT; i++) {
    touchAttachInterruptArg(pinNames[i], gotTouch, &pinIds[i], capTouchThreshold);
    touchSleepWakeUpEnable(pinNames[i], 500);
  }

  esp_sleep_enable_touchpad_wakeup();
}
```

And then if I call `esp_deep_sleep_start();` I lose my serial connection and never seem to wake up from any touch.  If I just remove this deep sleep call, everything works perfect for the 5 cap touch inputs.

PR #6194 (2022-02-03): Touch Sensor IDF Refactoring
## Summary
Touch Sensor refactoring to be based on IDF 4.4.
Adds support to ESP32S2 and future ESP32S3.

#### Adds some new APIs:
#### For all chips:
```void touchAttachInterruptArg(uint8_t pin, void (*userFunc)(void*), void *arg, uint32_t threshold);```
This function allows the user to add and pass a void* parameter to the ISR user callback.

```void touchDetachInterrupt(uint8_t pin);```
This function detaches ISR call back for the touch pad pin.

#### Only ESP32 chip
```void touchInterruptSetThresholdDirection(bool mustbeLower);```
This function allows the user t [... cut]
commit 0ea485e5 (2022-02-03): Touch Sensor IDF Refactoring (#6194)

Summary

Touch Sensor refactoring to be based on IDF 4.4.
Adds support to ESP32S2 and future ESP32S3.

Adds some new APIs:

For all chips:

void touchAttachInterruptArg(uint8_t pin, void (*userFunc)(void*), void *arg, uint32_t threshold);
This function allows the user to add and pass a void* parameter to the ISR user callback.

void touchDetachInterrupt(uint8_t pin);
This function detaches ISR call back for the touch pad pin.

Only ESP32 chip

void touchInterruptSetThresholdDirection(bool mustbeLower);
This function allows the user to set if the ISR callba [... cut]