espressif/arduino-esp32 #4323: External wakeup ext0 & ext1 not working together

Shuptuu, 2020-09-03:
Hi,
I try to use at the same time ext0 and ext1 to wake up an esp32. When I enable just one (ext0 or ext1), it works fine. But when enabling both, just ext0 is considered, ext1 is ignored.
Any idea?

A test code to illustrate:
```
RTC_DATA_ATTR int           bootcount=0;

void setup() {
  Serial.begin(115200);
  
  Serial.println("Hello");
  Serial.println("Boot number: " + String(bootcount));
  esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();
  wakeup_reason_txt(wakeup_reason);
  
  bootcount++;

  // initialize input pins:
  pinMode(27, INPUT_PULLUP);
  pinMode(35, INPUT);

  // going to deepsleep
  Serial.println("Going to deep sleep...");
  Serial.println();
  Serial.println();
  esp_sleep_enable_ext1_wakeup(0x8000000,ESP_EXT1_WAKEUP_ALL_LOW); // mask for pin 35
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_27,LOW);
  esp_deep_sleep_start();
}

void loop() {
  // put your main code here, to run repeatedly:

}

void wakeup_reason_txt(esp_sleep_wakeup_cause_t wakeup_reason){
  switch(wakeup_reason)
  {
    case 2  : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case 3  : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case 4  : Serial.println("Wakeup caused by timer"); break;
    case 5  : Serial.println("Wakeup caused by touchpad"); break;
    case 6  : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}
```

atanisoft, 2020-09-03:
Your mask looks incorrect, it appears to be setting GPIO 28 and not 35. Can you try:
```
esp_sleep_enable_ext1_wakeup((1 << 35) ,ESP_EXT1_WAKEUP_ALL_LOW); // mask for pin 35
```

Shuptuu, 2020-09-03:
yes, sorry, just discovered this too...
Thanks

closed 2020-09-03 by Shuptuu