espressif/arduino-esp32 #4326: error: 'struct esp_http_client_config_t' has no member named 'skip_cert_common_name_check'

EMHacker-alt, 2020-09-06:
### Hardware:
Board:							?ESP32 Dev Module?
IDE name:						?Platform.io?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no?
Upload Speed:					?921600?
Computer OS:                                         ?Mac OSX?

### Description:
Hi, I'm trying to use the "HTTPS_OTA_Update.ino" to perform OTA update of ESP32. But when I compile the code, an Error message appears "error: 'struct esp_http_client_config_t' has no member named 'skip_cert_common_name_check'". I go to the struct 'esp_http_client_config_t' and found no member of 'skip_cert_common_name_check'. Anyone can solve this problem.

Thanks!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
// This sketch provide the functionality of OTA Firmware Upgrade
#include <Arduino.h>
#include "WiFi.h"
#include "HttpsOTAUpdate.h"
// This sketch shows how to implement HTTPS firmware update Over The Air.
// Please provide your WiFi credentials, https URL to the firmware image and the server certificate.

static const char *ssid     = "your-ssid";  // your network SSID (name of wifi network)
static const char *password = "your-password"; // your network password

static const char *url = "https://example.com/firmware.bin"; //state url of your firmware image

static const char *server_certificate = "-----BEGIN CERTIFICATE-----\n" \
     "MIIEkjCCA3qgAwIBAgIQCgFBQgAAAVOFc2oLheynCDANBgkqhkiG9w0BAQsFADA/\n" \
     "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n" \
     "DkRTVCBSb290IENBIFgzMB4XDTE2MDMxNzE2NDA0NloXDTIxMDMxNzE2NDA0Nlow\n" \
     "SjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxIzAhBgNVBAMT\n" \
     "GkxldCdzIEVuY3J5cHQgQXV0aG9yaXR5IFgzMIIBIjANBgkqhkiG9w0BAQEFAAOC\n" \
     "AQ8AMIIBCgKCAQEAnNMM8FrlLke3cl03g7NoYzDq1zUmGSXhvb418XCSL7e4S0EF\n" \
     "q6meNQhY7LEqxGiHC6PjdeTm86dicbp5gWAf15Gan/PQeGdxyGkOlZHP/uaZ6WA8\n" \
     "SMx+yk13EiSdRxta67nsHjcAHJyse6cF6s5K671B5TaYucv9bTyWaN8jKkKQDIZ0\n" \
     "Z8h/pZq4UmEUEz9l6YKHy9v6Dlb2honzhT+Xhq+w3Brvaw2VFn3EK6BlspkENnWA\n" \
     "a6xK8xuQSXgvopZPKiAlKQTGdMDQMc2PMTiVFrqoM7hD8bEfwzB/onkxEz0tNvjj\n" \
     "/PIzark5McWvxI0NHWQWM6r6hCm21AvA2H3DkwIDAQABo4IBfTCCAXkwEgYDVR0T\n" \
     "AQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8EBAMCAYYwfwYIKwYBBQUHAQEEczBxMDIG\n" \
     "CCsGAQUFBzABhiZodHRwOi8vaXNyZy50cnVzdGlkLm9jc3AuaWRlbnRydXN0LmNv\n" \
     "bTA7BggrBgEFBQcwAoYvaHR0cDovL2FwcHMuaWRlbnRydXN0LmNvbS9yb290cy9k\n" \
     "c3Ryb290Y2F4My5wN2MwHwYDVR0jBBgwFoAUxKexpHsscfrb4UuQdf/EFWCFiRAw\n" \
     "VAYDVR0gBE0wSzAIBgZngQwBAgEwPwYLKwYBBAGC3xMBAQEwMDAuBggrBgEFBQcC\n" \
     "ARYiaHR0cDovL2Nwcy5yb290LXgxLmxldHNlbmNyeXB0Lm9yZzA8BgNVHR8ENTAz\n" \
     "MDGgL6AthitodHRwOi8vY3JsLmlkZW50cnVzdC5jb20vRFNUUk9PVENBWDNDUkwu\n" \
     "Y3JsMB0GA1UdDgQWBBSoSmpjBH3duubRObemRWXv86jsoTANBgkqhkiG9w0BAQsF\n" \
     "AAOCAQEA3TPXEfNjWDjdGBX7CVW+dla5cEilaUcne8IkCJLxWh9KEik3JHRRHGJo\n" \
     "uM2VcGfl96S8TihRzZvoroed6ti6WqEBmtzw3Wodatg+VyOeph4EYpr/1wXKtx8/\n" \
     "wApIvJSwtmVi4MFU5aMqrSDE6ea73Mj2tcMyo5jMd6jmeWUHK8so/joWUoHOUgwu\n" \
     "X4Po1QYz+3dszkDqMp4fklxBwXRsW10KXzPMTZ+sOPAveyxindmjkW8lGy+QsRlG\n" \
     "PfZ+G6Z6h7mjem0Y+iWlkYcV4PIWL1iwBi8saCbGS5jN2p8M+X+Q7UNKEkROb3N6\n" \
     "KOqkqm57TH2H3eDJAkSnh6/DNFu0Qg==\n" \
     "-----END CERTIFICATE-----";

static HttpsOTAStatus_t otastatus;

void HttpEvent(HttpEvent_t *event)
{
    switch(event->event_id) {
        case HTTP_EVENT_ERROR:
            Serial.println("Http Event Error");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            Serial.println("Http Event On Connected");
            break;
        case HTTP_EVENT_HEADER_SENT:
            Serial.println("Http Event Header Sent");
            break;
        case HTTP_EVENT_ON_HEADER:
            Serial.printf("Http Event On Header, key=%s, value=%s\n", event->header_key, event->header_value);
            break;
        case HTTP_EVENT_ON_DATA:
            break;
        case HTTP_EVENT_ON_FINISH:
            Serial.println("Http Event On Finish");
            break;
        case HTTP_EVENT_DISCONNECTED:
            Serial.println("Http Event Disconnected");
            break;
    }
}

void setup(){

    Serial.begin(115200);
    Serial.print("Attempting to connect to SSID: ");
    WiFi.begin(ssid, password);

    // attempt to connect to Wifi network:
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(1000);
    }

    Serial.print("Connected to ");
    Serial.println(ssid);
    
    HttpsOTA.onHttpEvent(HttpEvent);
    Serial.println("Starting OTA");
    HttpsOTA.begin(url, server_certificate); 

    Serial.println("Please Wait it takes some time ...");
}

void loop(){
    otastatus = HttpsOTA.status();
    if(otastatus == HTTPS_OTA_SUCCESS) { 
        Serial.println("Firmware written successfully. To reboot device, call API ESP.restart() or PUSH restart button on device");
    } else if(otastatus == HTTPS_OTA_FAIL) { 
        Serial.println("Firmware Upgrade Fail");
    }
    delay(1000);
}
```

### Debug Messages:
```
Building in release mode
Compiling .pio/build/esp32doit-devkit-v1/src/HttpsOTAUpdate.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib099/WiFi/WiFiUdp.cpp.o
src/HttpsOTAUpdate.cpp: In member function 'void HttpsOTAUpdateClass::begin(const char*, const char*, bool)':
src/HttpsOTAUpdate.cpp:91:12: error: 'struct esp_http_client_config_t' has no member named 'skip_cert_common_name_check'
     config.skip_cert_common_name_check = skip_cert_common_name_check;
            ^
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/IPv6Address.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/Print.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/StreamString.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/FrameworkArduino/WMath.cpp.o
*** [.pio/build/esp32doit-devkit-v1/src/HttpsOTAUpdate.cpp.o] Error 1
================================================================== [FAILED] Took 2.77 seconds ==================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.
```

lbernstone, 2020-09-06:
This was added 12 days ago and compiles in Arduino IDE.  Is platform.io up to date such that it has the correct idf?

EMHacker-alt, 2020-09-07:
I upgrade the platform.io to Core version 5.0.0 and reinstall the ESP32 Platform. While, the error still appear.

atanisoft, 2020-09-07:
@EMHacker-alt I'd suggest open this issue against PlatformIO instead since it would appear they are not using the supported version of ESP-IDF for arduino-esp32.

EMHacker-alt, 2020-09-09:
Thanks! I will open this issue against PlatformIO

EMHacker-alt, 2020-09-14:
Problem solved please refer to this https://community.platformio.org/t/error-struct-esp-http-client-config-t-has-no-member-named-skip-cert-common-name-check/15978

closed 2020-09-14 by EMHacker-alt