espressif/arduino-esp32 #4329: LEDs Flicker at below 30% Brightnes and Certain Patterns

sprewl, 2020-09-08:
#4072 # Hardware:
Board:							NODEMCU esp3266MOD
LED type:
Dream Color Pixel Light Strip, Double PCB, WS2812B, Each LED have built in IC, 5V, IP30 No Waterproof, 60 LED/m (5-Meter) 
IDE name:						Arduino IDE
Flash Frequency:				        40Mhz
Upload Speed:					15200
Computer OS:                                         Windows 10
Componets Value:                                   330 OHM resitors and 1000uf 10v Caps

### Description:
Hi,
I was trying to put LEDs on my Helmet and I realized this issue occurs ar random, I have several ESP-01 and Node MCU lying around, and it happens with all of them at times, and some never.
The flicker seems to be very random, the only times I have fixed it is by using the USB port on the Node MCU to power it up, and if I use the Vin pins the LEDs behave very erratic at low brightness or patterns like Pacifica.
The setup I have is at standard ie. 330-ohm Resistor and 1000uf 10v Cap also is present, so far while using my Powersupply and 2A Samsung or any brand phone charger recreates this problem. [Here's a video showcasing the issue.](https://youtu.be/WFRVWE48loE)
**Please note Before blaming the wiring or LEDs I have used a different NODE MCU to check the LEDs the worked fine even on low brightness**. It's not the ESP or the LEDs cause the same happened even if I use 30 LEDs or 900 Of them.
I really need some help, I'm sure many others would have faced similar issues, I'm not able to use any effect on low brightness

lbernstone, 2020-09-08:
esp8266 != esp32
https://github.com/esp8266/Arduino

sprewl, 2020-09-10:
> 
> 
> esp8266 != esp32
> https://github.com/esp8266/Arduino

Can you elaborate?

chegewara, 2020-09-10:
> Can you elaborate?

> Board: NODEMCU esp3266MOD

closed 2020-09-27 by sprewl