espressif/arduino-esp32 #4332: Unable to load more than two files from from the WebServer using SPIFFS

brad-colbert, 2020-09-09:
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation version:			?
IDE name:						Platform.io 5.0.0 3.3.0
Flash Frequency:					80Mhz
PSRAM enabled:                                      ?
Upload Speed:					921600
Computer OS:                                          Mac OSX 10.15.6

### Description:
I'm trying to run this example:

https://circuits4you.com/2018/02/03/esp8266-nodemcu-adc-analog-value-on-dial-gauge/

After loading two files (index.html and the first file listed in the index.html) the browser reports net::ERR_CONNECTION_RESET for the remaining files.

The serial console shows:
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found

but there are no other handlers defined.

I found people with similar issues and I tried the reported solution of turning off WIFI_AMPDU in sdkconfig:
(.platformio/packages/framework-arduinoespressif32/tools/sdk/sdkconfig)
CONFIG_ESP32_WIFI_AMPDU_TX_ENABLED=n
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=n

This had no effect.

Is there a solution for this?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```
/*
 * ESP8266 SPIFFS HTML Web Page with JPEG, PNG Image 
 * https://circuits4you.com
 */

#include <WiFi.h>
#include <WebServer.h>
#include <SPIFFS.h>   //Include File System Headers
#include <FS.h>   //Include File System Headers

const char* htmlfile = "/index.html";

//WiFi Connection configuration
const char *ssid = "Trebloc";
const char *password = "InYourFace1";

WebServer server(80);

void handleADC(){
  int a = analogRead(A13);
  a = map(a,0,1023,0,100);
  String adc = String(a);
  Serial.println(adc);
  server.send(200, "text/plane",adc);
}

void handleRoot()
{
  Serial.println("void handleRoot()");
  server.sendHeader("Location", "/index.html",true);   //Redirect to our html web page
  server.send(302, "text/plane","");
}

void handleWebRequests()
{
  Serial.println("void handleWebRequests()");
  if(loadFromSpiffs(server.uri())) return;
  String message = "File Not Detected\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " NAME:"+server.argName(i) + "\n VALUE:" + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  Serial.println(message);
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  //Initialize File System
  SPIFFS.begin(false, "/spiffs", 20);
  Serial.println("File System Initialized");

  
  //Connect to wifi Network
  WiFi.begin(ssid, password);     //Connect to your WiFi router
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  //If connection successful show IP address in serial monitor
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());  //IP address assigned to your ESP

  //Initialize Webserver
  server.on("/", handleRoot);
  server.on("/getADC",handleADC); //Reads ADC function is called from out index.html
  server.onNotFound(handleWebRequests); //Set setver all paths are not found so we can handle as per URI
  server.begin();  
}

void loop() {
 server.handleClient();
}

bool loadFromSpiffs(String path)
{
  Serial.println("bool loadFromSpiffs(" + path + ")");

  String dataType = "text/plain";
  if(path.endsWith("/")) path += "index.htm";

  if(path.endsWith(".src")) path = path.substring(0, path.lastIndexOf("."));
  else if(path.endsWith(".html")) dataType = "text/html";
  else if(path.endsWith(".htm")) dataType = "text/html";
  else if(path.endsWith(".css")) dataType = "text/css";
  else if(path.endsWith(".js")) dataType = "application/javascript";
  else if(path.endsWith(".png")) dataType = "image/png";
  else if(path.endsWith(".gif")) dataType = "image/gif";
  else if(path.endsWith(".jpg")) dataType = "image/jpeg";
  else if(path.endsWith(".ico")) dataType = "image/x-icon";
  else if(path.endsWith(".xml")) dataType = "text/xml";
  else if(path.endsWith(".pdf")) dataType = "application/pdf";
  else if(path.endsWith(".zip")) dataType = "application/zip";
  File dataFile = SPIFFS.open(path.c_str(), "r");
  if (server.hasArg("download")) dataType = "application/octet-stream";
  if (server.streamFile(dataFile, dataType) != dataFile.size()) {
    Serial.println("Problem with file" + path); 
  }

  dataFile.close();
  return true;
}
```

### Debug Messages:
I don't appear to be able to set this in PlatformIO.

[image]

ayushsharma82, 2020-11-14:
+1 Happening to me as well. Webserver fails to load more than 1 file.

lbernstone, 2020-11-14:
WebServer can only handle 1 connection at a time.  As shown in your error, you are using synchronous XHR.  In any case, [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer) will give better results than WebServer for significant httpd service.

closed 2021-01-30 by stale[bot]