espressif/arduino-esp32 #4334: Why does SPIFFS file.write take several attempts before succeding?

BenBergman, 2020-09-11:
### Hardware:
Board:							ESP32-CAM
Core Installation version:			I am using 1.0.4 of the ESP32 board config in Arduino, not sure if that is the correct value here
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:					921600
Computer OS:                                        Arch Linux

### Description:
I am capturing a frame with an OV2640 camera and saving to SPIFFS. I have found that `file.write()` takes several attempts before it successfully writes any bytes to SPIFFS. I've enabled verbose debugging but don't seem to be getting any extra debug info during the SPIFFS write (I see lots of other debug logging during setup though, so I know verbose debug is on). I've also tried adding a 10 second delay between capturing the image and writing to SPIFFS in case there was some timing issue grabbing the large image over DMA, but I still get 5-10 failed writes before a successful one.

Looking at [`vfs_api.cpp`](https://github.com/espressif/arduino-esp32/blob/67ee7c32e72ca6be40f3437316505dc208509c0d/libraries/FS/src/vfs_api.cpp) (which I think SPIFFS inherits from), it seems like `write()` would return 0 if the file is a directory, if the file ...doesn't exist? fails to open?, if the input data is empty (confirmed this not the case as it eventually does write), if the length of input data is 0 (confirmed this is non-zero before the first write attempt. It would also return 0 if `fwrite()` returns zero, but I'm not sure what would cause that. The lack of debug messages seems to indicate it doesn't think it is a directory or that the file is bad, so I'm at a loss for what to investigate next.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))

Below is my full application for context, but the SPIFFS write happens in `capturePhotoSaveSpiffs()`.

```cpp
/*********
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp32-cam-take-photo-display-web-server/
  
  IMPORTANT!!! 
   - Select Board "AI Thinker ESP32-CAM"
   - GPIO 0 must be connected to GND to upload a sketch
   - After connecting GPIO 0 to GND, press the ESP32-CAM on-board RESET button to put your board in flashing mode
  
  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.
*********/

#include "WiFi.h"
#include "esp_camera.h"
#include "esp_timer.h"
#include "img_converters.h"
#include "Arduino.h"
#include "soc/soc.h"           // Disable brownour problems
#include "soc/rtc_cntl_reg.h"  // Disable brownour problems
#include "driver/rtc_io.h"
#include <ESPAsyncWebServer.h>
#include <StringArray.h>
#include <SPIFFS.h>
#include <FS.h>
#include <ESPmDNS.h>

// Replace with your network credentials
const char* ssid = "80DBD5";
const char* password = "asdf;lkj";

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);
AsyncWebSocket asyncWs("/ws");

boolean takeNewPhoto = false;
camera_fb_t * fb = NULL; // pointer

// Photo File Name to save in SPIFFS
#define FILE_PHOTO "/photo.jpg"

// OV2640 camera module pins (CAMERA_MODEL_AI_THINKER)
#define PWDN_GPIO_NUM     32
#define RESET_GPIO_NUM    -1
#define XCLK_GPIO_NUM      0
#define SIOD_GPIO_NUM     26
#define SIOC_GPIO_NUM     27
#define Y9_GPIO_NUM       35
#define Y8_GPIO_NUM       34
#define Y7_GPIO_NUM       39
#define Y6_GPIO_NUM       36
[... 478 lines cut ...]
The picture has been saved in /photo.jpg - Size: 0 bytes
fb->format is jpeg
check size: 0
Picture file name: /photo.jpg
file.write() result: 0
The picture has been saved in /photo.jpg - Size: 0 bytes
fb->format is jpeg
check size: 0
Picture file name: /photo.jpg
file.write() result: 0
The picture has been saved in /photo.jpg - Size: 0 bytes
fb->format is jpeg
check size: 0
Picture file name: /photo.jpg
file.write() result: 0
The picture has been saved in /photo.jpg - Size: 0 bytes
fb->format is jpeg
check size: 0
Picture file name: /photo.jpg
file.write() result: 193513
The picture has been saved in /photo.jpg - Size: 193408 bytes
fb->format is jpeg
check size: 193513

```

lbernstone, 2020-09-11:
Turn on verbose debugging, by default logging is none.

BenBergman, 2020-09-11:
For the debug output above I had `Core Debug Level` set to `Verbose` in the Arduino IDE.

BenBergman, 2020-09-11:
Originally I had trimmed the debug output to just that of my affected function. I have updated it to be the full output after boot.

JL-EU, 2020-11-22:
I see the same issue on a ESP32 Wrover, often 3 trials before the file is stored successfully. Verbose debug info don't show additional information.

Taking a photo...
Picture size : 141713 bytes
Picture file name: /photo.jpg
File size before: 131200
Failed to write image
File size before: 0
Failed to write image
File size before: 0
Failed to write image
File size before: 0
/photo.jpg - Size: 141696 bytes
Image has been stored correctly

grindylow, 2021-01-02:
Same issue here, also using ESP32-CAM and Rui Santos's "RandomNerdTutorials.com/esp32-cam-take-photo-display-web-server".

I "solved" the issue by switching to LittleFS. According to [the official docs](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#spiffs-deprecation-warning), SPIFFS is deprecated, so this "solution" might actually be quite reasonable.

Replace `#include <SPIFFS.h>` with `#include <LITTLEFS.h>`.

Then replace all occurences of `SPIFFS` with `LITTLEFS`.

So far, every call to `File.write()` seems to succeed the first time around.

krulkip, 2021-02-23:
I will try LITTLEFS but in the mean time i have some observations.

Using SPIFFS i can write data to a file like below. This works all the time.
      for (int i=0;i<data_offset/1024;i++){
        for(int j=0;j<1024;j++){
           read_value[j]=psram_buffer[i*1024+j];  
        }   
        file.write(read_value,1024); 
      }

When i try to write the same data to the same file directly from PSRAM to the file it only works sporadically like below.
file.write((byte *)psram_buffer,data_offset);

It seems PSRAM and SPIFFS are not working well together. Anyone seen this also and solved it ?

lbernstone, 2021-02-23:
Please test with release 1.0.5, which includes #4850

krulkip, 2021-02-23:
Tried with 1.05 and still gets only sporadic success.
Tried with LITTLEFS and works every time. It is also about 4x faster than SPIFFS.

closed 2021-07-09 by stale[bot]