espressif/arduino-esp32 #4339: Arduino IDE - Invalid head of packet (0x50)

Aaronprat, 2020-09-13:
**What you are trying to do?** 
trying to upload code to esp32-s2-Saola-1 v1.2
**Describe your system( Hardware, computer, O/S, core version, environment).**
OSX, 10.15.5, Arduino 1.8.13
**Describe what is failing.**
```
A fatal error occurred: Invalid head of packet (0x50)
```

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3? - not sure
IDE name:					      Arduino IDE 1.8.13
Flash Frequency:				      80Mhz
PSRAM enabled:                                    no
Upload Speed:				      115200
Computer OS:                                        Mac OSX

### Description:
Failing to upload code to the esp32-s2-Saola-1 v1.2 module

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);     // opens serial port, sets data rate to 115200 bps

}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Test");
  delay(1000);  
}
```

### Debug Messages:
```
Sketch uses 215305 bytes (16%) of program storage space. Maximum is 1310720 bytes.
Global variables use 15436 bytes (4%) of dynamic memory, leaving 312244 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32D2WDQ5 (revision 1)
Features: WiFi, Single Core, Embedded Flash, VRef calibration in efuse, Coding Scheme Repeat (UNSUPPORTED)
MAC: d1:8b:bb:0a:b8:93
Uploading stub...
Running stub...

A fatal error occurred: Invalid head of packet (0x50)
A fatal error occurred: Invalid head of packet (0x50)
```

I've been looking into how to resolve this issue, I see that other people had it and didnt find a way to fix it for the arduino IDE app. I'm seeing that I need esptools 3.0 dev, though failing to find how to upgrade it.
any help will be appreciated. lmk if you need more information

thanks

lbernstone, 2020-09-13:
esp32-s2 is not yet supported.  It is available in the esp32s2 branch.  Follow the [gitrepo installation instructions](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md), then `git pull esp32s2`.  You'll then see an ESP32S2 Dev Module in your 'sketchbook' boards.

Aaronprat, 2020-09-14:
I did that. sorry if it was not clear.

chegewara, 2020-09-14:
Are you sure your device is saola? Here is mine log when flashing:
```
Connecting....
Chip is ESP32-S2
Features: WiFi, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz
```

Yours looks like old esp32, single core:
```
Connecting....
Chip is ESP32D2WDQ5 (revision 1)
Features: WiFi, Single Core, Embedded Flash, VRef calibration in efuse, Coding Scheme Repeat (UNSUPPORTED)
```

Aaronprat, 2020-09-16:
This is what written on it: `esp32-s2-Saola-1 v1.2`. On the chip itself it says `ESPRESSIF ESP32-S2-WROOM`

[This](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/get-started/index.html#get-started-step-by-step) instructions is working, through CLI. I'm trying to make it work from the Arduino IDE.

thanks

lbernstone, 2020-09-16:
It sounds like your arduino is using an older version of esptool.  Search your computer for rogue versions of esptool.py.  You should just be able to update them in place.

Aaronprat, 2020-09-16:
"I'm seeing that I need esptools 3.0 dev, though failing to find how to upgrade it."

My esptool version on the arduino IDE is 2.6 as you can see in the debug message. I'm trying to find how to upgrade it for the IDE and dont find documentation for it.

thanks

lbernstone, 2020-09-16:
https://github.com/espressif/arduino-esp32/blob/esp32s2/tools/esptool.py
It is a text file- you should just be able to replace any instances on your machine with this code.
But, it sounds like your installation is damaged.  You will probably run into a lot more issues than this.  I would recommend you uninstall arduino, clean up, and then start over from scratch.

gavan1, 2020-11-11:
> esp32-s2 is not yet supported. It is available in the esp32s2 branch. Follow the [gitrepo installation instructions](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md), then `git pull esp32s2`. You'll then see an ESP32S2 Dev Module in your 'sketchbook' boards.

ran git pull esp32s2

fatal: 'esp32s2' does not appear to be a git repository

lbernstone, 2020-11-11:
Sorry, `git pull origin esp32s2` (the "tip" branch atm).  You will also need to run get.py(exe) again after you pull.

closed 2021-01-24 by stale[bot]