espressif/arduino-esp32 #4340: Error in WiFiSecureClient : Returning UNKNOWN ERROR CODE (0050) from start_ssl_client()

HamzaHajeir, 2020-09-14:
I'm Using PlatformIO Core version 5.0.1 and Home 3.3.0 over VSCode.
Using Windows 10 (if it helps).

### Hardware:
Board:							ESP32 DevkitV1 Module
Core Installation version:			[V2.0.0](https://github.com/platformio/platform-espressif32/releases/tag/v2.0.0)
IDE name:						PlatformIO

### Description:

I'm trying to connect Securely to a server using SSL. I'm facing a problem in connecting using WiFiClientSecure  and BasicHttpsClient tweaked examples.

I've copied RootCA of the [server](https://sila-admin.com/), Which is GoDaddy class 2.

### Tweaked BasicHttpsClient.ino : 
```cpp
#include <Arduino.h>

const char *server = "www.sila-admin.com"; // Server URL

const char *test_root_ca =
    "-----BEGIN CERTIFICATE-----\n"
    "MIIEADCCAuigAwIBAgIBADANBgkqhkiG9w0BAQUFADBjMQswCQYDVQQGEwJVUzEh\n"
    "MB8GA1UEChMYVGhlIEdvIERhZGR5IEdyb3VwLCBJbmMuMTEwLwYDVQQLEyhHbyBE\n"
    "YWRkeSBDbGFzcyAyIENlcnRpZmljYXRpb24gQXV0aG9yaXR5MB4XDTA0MDYyOTE3\n"
    "MDYyMFoXDTM0MDYyOTE3MDYyMFowYzELMAkGA1UEBhMCVVMxITAfBgNVBAoTGFRo\n"
    "ZSBHbyBEYWRkeSBHcm91cCwgSW5jLjExMC8GA1UECxMoR28gRGFkZHkgQ2xhc3Mg\n"
    "MiBDZXJ0aWZpY2F0aW9uIEF1dGhvcml0eTCCASAwDQYJKoZIhvcNAQEBBQADggEN\n"
    "ADCCAQgCggEBAN6d1+pXGEmhW+vXX0iG6r7d/+TvZxz0ZWizV3GgXne77ZtJ6XCA\n"
    "PVYYYwhv2vLM0D9/AlQiVBDYsoHUwHU9S3/Hd8M+eKsaA7Ugay9qK7HFiH7Eux6w\n"
    "wdhFJ2+qN1j3hybX2C32qRe3H3I2TqYXP2WYktsqbl2i/ojgC95/5Y0V4evLOtXi\n"
    "EqITLdiOr18SPaAIBQi2XKVlOARFmR6jYGB0xUGlcmIbYsUfb18aQr4CUWWoriMY\n"
    "avx4A6lNf4DD+qta/KFApMoZFv6yyO9ecw3ud72a9nmYvLEHZ6IVDd2gWMZEewo+\n"
    "YihfukEHU1jPEX44dMX4/7VpkI+EdOqXG68CAQOjgcAwgb0wHQYDVR0OBBYEFNLE\n"
    "sNKR1EwRcbNhyz2h/t2oatTjMIGNBgNVHSMEgYUwgYKAFNLEsNKR1EwRcbNhyz2h\n"
    "/t2oatTjoWekZTBjMQswCQYDVQQGEwJVUzEhMB8GA1UEChMYVGhlIEdvIERhZGR5\n"
    "IEdyb3VwLCBJbmMuMTEwLwYDVQQLEyhHbyBEYWRkeSBDbGFzcyAyIENlcnRpZmlj\n"
    "YXRpb24gQXV0aG9yaXR5ggEAMAwGA1UdEwQFMAMBAf8wDQYJKoZIhvcNAQEFBQAD\n"
    "ggEBADJL87LKPpH8EsahB4yOd6AzBhRckB4Y9wimPQoZ+YeAEW5p5JYXMP80kWNy\n"
    "OO7MHAGjHZQopDH2esRU1/blMVgDoszOYtuURXO1v0XJJLXVggKtI3lpjbi2Tc7P\n"
    "TMozI+gciKqdi0FuFskg5YmezTvacPd+mSYgFFQlq25zheabIZ0KbIIOqPjCDPoQ\n"
    "HmyW74cNxA9hi63ugyuV+I6ShHI56yDqg+2DzZduCLzrTia2cyvk0/ZM/iZx4mER\n"
    "dEr/VxqHD3VILs9RaRegAhJhldXRQLIQTO7ErBBDpqWeCtWVYpoNz4iCxTIM5Cuf\n"
    "ReYNnyicsbkqWletNw+vHX/bvZ8=\n"
    "-----END CERTIFICATE-----\n";

#include <WiFiClientSecure.h>
#include <WiFi.h>
#include <HTTPClient.h>

void setClock();
void connect();
void setup()
{
    Serial.begin(115200);

    WiFi.mode(WIFI_STA);
    
    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

    WiFi.begin("xxxxxx","xxxxxxx");

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
[... 159 lines cut ...]
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
```

I've tried tweaking WiFiClientSecure.ino example with the following results : 

```
Starting connection to server...
[V][ssl_client.cpp:56] start_ssl_client(): Free internal heap before TLS 174688
[V][ssl_client.cpp:58] start_ssl_client(): Starting socket
[V][ssl_client.cpp:93] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:102] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:115] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:180] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:195] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-80) UNKNOWN ERROR CODE (0050)
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -80
[V][ssl_client.cpp:248] stop_ssl_socket(): Cleaning SSL connection.
Connection failed!
UNKNOWN ERROR CODE (0050)
```

It seems has a returning error message (unknown one) with -0x50 (-80), I tried to have a meaning with no success. 

I appreciate any kind of help.

closed 2020-11-28 by stale[bot]

lrodorigo, 2021-03-07:
Same (random) issue here,  using **espressif32@3.0.0**
using SSL Client to connect to MQTT TLS Broker (Google IoT Core).

```
[E][ssl_client.cpp:33] _handle_error(): [start_ssl_client():199]: (-80) UNKNOWN ERROR CODE (0050)
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -80
```

HamzaHajeir, 2021-03-08:
I've not proceed with this, Changed the approach to using AES encryption to the payload