espressif/arduino-esp32 #4344: ESP32 constantly rebooting

josetomas, 2020-09-17:
Hardware:
Adafruit feather ESP32 WROOM - 32
IDE: Platformio
Upload Speed: 115200
Computer OS: Windows
Core Installation version: No idea, as it came from fabric
Flash frequency: 40 MHz

### Description:
I'm trying to move some motors, when the ESP32 gets a variable from an http client. But it is constantly rebooting. 

### Sketch: 
```
#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>
#include "analogWrite.h"

/***********CONFIGURATION************
 * This section will define all the variables the user is spected to
 * configurate, please don't modify the other sections*/

bool wifiPass = false; //True if the WiFi network has password; False if not. 
const char* ssid = "MIT"; //WiFi ssid
const char* password = ""; //WiFi password

/*******CONECTIONS: Modify the next 2 lines to define the pins your are going to use. 
 * Dont modifiy nothing of you are using the same conections presented in the Fritzing 
 * image. */

const int pinMotor[]= {12,27,15,33,22,23};// {moto1CW,motor1CCW,moto2CW,motor2CCW,moto3CW,motor3CCW}
const int pinEnable[]={13,32,14};

const int motorSpeed = 255; //Motor speed, this variable will define the motor speed, to define the number follow the following ecuation: Max_RPM*motorSpeed/255=Speed_expected. 

/******DON NOT MODIFY PASS HERE*********/

const char* serverNameUp = "http://10.31.209.244/UP_Client";
const char* serverNameDown = "http://10.31.209.244/DOWN_Client";
const char* serverNameLeft = "http://10.31.209.244/LEFT_Client";
const char* serverNameRight = "http://10.31.209.244/RIGHT_Client";
const char* serverNameCW = "http://10.31.209.244/CW_Client";
const char* serverNameCCW = "http://10.31.209.244/CCW_Client";
const char* serverNameStop = "http://10.31.209.244/STOP_Client";

String httpGETRequest(const char* server);
void moveMotor(int motor, String direction); 
void forward();
void left();
void right();
void left();
void down();
void CCW();
void CW();
void scan();
void stop();

void setup() {
for (byte i=0; i< sizeof(pinMotor); i++ ){
  pinMode(pinMotor[i],OUTPUT);
}
for (byte i=0; i< sizeof(pinEnable); i++ ){
  pinMode(pinEnable[i],OUTPUT);
}
Serial.begin(115200);
if (wifiPass){
  WiFi.begin(ssid,password);
}
else
{
WiFi.begin(ssid);
}
while(WiFi.status() != WL_CONNECTED) { 
    delay(500);
    Serial.print(".");
}
Serial.println("");
Serial.print("Connected to WiFi network with IP Address: ");
Serial.println(WiFi.localIP());
moveMotor(1,"Off");
moveMotor(2,"Off");
moveMotor(3,"Off");
}

void loop() {
  scan();
}

void scan(){
  String upStatus = httpGETRequest(serverNameUp);
  if(upStatus=="move"){
    forward();
    Serial.println("forward");
  }
  String downStatus = httpGETRequest(serverNameDown);
  if(downStatus=="move"){
    down();
    Serial.println("down");
  }
  String leftStatus = httpGETRequest(serverNameLeft);
  if(leftStatus=="move"){
    left();
    Serial.println("left");
  }
  String rightStatus = httpGETRequest(serverNameRight);
  if(rightStatus=="move"){
    right();
    Serial.println("right");
  }
  String cwStatus = httpGETRequest(serverNameCW);
  if(cwStatus=="move"){
    CW();
    Serial.println("cw");
  }
  String ccwStatus = httpGETRequest(serverNameCCW);
  if(ccwStatus=="move"){
    CCW();
    Serial.println("ccw");
  }
  String stopStatus = httpGETRequest(serverNameStop);
  if(stopStatus=="dont move"){
    stop();
    Serial.println("stop");
  }
}

String httpGETRequest(const char* server) {
  HTTPClient http;
  http.begin(server);
  int httpResponseCode = http.GET();
  String payload = "--"; 
  if (httpResponseCode>0) {
    payload = http.getString();
  }
  else {
  }
  http.end();

  return payload;
}
//Standar functions for bluetooth and WiFi
void forward(){
  moveMotor(1,"CCW");
  moveMotor(2,"CW");
  moveMotor(3,"Off");
  delay(100);
}

void left(){
  moveMotor(1,"CCW");
  moveMotor(2,"CCW");
  moveMotor(3,"CW");
}

void right(){
  moveMotor(1,"CW");
  moveMotor(2,"CW");
  moveMotor(3,"CCW");
}

void down(){
  moveMotor(1,"CW");
  moveMotor(2,"CCW");
  moveMotor(3,"Off");
}

void stop(){
  moveMotor(1,"Off");
  moveMotor(2,"Off");
  moveMotor(3,"Off");
}

void CW(){
  moveMotor(1,"CW");
  moveMotor(2,"CW");
  moveMotor(3,"CW");
}

void CCW(){
  moveMotor(1,"CCW");
  moveMotor(2,"CCW");
  moveMotor(3,"CCW");
}

void moveMotor(int motor, String direction){
  switch (motor){
    case 1:
     digitalWrite(pinEnable[0],HIGH);
      if(direction == "CCW"){
        digitalWrite(pinMotor[0],LOW);
        digitalWrite(pinMotor[1],HIGH);
        Serial.print("Motor1: ");
        Serial.println("CCW");
      }
      else if (direction=="CW")
      {
        digitalWrite(pinMotor[0],HIGH);
        digitalWrite(pinMotor[1],LOW);
        Serial.print("Motor1: ");
        Serial.println("CW");
      }
      else 
      {
        digitalWrite(pinEnable[0],LOW);
        digitalWrite(pinMotor[0],LOW);
        digitalWrite(pinMotor[1],LOW);
        Serial.print("Motor1: ");
        Serial.println("Off");
      }
      break;
    case 2: 
      digitalWrite(pinEnable[1],HIGH);
      if(direction == "CCW"){
        digitalWrite(pinMotor[2],LOW);
        digitalWrite(pinMotor[3],HIGH);
        Serial.print("Motor2: ");
        Serial.println("CCW");
      }
      else if (direction=="CW")
      {
        digitalWrite(pinMotor[2],HIGH);
        digitalWrite(pinMotor[3],LOW);
        Serial.print("Motor2: ");
        Serial.println("CW");
      }
      else 
      {   
        digitalWrite(pinEnable[1],LOW);
        digitalWrite(pinMotor[2],LOW);
        digitalWrite(pinMotor[3],LOW);
        Serial.print("Motor2: ");
        Serial.println("Off");
      }
      break;
    case 3:
      analogWrite(pinEnable[2],motorSpeed);
      if(direction == "CCW"){
        digitalWrite(pinMotor[4],LOW);
        digitalWrite(pinMotor[5],HIGH);
        Serial.print("Motor3: ");
        Serial.println("CCW");
      }
      else if (direction=="CW")
      {
        digitalWrite(pinMotor[4],HIGH);
        digitalWrite(pinMotor[5],LOW);
        Serial.print("Motor3: ");
        Serial.println("CW");
      }
      else 
      {
        analogWrite(pinEnable[2],LOW);
        digitalWrite(pinMotor[4],LOW);
        digitalWrite(pinMotor[5],LOW);
        Serial.print("Motor3: ");
        Serial.println("Off");
      }
      break;
  }
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
.
Connected to WiFi network with IP Address: 10.31.227.148
Motor1: Off
Motor2: Off
Motor3: Off
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060830  A0      : 0x800d2b02  A1      : 0x3ffb1df0  
A2      : 0x3ffb1e70  A3      : 0x00000000  A4      : 0x3ffcb928  A5      : 0x0000000e  
A6      : 0x3ffcba10  A7      : 0x0000000e  A8      : 0x8015e832  A9      : 0x3ffb1dd0  
A10     : 0x3ffba7e0  A11     : 0x00000000  A12     : 0x0000000e  A13     : 0x00000000  
A14     : 0x0000000e  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x00000000:0x3ffb1df0 0x400d2aff:0x3ffb1e10 0x400d2b71:0x3ffb1e30 0x400d1267:0x3ffb1e50 0x400d1a1a:0x3ffb1f10 0x400d1c13:0x3ffb1f90 0x400d6161:0x3ffb1fb0 0x40088b9d:0x3ffb1fd0

Rebooting...

```

lbernstone, 2020-09-17:
This forum is for issues with the repository codebase, not your programs.
https://learn.adafruit.com/adafruit-arduino-lesson-13-dc-motors/transistors

josetomas, 2020-09-17:
Ok, thanks!

closed 2020-09-17 by josetomas