espressif/arduino-esp32 #4345: Arduino IDE isn't recognizing ESPASyncWebServer

josetomas, 2020-09-17:
### Hardware:
Adafruit feather ESP32 WROOM - 32
IDE: Platformio
Upload Speed: 115200
Computer OS: Windows
Core Installation version: No idea, as it came from fabric
Flash frequency: 40 MHz

### Description:
The Arduino IDE isn't recognizing the ESPAsyncWebServer.h.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
Same problem with every sketch. 

### Debug Messages:
```
C:\Users\corne\OneDrive\Documentos\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:33:22: fatal error: AsyncTCP.h: No such file or directory
compilation terminated.
exit status 1
Error compilando para la tarjeta Adafruit ESP32 Feather.
```

lbernstone, 2020-09-17:
This forum is for issues with the repository codebase, not your programs.

josetomas, 2020-09-17:
Sorry, I thought ESPAsyncWebServer was also a librarie of your repo. If anyone has the same problem you can follow the issue [here](https://github.com/me-no-dev/ESPAsyncWebServer/issues/848)

closed 2020-09-17 by josetomas