espressif/arduino-esp32 #4346: ESP LOG redirect to own function bug.

zekageri, 2020-09-17:
If i set the `esp_log_set_vprintf(redirectToF_System);` than my function is never called when ESP crashes.

```
static char log_print_buffer[512];
static char APP_NAME[] = "My_Home";
static char Crash_File_Path[] = "/Crash_LOGS.txt";

int redirectToF_System(const char *szFormat, va_list args) {
    int ret = vsnprintf(log_print_buffer, sizeof(log_print_buffer), szFormat, args);
    if (ret >= 0){
        File g;
        if(LITTLEFS.exists(Crash_File_Path)){
            g = LITTLEFS.open(Crash_File_Path, "a");
        }else{
            g = LITTLEFS.open(Crash_File_Path, "w");
        }
        if (!g) {}else{
            g.write((uint8_t *)log_print_buffer, (size_t)ret);
        }
        g.close();
    }
    return ret;
}
static const inline void Crash_Log_Init(){
    esp_log_set_vprintf(redirectToF_System);
    esp_log_level_set("*", ESP_LOG_VERBOSE);
    //LITTLEFS.remove(Crash_File_Path);
    ESP_LOGI(APP_NAME, "Writing via ESP_LOGI!\n");
    esp_log_write(ESP_LOG_VERBOSE, APP_NAME, "I can write like this all day.\n"); // THIS LINE IS WORKING PROPERLY!
}
void setup(){
Serial.begin(115200);
Crash_Log_Init();
}
```

lbernstone, 2020-09-17:
Two issues:
1) `esp_log_level_set("*", ESP_LOG_VERBOSE);` does not change the logging level.  Use ESP_LOGE in your testing.  AFAIK you will need to rebuild the libraries to change the IDF logging level.
2) esp32-hal-log.h redefines all the log functions such that vprintf is not used.  If you modify the file as follows, you can redirect to file.
```
#ifndef __ARDUHAL_LOG_H__
#define __ARDUHAL_LOG_H__

#ifdef __cplusplus
extern "C"
{
#endif

#include "sdkconfig.h"
#include "esp_timer.h"

#define ARDUHAL_LOG_LEVEL_NONE       (0)
#define ARDUHAL_LOG_LEVEL_ERROR      (1)
#define ARDUHAL_LOG_LEVEL_WARN       (2)
#define ARDUHAL_LOG_LEVEL_INFO       (3)
#define ARDUHAL_LOG_LEVEL_DEBUG      (4)
#define ARDUHAL_LOG_LEVEL_VERBOSE    (5)

const char * pathToFileName(const char * path);
int log_printf(const char *fmt, ...);

#define log_v(format, ...)
#define isr_log_v(format, ...)

#define log_d(format, ...)
#define isr_log_d(format, ...)

#define log_i(format, ...)
#define isr_log_i(format, ...)

#define log_w(format, ...)
#define isr_log_w(format, ...)

#define log_e(format, ...)
#define isr_log_e(format, ...)

#define log_n(format, ...)
#define isr_log_n(format, ...)

#include "esp_log.h"

#ifdef __cplusplus
}
#endif

#endif /* __ESP_LOGGING_H__ */
```

zekageri, 2020-09-17:
> Two issues:
>
> 1. `esp_log_level_set("*", ESP_LOG_VERBOSE);` does not change the logging level.  Use ESP_LOGE in your testing.  AFAIK you will need to rebuild the libraries to change the IDF logging [...]

Thank you for your support. I will try it with your suggestions and report back!

zekageri, 2020-09-18:
I don't really understand what should i modify.

In the `esp32-hal-log.h` there is one reference to `const char * pathToFileName(const char * path);` and it is used in one definition:

`#define ARDUHAL_LOG_FORMAT(letter, format)  ARDUHAL_LOG_COLOR_ ## letter "[" #letter "][%s:%u] %s(): " format ARDUHAL_LOG_RESET_COLOR "\r\n", pathToFileName(__FILE__), __LINE__, __FUNCTION__`

**When i do:**

```
static const inline void Crash_Log_Init(){
    esp_log_set_vprintf(redirectToF_System);
    ESP_LOGE(APP_NAME, "Writing via ESP_LOGI!\n");
    esp_log_write(ESP_LOG_ERROR, APP_NAME, "I can write like this all day.\n");
}
```
**I get some loggings to my file:**

```
I can write like this all day.
I can write like this all day.
I can write like this all day.
E (1710913) esp_littlefs: Failed to unlink path "/Home_index.html". Has open FD.
E (2672737) esp_littlefs: Failed to unlink path "/Home_index.html". Has open FD.
E (3012752) esp_littlefs: Failed to unlink path "/Home_index.html". Has open FD.
E (4303186) esp_littlefs: Failed to unlink path "/Home_index.html". Has open FD.
I can write like this all day.
I can write like this all day.
I can write like this all day.
I can write like this all day.
I can write like this all day.
I can write like this all day.
```

lbernstone, 2020-09-18:
What I have showed you is how to rip out the arduino logging, so there is no ARDUHAL_LOG_FORMAT.  You will now need to fill those empty defines with the format that you want, and have them use your new function.

closed 2020-09-24 by zekageri

martirius, 2021-02-22:
Hi. I have opened a PR related to this issue:
https://github.com/espressif/arduino-esp32/pull/4845

tbertels, 2021-10-05:
> Hi. I have opened a PR related to this issue: #4845

With this PR and adding `-DUSE_ESP_IDF_LOG` to build.extra_flags platform.txt and removing the `#define ESP_LOG` lines [in esp32-hal-log.h](https://github.com/espressif/arduino-esp32/blob/1.0.4/cores/esp32/esp32-hal-log.h#L142-L146) (per https://stackoverflow.com/questions/60442350/arduinos-esp-log-set-vprintf-does-not-work-on-esp32 question comments), I can get some logging messages written to SPIFFS. Logs messages related to Wifi and such.

But I still can't get crash logs written to SPIFFS. They're still getting sent to UART.

Did anybody manage to make it work?

zekageri, 2021-10-20:
> Hi @tbertels , I don't think that is possible to do, because wiritng on the SPIFFS it's an heavy work and something you can't do in an interrupt.
> 

Why can't we add to queue and write in another task?

zekageri, 2022-04-22:
I'm still trying. Nothing seems to work

awaisnaeem1004, 2022-05-01:
Hi @martirius @tbertels,

I am facing an implementation issue regarding the use of the above PR. When I use `log_x()` by defining `USE_ESP_IDF_LOG` and `DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE` I only get the logs printed on serial monitor for `log_e()` and not for any other levels i.e., `log_d()`, `log_w()`, `log_i()`, etc.

However, when I only define DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE, I get the logs printed for each level.

Sketch: 
```
#include <Arduino.h>

void setup()
{
    Serial.begin(115200);
    Serial.println("Configuring the SPIFFS of ESP32...");
    log_e("[Error] Hey there");
    log_w("[Warn] Hey There");
    log_i("[Info] Hey There");
    log_d("[Debug] Hey There");
    log_v("[Verbose] Hey there");
}

void loop()
{
}
```

Output: [With `DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE`]
```
Configuring the SPIFFS of ESP32...
[E][ESP32_Logging_SPIFFS_WebServer.cpp:18] setup(): [Error] Hey there
[W][ESP32_Logging_SPIFFS_WebServer.cpp:19] setup(): [Warn] Hey There
[I][ESP32_Logging_SPIFFS_WebServer.cpp:20] setup(): [Info] Hey There
[D][ESP32_Logging_SPIFFS_WebServer.cpp:21] setup(): [Debug] Hey There
[V][ESP32_Logging_SPIFFS_WebServer.cpp:22] setup(): [Verbose] Hey there
```

Output: [With `DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_VERBOSE` and `USE_ESP_IDF_LOG`]
```
Configuring the SPIFFS of ESP32...
E (28) ARDUINO: [Error] Hey there
```

Can you please guide me in the right direction as to why this might be happening? Do I need to set some logging define or anything else? Thanks.