espressif/arduino-esp32 #4348: esp32s2 tree: Slow response from HelloServer.ino and ICMP weirdness

Miraculix200, 2020-09-18:
### Hardware/Software:

Arduino IDE version: 1.8.13
Board: ESP32-S2 Wrover
SDK Version: v4.3-dev-907-g6c17e3a64-dirty
Arduino ESP32 core: 1_0_4
PSRAM enabled: yes and no
Upload speed: 921600
OS: Windows 10

### Description

Using the HelloServer.ino example on a ESP32-S2 Wrover I get slow responses when pointing my browser at the ESP32-S2 IP address.

A similar server (with ESP8266 default libraries) running on an ESP-12F or ESP32-CAM has reasonable response times (both have not been compiled with the esp32s2 tree of the Arduino IDE however).

According to the network request monitor in Firefox, connecting takes 600ms. Then there is a wait period of over 2000ms. After that, receiving takes 0ms (so it's most likely not a connection speed problem). So when 3 pages are requested at once (e.g. 1 html, 1 CSS and 1 JS) the response needs over 6 seconds.

Interesting is the ping response from the ESP32-S2.

Right after reset, before pointing the browser at the webserver:

```
PING 192.168.0.29 (192.168.0.29) 56(84) bytes of data.
64 bytes from 192.168.0.29: icmp_seq=2 ttl=255 time=4.97 ms
64 bytes from 192.168.0.29: icmp_seq=3 ttl=255 time=2.77 ms
64 bytes from 192.168.0.29: icmp_seq=4 ttl=255 time=10.5 ms
64 bytes from 192.168.0.29: icmp_seq=5 ttl=255 time=7.51 ms
```

After pointing the browser at the webserver (loading the page only 1 time):

```
PING 192.168.0.29 (192.168.0.29) 56(84) bytes of data.
64 bytes from 192.168.0.29: icmp_seq=1 ttl=255 time=84.2 ms
64 bytes from 192.168.0.29: icmp_seq=2 ttl=255 time=1085 ms
64 bytes from 192.168.0.29: icmp_seq=3 ttl=255 time=77.4 ms
64 bytes from 192.168.0.29: icmp_seq=4 ttl=255 time=1074 ms
```

The first connection after resetting the board sometimes produces an immediate response from the webserver.

**Maybe related problem**: When I set a static IP in WiFi config, the device connects to the router but does not respond to TCP or ping requests (maybe related?). It only works when letting DHCP choose the IP address.

What I've tried to fix the issue:

- Disabling WIFI powersaving ( esp_wifi_set_ps(WIFI_PS_NONE); )
- Disabling PSRAM
- Using a AMS1117 voltage regulator instead of MCP1825
- Adding a 1000uF electrolytic cap
- Adding a 100nF ceramic cap

I designed the PCB myself (and I'm a PCB design beginner). Here is the schematic and PCB: https://easyeda.com/Miraculix200/esp32-s2-test

Anyway, I think a hardware problem can be ruled out, because the ping response is fine BEFORE connecting to the webserver.

### Compiler messages

```
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Using library ESPmDNS at version 1.0 in folder: C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS 
Using library WebServer at version 1.0 in folder: C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer 
Using library WiFi at version 1.0 in folder: C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 
Using library FS at version 1.0 in folder: C:\Users\xxx\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS 
```

Unfortunately I wasn't able to install ESP-IDF properly, neither using Eclipse nor Visual Studio Code, so I can't tell if it's a problem with the esp32s2 tree of Arduino IDE, or something else.

lbernstone, 2020-09-18:
I'm unable to reproduce with a saola-1_v1.2
Chrome has a known issue with Webserver.  It likes to leave requests hanging, but that's the signal that WebServer is using to send.  Test from the command line with wget to see if the issue is client specific.  Use EspAsyncWebServer if you intend to deliver content through http.
I can ping just fine with a static address.  It is slow when modem sleep is enabled, as expected.  Make sure your subnetting is correct.

Miraculix200, 2020-09-18:
It's not the browser. Before connecting with "w3m" (Linux command line browser) the ping response is fast. After connecting with "w3m" the ping response is fluctuating. And it keep fluctuating for minutes (I don't know if it ever stops).

The same web page works fine on ESP12 and ESP32-CAM. 

Subnetting is also correct, as I simply copied the code from my ESP12, and only changed the IP. It connects to the router with the correct static IP. But it doesn't respond to anything.

Did you use the most recent version of the esp32s2 tree for compiling?

lbernstone, 2020-09-18:
Ok, I am able to reproduce the slowdown now.  This looks to be an issue in WebServer, not esp32s2.  I'll have to think about exactly why it presents like this, but if you add a delay, even delay(1), in loop, it will behave.

Miraculix200, 2020-09-19:
I can confirm that delay(1) in the loop() fixes it. And my (large) code doesn't crash the board anymore when PSRAM is disabled and I connect to the webserver. Most likely a problem with not enough free heap though, not directly related.

closed 2020-09-19 by Miraculix200

Bayliner4387, 2021-06-12:
Was this really resolved? I tried the Delay(1) solution but with no luck. Seems to be more of a problem the more callbacks. Changing the callback timing so there is less likelihood of timing collisions helps but its still very slow by comparison to a 8266

lbernstone, 2021-06-12:
Issue was solved and closed.  If you have a reproducible issue, open a new issue and follow the issue template.

Bayliner4387, 2021-06-12:
I managed to "Band-Aid" this problem. The Delay(1) before WebServer.handleClient didnt help so on a guess it tried replacing the Delay call with a more "Non-Blocking solution". Hope this helps others.

void DoEvents(int mils) {
    unsigned long currentMillis = millis();
    if (currentMillis + mils > 4294967295)
        delay(mils);
    else
        while ((currentMillis + mils) > millis()) {
            server.handleClient();
            yield();
        }
}

scottchiefbaker, 2023-03-10:
This `delay()` solution is causing major slowdowns in [other places](https://github.com/espressif/arduino-esp32/issues/7708). Since everyone puts `server.handeClient()` in `loop()` this essentially limits your main `loop()` to never going higher than 1000Hz.

I'm not an internals expert, but wouldn't `yield()` make more sense than this?

PR #4350 (2020-10-01): WebServer.handleClient delay
If WebServer.handleClient is run in a tight loop, it will starve other processes.  So, if there is no connection, throw in a delay(1).  Fixes #4348
commit 9e7b13e4 (2020-10-01): WebServer.handleClient delay (#4350)

* If WebServer.handleClient is run in a tight loop, it will starve other processes.  So, if there is no connection, throw in a delay(1).  Fixes #4348

* Made a variable to control the delay behavior