espressif/arduino-esp32 #4351: Can't read HID report map over BLE

sasodoma, 2020-09-19:
### Hardware:
Board:							NodeMCU-32S
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80 MHz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
Hello, I have a Logitech MX Anywhere 2S mouse, that I'm trying to connect to over BLE. For testing purposes I modified the BLE_client example to allow me to input first a service UUID and then a characteristic UUID. The sketch then reads the characteristic and outputs the value. The sketck works with 180a and 2a29 (Manufacturer name) and also 180f and 2a19 (battery level - ASCII value for 2 is 50, battery is 50%). But when I try 1812 and 2a4b (HID report map) the value is apparently empty. Is there a special procedure requred to read the HID report map or am I doing something wrong?

### Serial output:
```
BLE Advertised Device found: Name: MX Anywhere 2S, Address: ce:57:0e:06:32:79, appearance: 962, serviceUUID: 00001812-0000-1000-8000-00805f9b34fb, txPower: 4
Forming a connection to ce:57:0e:06:32:79
 - Created client
 - Connected to server
We are now connected to the BLE Server.
 - Found our service: 0000180a-0000-1000-8000-00805f9b34fb
 - Found our characteristic: 00002a29-0000-1000-8000-00805f9b34fb
The characteristic value was: Logitech
 - Found our service: 00001812-0000-1000-8000-00805f9b34fb
 - Found our characteristic: 00002a4b-0000-1000-8000-00805f9b34fb
The characteristic value was: 
 - Found our service: 0000180f-0000-1000-8000-00805f9b34fb
 - Found our characteristic: 00002a19-0000-1000-8000-00805f9b34fb
The characteristic value was: 2
```

### Sketch:
```cpp

/**
   A BLE client example that is rich in capabilities.
   There is a lot new capabilities implemented.
   author unknown
   updated by chegewara
*/

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("00001812-0000-1000-8000-00805f9b34fb");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("00002a4b-0000-1000-8000-00805f9b34fb");

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;
BLEClient*  pClient;
BLERemoteService* pRemoteService;

boolean haveChar = true;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  Serial.print("Notify callback for characteristic ");
  Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
  Serial.print(" of data length ");
  Serial.println(length);
  Serial.print("data: ");
  Serial.println((char*)pData);
}

class MyClientCallback : public BLEClientCallbacks {
    void onConnect(BLEClient* pclient) {
    }

    void onDisconnect(BLEClient* pclient) {
      connected = false;
      Serial.println("onDisconnect");
    }
};

bool connectToServer() {
  Serial.print("Forming a connection to ");
  Serial.println(myDevice->getAddress().toString().c_str());

  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");

  pClient->setClientCallbacks(new MyClientCallback());

  // Connect to the remove BLE Server.
  pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
  Serial.println(" - Connected to server");

  connected = true;
  return true;
}
/**
   Scan for BLE servers and find the first one that advertises the service we are looking for.
*/
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    /**
        Called for each advertising BLE server.
    */
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.print("BLE Advertised Device found: ");
      Serial.println(advertisedDevice.toString().c_str());

      // We have found a device, let us now see if it contains the service we are looking for.
      //if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {
      if (!strcmp(advertisedDevice.getName().c_str(), "MX Anywhere 2S")) {
        BLEDevice::getScan()->stop();
        myDevice = new BLEAdvertisedDevice(advertisedDevice);
        doConnect = true;
        doScan = true;

      } // Found our server
    } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 5 seconds.
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, false);
} // End of setup.

// This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }
  if (Serial.available()) {
    if (haveChar) {
      haveChar = false;
      String incoming = Serial.readString();
      serviceUUID = BLEUUID::fromString(incoming.c_str());
    } else {
      haveChar = true;
      String incoming = Serial.readString();
      charUUID = BLEUUID::fromString(incoming.c_str());

      if (connected) {
        // Obtain a reference to the service we are after in the remote BLE server.
        pRemoteService = pClient->getService(serviceUUID);
        if (pRemoteService == nullptr) {
          Serial.print("Failed to find our service UUID: ");
          Serial.println(serviceUUID.toString().c_str());
          pClient->disconnect();
        }
        Serial.print(" - Found our service: ");
        Serial.println(serviceUUID.toString().c_str());

        // Obtain a reference to the characteristic in the service of the remote BLE server.
        pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
        if (pRemoteCharacteristic == nullptr) {
          Serial.print("Failed to find our characteristic UUID: ");
          Serial.println(charUUID.toString().c_str());
          pClient->disconnect();
        }
        Serial.print(" - Found our characteristic: ");
        Serial.println(charUUID.toString().c_str());

        // Read the value of the characteristic.
        if (pRemoteCharacteristic->canRead()) {
          std::string value = pRemoteCharacteristic->readValue();
          Serial.print("The characteristic value was: ");
          Serial.println(value.c_str());
        }

      //  if (pRemoteCharacteristic->canNotify())
        //  pRemoteCharacteristic->registerForNotify(notifyCallback);
      }
    }
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {

  } else if (doScan) {
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }

  delay(1000); // Delay a second between loops.
} // End of loop
```

chegewara, 2020-09-21:
> Is there a special procedure requred to read the HID report map or am I doing something wrong?

No, you should be able to read it. Did you try with nRF connect?

sasodoma, 2020-09-21:
> > Is there a special procedure requred to read the HID report map or am I doing something wrong?
> 
> No, you should be able to read it. Did you try with nRF connect?

Yes, but with Android you are not allowed to read HID without a system signature. It's a safety and that's why I went to an ESP32 in the first place. My original idea was an Android app.

chegewara, 2020-09-21:
Youre right, i forgot about it, sorry.

> Perhaps the mouse does something strange.

Its a standard, code cant make anything strange.

sasodoma, 2020-09-21:
Is it possible that some sort of authentication is required? I know on my phone or laptop I don't need a code to pair the mouse but there is probably still some encryption happening in the background right?

closed 2020-12-04 by stale[bot]

sasodoma, 2020-12-29:
So I decided that the best thing to do is to buy an NRF dongle and try to read the report map with that. I installed nRF Connect on Windows and connected to my mouse. When trying to read the report map I get "BLE_GATT_STATUS_ATTERR_INSUF_AUTHENTICATION". If I then pair the mouse, the security gets updated to mode 1 and level 2. After that I can read the report map sucessfully. I'm guessing that in order to read the report map with my ESP I need to somehow raise the security level to 2. Is that something that can be done?

sasodoma, 2020-12-29:
Ok so it turns out that the problem here was, that the mouse didn't allow reading the HID service over an unencrypted connection. The solution was to add `BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);` before the createClient command, after this I was able to read the report map correctly.