espressif/arduino-esp32 #4355: advertisedDevice.haveServiceUUID() always returns false

nathan-casabieille, 2020-09-21:
### Hardware:
Board: ESP-WROOM-32
Core Installation version: 1.0.4
IDE name: Arduino IDE
Flash Frequency: 80Mhz
PSRAM enabled: no
Upload Speed: 115200
Computer OS: Arch Linux

Hello,

I would like to connect my esp32 to a Bluetooth watch to retrieve my medical data.

As I have several watches, I just have to ask the esp32 to connect to the device if it has one of these services:

- Generic Access:
  UUID: 0x1800
   - Device Name
     UUID: 0x2A00
     Properties: READ, WRITE
  - Appearance
    UUID: 0x2A01
    Properties: READ
- Peripheral Preffered Connection Parameters
    UUID: 0x2A04
    Properties: READ
    - Central Address Resolution
      UUID: 0x2AA6
      Properties: READ
- Generic Attribute:
  UUID: 0x1801
  - Service Changed
    UUID: 0x2A05
    Properties: INDICATE
    Descriptors:
    - Client Characteristic Configuration
      UUID: 0x2902
- Secure DFU Service:
  UUID: 0xFE59
  - Buttonless DFU
     UUID: 8ec90003-f315-4f60-9fb8-838830daea50
     Properties: INDICATE, WRITE
     Descriptors:
     - Client Characteristic Configuration
       UUID: 0x2A29
       Properties: READ
- Device Information
  UUID: 0x180A
  - Manufacturer Name String
    UUID: 0x2A29
    Properties: READ
- Unknown Service:
  UUID: 14839ac4-7d7e-415c-9a42-167340cf2339
  - Unknown Characteristic:
    UUID: 8b00ace7-eb0b-49b0-bbe9-9aee0a26e1a3
    Properties: WRITE NO RESPONSE
  - Unknown Characteristic:
    UUID: 0734594a-a8e7-4b1a-a6b1-cd5243059a57
    Properties: NOTIFY
    Descriptors:
    - Client Characteristic Configuration
      UUID: 0x2902

So here is what I wrote:

```
BLEUUID serviceUUID("14839ac4-7d7e-415c-9a42-167340cf2339");

void MyAdvertisedDeviceCallbacks::onResult(BLEAdvertisedDevice advertisedDevice)
{
    if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {
        // connect to device
    }
}
```
However, when it detects a watch, it does not connect to it.

It was then that I realized that the function `haveServiceUUID()` always returns false while it shouldn't be.

Does anyone know the cause of this problem ?

Thank you very much.

chegewara, 2020-09-21:
Hi,
all info you provided here has nothing to do with `haveServiceUUID()`. What you need is to check advertising info/data and info you provided is taken after connecting.

closed 2020-10-07 by nathan-casabieille

Rob58329, 2025-07-26:
Note also that the "advertisedDevice.haveServiceUUID()`" command (eg as used in the current "ESP32 BLE Client example") also returns false/wont work for BLE-servers which don't transmit the "ServiceUUID" in the Initial Advertising packet.

For such devices (eg. Shelly Home-Automation-Relays) the ServiceUUID is sent after the initial-Advertising-Packet, in a later "Scan Responce Packet".  However, I  cant currently find a way to access this in arduino (eg. with "BLESCANPARAM" or similar), but instead it's possible to use the Device's MAC address (which IS sent in the Initial Advertising packet) to identify the Device you want (and so you can then just connect to it and start looking for the CharacteristicUUID you want).