espressif/arduino-esp32 #4363: OTA CHANGE

beeedee, 2020-09-27:
The size of my application is 2.9MB and I am using Arduino IDE
My new custom partition table below ;

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0xC0000, //webuptade application
app1,     app,  ota_1,   0xD0000, 0x330000, // my application

app0 partition for webuptade application and app1partition for my application.
I can install my app on app1 partition with webuptade application.
But when necessary, I have to go to the other section by pressing a button in the application.
After all, I will not have installed the applications 2 times.

How can I do that ?

beeedee, 2020-09-27:
Looks like it will happen if I update Otadata from the app and use the code ESP.restart(). but what is the code to update otadata.

lbernstone, 2020-09-27:
https://docs.espressif.com/projects/esp-idf/en/v3.3.3/api-reference/system/ota.html

beeedee, 2020-09-27:
I do not have enough command of English to understand the document. Can't you sample it with a code for me?

beeedee, 2020-09-27:
and I could not understand if it is possible to do this in arduino ide :(

lbernstone, 2020-09-27:
To switch to the not running OTA partition, you can call:
```
#include <esp_ota_ops.h>
esp_ota_set_boot_partition(esp_ota_get_next_update_partition(NULL));
```

beeedee, 2020-09-27:
You are a magnificent detail, thank you.

closed 2020-10-08 by everslick