espressif/arduino-esp32 #4366: Delta firmware OTA (DFOTA)

nzagorec, 2020-09-28:
Hi,

Has anyone tried to generate delta between two binaries and use it to upgrade the firmware on ESP32?
Would this require some support from IDF or can it be done completely in code?
I'd like to shrink my upgrade packages as much as possible so I can use LPWAN to transfer them to the MCU.

Thanks!
N

lbernstone, 2020-09-28:
You will certainly need to use the idf functions to pull and modify the ota partitions, but it should be possible to it with a stream oriented diff method and the release version arduino libraries.

nzagorec, 2020-09-29:
Thanks @lbernstone !

I tried using bsdiff to generate delta between two binaries and it worked great, delta bin is only 17KB big. When I convert it to hex it's 75KB which is still ok to transfer.

Imagine I have this hex on my ESP's flash, what would be the next step towards re-constructing an original binary?

Appreciate your help!
N

lbernstone, 2020-09-29:
This forum is for discussing issues about the codebase.  For general discussion or fishing for someone to write your code, please take it to esp32.com.

nzagorec, 2020-09-29:
No need to be rude, I'm not fishing for anything, just wanted someone to point me into right direction. Will tackle it on my own. 

Cheers!
N

closed 2020-09-29 by nzagorec