espressif/arduino-esp32 #4368: Configuring a new IP in web server causes a crash

kordhrufan, 2020-09-29:
Trying to config a new IP, gateway and mask causes a crash.

What you are trying to do?
Hi, I'm trying to configure a web page where you can change its IP, gateway and mask of the server.

Describe your system( Hardware, computer, O/S, core version, environment).
Olimex ESP32-EVB, Windows 10, platformio
Describe what is failing.
When you use the WiFi.config funcion it causes a crash. This is after initiating a DHCP connection and starting the web server
Show the shortest possible code that will duplicate the error.

### Hardware:
Board:							?ESP32-EVB
Core Installation version:			        ?1.0.3
IDE name:						?Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? 
Upload Speed:					     ?115200?
Computer OS:                                        ?Windows 10?

### Description:
Trying to config a new IP, gateway and mask causes a crash after sending an HTTP_POST on server.

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include "SPIFFS.h"

// Replace with your network credentials
const char* ssid = "x";
const char* password = "x";

const char* PARAM_INPUT_1 = "ip";
const char* PARAM_INPUT_2 = "gateway";
const char* PARAM_INPUT_3 = "mask";
// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

void setup() {
  Serial.println("Inicio");
  Serial.begin(115200);
  if(!SPIFFS.begin(true)){
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  server.begin();
   // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", "text/html");
  });
  delay(1000);
  Serial.print("IP config:");
  Serial.println(WiFi.localIP());
  Serial.println("Preparandose para empezar");
  // Send a POST request to <ESP_IP>/update
  server.on("/update", HTTP_POST, [] (AsyncWebServerRequest *request) {
      IPAddress localIp(192,168,0,50);
      IPAddress gateway(192,168,0,1);
      IPAddress mask(255,255,255,0);
      Serial.print("IP cambiada a: ");
      Serial.println(localIp.toString());
      Serial.print("gateway cambiada a: ");
      Serial.println(gateway.toString());
      Serial.print("mask cambiada a: ");
      Serial.println(mask.toString());
      WiFi.config(localIp,gateway,mask);
      //Se redirecciona a la nueva ip
      request->redirect("http://"+WiFi.localIP().toString()+":80");
  });
}

void loop() {
}
```

### Debug Messages:
```
--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
x �� x�x�x��ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
IP config:0.0.0.0
Preparandose para empezar
IP cambiada a: 192.168.0.50
gateway cambiada a: 192.168.0.1
mask cambiada a: 255.255.255.0

assertion "next > (intptr_t)block" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c", line 124, function: get_next_block
abort() was called at PC 0x401087df on core 1

Backtrace: 0x4008c434:0x3ffd00f0 0x4008c665:0x3ffd0110 0x401087df:0x3ffd0130 0x4008bc2f:0x3ffd0160 0x4008c0b2:0x3ffd0180 0x40084b12:0x3ffd01a0 0x40084f19:0x3ffd01c0 0x4000bec7:0x3ffd01e0 0x400d77ff:0x3ffd0200 0x400d780d:0x3ffd0220 0x400d21e4:0x3ffd0240 0x400d4117:0x3ffd0260 0x400d49ad:0x3ffd0280 0x400d6c09:0x3ffd02a0 0x400d2357:0x3ffd02c0 0x400d2365:0x3ffd02e0 0x4013b5a2:0x3ffd0300 0x4013b781:0x3ffd0330 0x4013b9db:0x3ffd0360 0x40088b7d:0x3ffd0390
  #0  0x4008c434:0x3ffd00f0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008c665:0x3ffd0110 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x401087df:0x3ffd0130 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x4008bc2f:0x3ffd0160 in get_next_block at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:380
      (inlined by) multi_heap_free_impl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap.c:496
  #4  0x4008c0b2:0x3ffd0180 in multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:315
  #5  0x40084b12:0x3ffd01a0 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:131
  #6  0x40084f19:0x3ffd01c0 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #7  0x4000bec7:0x3ffd01e0 in ?? ??:0
  #8  0x400d77ff:0x3ffd0200 in String::invalidate() at C:\Users\neko_\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #9  0x400d780d:0x3ffd0220 in String::~String() at C:\Users\neko_\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #10 0x400d21e4:0x3ffd0240 in AsyncWebHeader::~AsyncWebHeader() at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/WebRequest.cpp:797
      (inlined by) operator() at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/WebRequest.cpp:56
      (inlined by) _M_invoke at c:\users\neko_\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #11 0x400d4117:0x3ffd0260 in std::function<void (AsyncWebHeader* const&)>::operator()(AsyncWebHeader* const&) const at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/WebRequest.cpp:797
  #12 0x400d49ad:0x3ffd0280 in LinkedList<AsyncWebHeader*, LinkedListNode>::free() at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/WebRequest.cpp:797
      (inlined by) AsyncWebServerRequest::~AsyncWebServerRequest() at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/WebRequest.cpp:81
  #13 0x400d6c09:0x3ffd02a0 in AsyncWebServer::_handleDisconnect(AsyncWebServerRequest*) at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/StringArray.h:73
  #14 0x400d2357:0x3ffd02c0 in AsyncWebServerRequest::_onDisconnect() at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/WebRequest.cpp:797
  #15 0x400d2365:0x3ffd02e0 in std::_Function_handler<void (void*, AsyncClient*), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at .pio\libdeps\esp32-evb\ESPAsyncWebServer-esphome\src/WebRequest.cpp:797
      (inlined by) _M_invoke at c:\users\neko_\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #16 0x4013b5a2:0x3ffd0300 in std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at .pio\libdeps\esp32-evb\AsyncTCP-esphome\src/AsyncTCP.cpp:1040
  #17 0x4013b781:0x3ffd0330 in AsyncClient::_error(signed char) at .pio\libdeps\esp32-evb\AsyncTCP-esphome\src/AsyncTCP.cpp:1040
  #18 0x4013b9db:0x3ffd0360 in AsyncClient::_s_error(void*, signed char) at .pio\libdeps\esp32-evb\AsyncTCP-esphome\src/AsyncTCP.cpp:1040
      (inlined by) _handle_async_event at .pio\libdeps\esp32-evb\AsyncTCP-esphome\src/AsyncTCP.cpp:171
      (inlined by) _async_service_task at .pio\libdeps\esp32-evb\AsyncTCP-esphome\src/AsyncTCP.cpp:194
  #19 0x40088b7d:0x3ffd0390 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

lbernstone, 2020-09-29:
Your code is intended to change the IP address when a specific URL is requested?  You cannot tear the stack down while it is active.  Set a flag, and change the IP address in the main loop when the flag is active.

kordhrufan, 2020-09-29:
I see, now it works! Thank you.

closed 2020-09-29 by kordhrufan