espressif/arduino-esp32 #4373: Not able to run Classic BT on esp32

prakharnigam-hsc, 2020-09-30:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.0
IDE name:						Platform.io
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10 using WSL

### Description:
I am trying to run esp as classic BT and I am using example given below but facing runtime issue
https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/classic_bt/bt_discovery/main/bt_discovery.c

But while running I am facing error such as 
esp_bt_controller_init: rc=259 ESP_ERR_INVALID_STATE

When i try to run BLE it is working fine but in Classic BT seems like something not right.

### Debug Messages:
```
esp_bt_controller_init: rc=259 ESP_ERR_INVALID_STATE
```

prakharnigam-hsc, 2020-09-30:
sdkconfig which i am using
[sdkconfig.txt](https://github.com/espressif/arduino-esp32/files/5303517/sdkconfig.txt)

HSC-Project, 2020-10-01:
Hi all, can somebody help to look at this issue, I am also block due to this issue.

TradieUte, 2020-11-04:
I have what may be the same issue. menuconfig set for BT classic only, but running code fails with "BT enable failed" (Have logging turned on.) Traced to questionable #ifdef logic in esp32-hal-bt.c.

//#ifdef CONFIG_CLASSIC_BT_ENABLED
//#define BT_MODE ESP_BT_MODE_BTDM
//#else
//#define BT_MODE ESP_BT_MODE_BLE
//#endif

The above sets BT_MODE to 3, whereas the value loaded to &cfg is 2.

#if defined(CONFIG_CLASSIC_BT_ENABLED)
#define BT_MODE ESP_BT_MODE_CLASSIC_BT
#elif defined(CONFIG_BT_BLE_ENABLED)
#define BT_MODE ESP_BT_MODE_BLE
#else
#define BT_MODE ESP_BT_MODE_BTDM
#endif

replaced as above, and now works fine (however still can't get pin code pairing). Do not know for sure that modified logic is correct for other than BT Classic.

TradieUte, 2021-01-03:
Have got it going, have a look at code on https://github.com/TradieUte/ESP32-WiFi-BT-Server-with-legacy-pairing

closed 2021-01-17 by stale[bot]

ketan, 2021-04-07:
@TradieUte — I can see some change you've made compared to master, it's unclear what version of the arduino framework you were on, when you got this work. Could please share your arduino framework version a the time?

Thanks