espressif/arduino-esp32 #4375: ESP32-S2 I2C interface with MLX90614

vigneshtg, 2020-09-30:
Hello,
I'm using ESP32-S2 (ESP32-S2-Saola-1 Board, having ESP32-S2 Wroover model). I'm trying to connect ESP32-S2 with MLX90614 via I2C interface and have connected SDA to PIN8 and SCL to PIN9 of ESP32-S2. But the problem is nothing is displaying on Serial Monitor. Also, I've checked with I2C Scanner code and it's fine and showing I2C address that I setup, but the temperature is not showing on Serial Monitor and it is blank.

May I know the reason and please help me out.

Thank you.

chegewara, 2020-09-30:
Maybe some code to show?

vigneshtg, 2020-09-30:
I'm using the following library from Sparkfun for MLX90614
https://github.com/sparkfun/SparkFun_MLX90614_Arduino_Library

It also includes example code (MLX90614_Serial_Demo)

chegewara, 2020-09-30:
Are you having issue with example or your own code?

vigneshtg, 2020-09-30:
I'm using example code only and having issue with example code. It is successfully compiled and uploaded but nothing is showing on Serial Monitor.

chegewara, 2020-09-30:
Try to comment out this line:
https://github.com/sparkfun/SparkFun_MLX90614_Arduino_Library/blob/master/src/SparkFunMLX90614.cpp#L35
or call `Wire.begin(SDA, SCL)` after `therm.begin(); `.
I think that `therm.begin(); `  is reinit wire with default pins.

vigneshtg, 2020-09-30:
Let me try it and update you. Thanks!

vigneshtg, 2020-10-01:
I've checked by commenting this line:
https://github.com/sparkfun/SparkFun_MLX90614_Arduino_Library/blob/master/src/SparkFunMLX90614.cpp#L35
Also, called Wire.begin(SDA, SCL) after therm.begin();  
But nothing is dsiplayed on Serial Monitor. Further, I've checked other I2C based sensor with ESP32-S2, there also nothing is displayed, but everything working fine with ESP32 but not with ESP32-S2. I think the problem is with I2C driver of ESP32-S2
Please help me out!
Thank you

chegewara, 2020-10-01:
Time to use logic analyzer?

chegewara, 2020-10-01:
https://github.com/espressif/arduino-esp32/blob/esp32s2/variants/esp32s2/pins_arduino.h#L17-L18

vigneshtg, 2020-10-01:
> https://github.com/espressif/arduino-esp32/blob/esp32s2/variants/esp32s2/pins_arduino.h#L17-L18

Yes, when I've started the connection, I followed this only, but still I don't understand why nothing is displayed on Serial Monitor and it's blank?

chegewara, 2020-10-01:
I dont have that sensor, so no much help about it, but just tested I2C with DS1307 RTC and it is working just fine.

vigneshtg, 2020-10-01:
> I dont have that sensor, so no much help about it, but just tested I2C with DS1307 RTC and it is working just fine.

Have you interfaced DS1307 RTC with ESP32-S2 or ESP32? If you've done with ESP32-S2, could you please share the code so that it'll be helpful to check it. Also I can check with  DS1307 RTC from my side and try to fix it with MLX90614

chegewara, 2020-10-01:
> Have you interfaced DS1307 RTC with ESP32-S2 or ESP32?

ESP32-S2 of course.

i am using example from RTClib library:
```
// Date and time functions using a DS1307 RTC connected via I2C and Wire lib
#include "RTClib.h"
#include "Wire.h"
RTC_DS1307 rtc;

char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};

void setup () {
  Serial.begin(115200);

#ifndef ESP8266
  while (!Serial); // wait for serial port to connect. Needed for native USB
#endif
// Wire.begin(8,9);
  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    Serial.flush();
    abort();
  }

  if (! rtc.isrunning()) {
    Serial.println("RTC is NOT running, let's set the time!");
    // When time needs to be set on a new device, or after a power loss, the
    // following line sets the RTC to the date & time this sketch was compiled
    rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
    // This line sets the RTC with an explicit date & time, for example to set
    // January 21, 2014 at 3am you would call:
    // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
  }

  // When time needs to be re-set on a previously configured device, the
  // following line sets the RTC to the date & time this sketch was compiled
  // rtc.adjust(DateTime(F(__DATE__), F(__TIME__)));
  // This line sets the RTC with an explicit date & time, for example to set
  // January 21, 2014 at 3am you would call:
  // rtc.adjust(DateTime(2014, 1, 21, 3, 0, 0));
}

void loop () {
    DateTime now = rtc.now();

    Serial.print(now.year(), DEC);
    Serial.print('/');
    Serial.print(now.month(), DEC);
    Serial.print('/');
    Serial.print(now.day(), DEC);
    Serial.print(" (");
    Serial.print(daysOfTheWeek[now.dayOfTheWeek()]);
    Serial.print(") ");
    Serial.print(now.hour(), DEC);
    Serial.print(':');
    Serial.print(now.minute(), DEC);
    Serial.print(':');
    Serial.print(now.second(), DEC);
    Serial.println();

    Serial.print(" since midnight 1/1/1970 = ");
    Serial.print(now.unixtime());
    Serial.print("s = ");
    Serial.print(now.unixtime() / 86400L);
    Serial.println("d");

    // calculate a date which is 7 days, 12 hours, 30 minutes, and 6 seconds into the future
    DateTime future (now + TimeSpan(7,12,30,6));

    Serial.print(" now + 7d + 12h + 30m + 6s: ");
    Serial.print(future.year(), DEC);
    Serial.print('/');
    Serial.print(future.month(), DEC);
    Serial.print('/');
    Serial.print(future.day(), DEC);
    Serial.print(' ');
    Serial.print(future.hour(), DEC);
    Serial.print(':');
    Serial.print(future.minute(), DEC);
    Serial.print(':');
    Serial.print(future.second(), DEC);
    Serial.println();

    Serial.println();
    delay(3000);
}
```

vigneshtg, 2020-10-01:
> ESP32-S2 of course.
>
> i am using example from RTClib library:
>
> ```
> // Date and time functions using a DS1307 RTC connected via I2C and Wire lib
> #include "RTClib.h"
> #include "Wire.h"
> RTC_DS1307 rtc; [...]

Thanks. In the code you've commented Wire.begin(8,9). Is it like that? Also, I think may be problem in installing EP32-S2 board on Arduino platform. May I know the right procedure to follow it? Further, after adding ESP32-S2 board, when I choose ESP32 Dev board, I'm getting a compilation error like this:

**xtensa-esp32-elf-g++: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.**

But when I choose ESP32-S2 board it's compiling fine and I've read that when using ESP32-S2, ESP32 based boards can't be used. Is it so?

chegewara, 2020-10-01:
At first you have to delete old arduino-esp32, make sure you did delete all. I believe you have this error because some old files, maybe toolchain, are still not deleted. Then follow instruction for development repository, just use esp32s2 branch instead of master.

There is also video on youtube from @UnexpectedMaker 
https://www.youtube.com/watch?v=yiYyTQDRFZI

This installation will work with old esp32 and esp32-s2, when you pick proper board. It is still dev branch, so expect bugs and issues and please report if you find any.

vigneshtg, 2020-10-01:
> At first you have to delete old arduino-esp32, make sure you did delete all. I believe you have this error because some old files, maybe toolchain, are still not deleted. Then follow instruction for [...]

Sure, Thanks. I've seen this video earlier and he's doing for Mac and not windows. Hence I've followed this video:
https://www.youtube.com/watch?v=L6IoSVdKwNM

Is the procedure on this video is right to follow it?  Also you made installation for Windows or other operating system?

chegewara, 2020-10-01:
Currently i am working on ubuntu 20.04, but i also have prepared windows 10 installation and i did have windows 10 in virtualbox.

vigneshtg, 2020-10-01:
Have you took the video that you've shared as a reference for windows installation or some other procedure?

> Currently i am working on ubuntu 20.04, but i also have prepared windows 10 installation and i did have windows 10 in virtualbox.

chegewara, 2020-10-01:
My own experience and git command line.

vigneshtg, 2020-10-01:
> My own experience and git command line.

Fine, Thanks. If possible you can share your procedure for windows installation which will be very helpful.  Also, I can try it out and let you know the updates.

vigneshtg, 2020-10-09:
Hello,
I've checked DS3231 RTC Timer with ESP32-S2. It's working fine. But except DS3231, other I2C based devices are not working with ESP32-S2. The program is successfully uploaded but Serial Monitor is blank and showing nothing. Please help me out.
Thank you.

closed 2020-12-25 by stale[bot]

SuGlider, 2021-09-14:
Arduino Core 2.0.0 and PR #5664 may solve this issue.
Please test it and let me know.
Thanks.