espressif/arduino-esp32 #4376: OTA update : Guru meditation Error

naturalmechanics, 2020-09-30:
I am trying to download a bin file via tinyGSM library, and use it as the new firmware (OTA over LTE). The file is exported via arduino 1.8.11 as compressed binary file.

System : [ESP 32 on TTGO / lilygo sim 7000g](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G), Arduino on arch linux 64 bit.

 I have followed [this](https://github.com/espressif/arduino-esp32/issues/3976), because I was having the invalid chip ID error, as well as invalid magic byte error. I have turned off the magic byte check, and edited the files Esp.cpp, Esp.h, Updater.cpp, and Update.h as described in the [PR](https://github.com/espressif/arduino-esp32/pull/3898).

However, now I have this error: 

```
Inteding to start update
21:35:34.306 -> [D][Updater.cpp:130] begin(): OTA Partition: app1
21:35:34.505 -> Escritos : 17995 successfully
21:35:34.505 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
21:35:34.505 -> Core 1 register dump:
21:35:34.505 -> PC      : 0x4000c2e0  PS      : 0x00060930  A0      : 0x80087b10  A1      : 0x3ffb1bf0  
21:35:34.505 -> A2      : 0x3ffb1c10  A3      : 0x00000000  A4      : 0x00000010  A5      : 0x3ffb1c10  
21:35:34.538 -> A6      : 0x3ffb0019  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00150000  
21:35:34.538 -> A10     : 0x3ffbd49c  A11     : 0x0000464b  A12     : 0x3ffc1a00  A13     : 0x3ffc1d38  
21:35:34.538 -> A14     : 0x3ffc1ee4  A15     : 0x00000000  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c  
21:35:34.571 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
21:35:34.571 -> 
21:35:34.571 -> Backtrace: 0x4000c2e0:0x3ffb1bf0 0x40087b0d:0x3ffb1c00 0x4013b339:0x3ffb1c50 0x400db99e:0x3ffb1c70 0x400d94b2:0x3ffb1c90 0x400d965e:0x3ffb1cb0 0x400d9809:0x3ffb1cd0 0x400d1761:0x3ffb1d10 0x400d3925:0x3ffb1d60 0x400d6639:0x3ffb1da0 0x400d6843:0x3ffb1e60 0x400d6d86:0x3ffb1f20 0x400d6db6:0x3ffb1f40 0x400d6dff:0x3ffb1f60 0x400d6e87:0x3ffb1f80 0x400dccab:0x3ffb1fb0 0x400896f9:0x3ffb1fd0
21:35:34.604 -> 
21:35:34.604 -> Rebooting...
21:35:34.604 -> ets Jun  8 2016 00:22:57

```

Using the backtrace decoder, i find : 
```
0x40087b0d: spi_flash_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 393
0x4013b339: esp_partition_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/partition.c line 274
0x400db99e: EspClass::partitionWrite(esp_partition_t const*, unsigned int, unsigned int*, unsigned int) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Esp.cpp line 203
0x400d94b2: UpdateClass::_enablePartition(esp_partition_t const*) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Update/src/Updater.cpp line 167
0x400d965e: UpdateClass::_verifyEnd() at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Update/src/Updater.cpp line 167
0x400d9809: UpdateClass::end(bool) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Update/src/Updater.cpp line 167
0x400d1761: performUpdate(Stream&, unsigned int) at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 2004 (discriminator 14)
0x400d3925: updateFromFS() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1977
0x400d6639: check_OTA() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1799
0x400d6843: reset_GSM() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 516
0x400d6d86: reactivate_cycle() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1953
0x400d6db6: activate_wakeupProtocol(int) at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1929
0x400d6dff: differentiate_Wakeup() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1912
0x400d6e87: setup() at /home/hedgehog/Work/GPS_Moving/GPS_ITER_12_OTA/GPS_ITER_12_OTA.ino line 1653
0x400dccab: loopTask(void*) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 14
0x400896f9: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 355 (discriminator 1)
```

Going to ` EspClass::partitionWrite(esp_partition_t const*, unsigned int, unsigned int*, unsigned int) at /home/hedgehog/.arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/Esp.cpp line 203`, I find that the code is : 

```
  md5.calculate();
    result = md5.toString();
    return result;
}

```
It does not call  partitionWrite. PartitionWrite is in line 317. 

I am trying to download the OTA file via GSM modem. The relevant code is here: 

 ```
uint32_t contentLength = knownFileSize;
		 
		 File file = SPIFFS.open("/update.bin", FILE_APPEND);
		 
		 while (client.available())
		 {
			 String line = client.readStringUntil('\n');
			 line.trim();
			 line.toLowerCase();
			 if (line.startsWith("content-length:"))
			 {
				 contentLength = line.substring(line.lastIndexOf(':') + 1).toInt();
			 }
			 else if (line.length() == 0)
			 {
				 break;
			 }
		 }
		 
		 timeout = millis();
		 uint32_t readLength = 0;
		 
		 unsigned long timeElapsed = millis();
		 
		 while (readLength < contentLength && client.connected() && millis() - timeout < 10000L){
			 int i = 0;
			 while (client.available())
			 {
				 
				 if (!file.print(char(client.read()))) {																	Serial.println("Fallo Append");
				 }
				 
				 readLength++;
				 
				 timeout = millis();
			 }
		 }
		 
		 file.close();
		 
		 
		 timeElapsed = millis() - timeElapsed;
		 Serial.println();
		 
		 client.stop();
		 
		 
		 updateFromFS();

 void updateFromFS(){
	 File updateBin = SPIFFS.open("/update.bin");
	 if (updateBin)
	 {
		 if (updateBin.isDirectory())
		 {
			 Serial.println("Error, en el directorio");
			 updateBin.close();
			 return;
		 }
		 
		 size_t updateSize = updateBin.size();
		 
		 if (updateSize > 0)		 {																					Serial.println("Inteding to start update");
			 performUpdate(updateBin, updateSize);
		 }
		 else
		 {
			 Serial.println("Error, archivo vacío");
		 }
		 
		 updateBin.close();
		 
		 // whe finished remove the binary from sd card to indicate end of the process
		 //fs.remove("/update.bin");
	 }
	 else
	 {
		 Serial.println("No se puede cargar el archivo");
	 }
 }
 
 
void performUpdate(Stream &updateSource, size_t updateSize)
{
	if (Update.begin(updateSize))
	{
		 size_t written = Update.writeStream(updateSource);
		 if (written == updateSize) {																					Serial.println("Escritos : " + String(written) + " successfully");
		 } else {																										Serial.println("wrote only ...  : " + String(written) + "/" + String(updateSize) + ". Retry?");
		 }
		 if (Update.end()) {																							Serial.println("OTA finish!");
			 if (Update.isFinished())			 {																		Serial.println("Ota exited ... rebotting!");
				 ESP.restart();
			 }
			 else
			 {
				 Serial.println("Ota no terminó? Algo salió mal!");
			 }
		 }
		 else
		 {
			 Serial.println("Ocurrió Error #: " + String(Update.getError()));
		 }
	 }
	 else
	 {
		 Serial.println("Sin espacio suficiente para hacer OTA");
	 }
 }
```

sorry for the tab formatting, It works perfectly on kate.

The target file is a *.ino file, exported as compressed binary via arduino 1.8.11 IDE. I wonder what I wrong. Thank you.

### Hardware:
Board: TTGO SIM 7000G
IDE name:							Arduino 1.8.11
Flash Frequency:					80 MHz
PSRAM enabled:                                          NO
Upload Speed:						921600
Computer OS:                                           Arch 64 bit

robertpoll, 2020-10-01:
Are you using encrypted flash? You would need to have enabled it specifically using something like `espsecure.py` - it's not possible from the Arduino IDE afaik so it sounds like you might not be. 

Assuming you're not, then take a look at the binary you've created from Arduino - the first byte should be `0xE9`. Also I would check it once the file is in your SPIFFS filesystem, just before you run `performUpdate()` to check it is still OK.

When you say you've turned off the magic byte check, are you using the files in the PR as-is or have you modified them?

naturalmechanics, 2020-10-01:
i turned off the magic byte check in updater.cpp and also modified them by following the link in the PR, following the [diff display](https://github.com/espressif/arduino-esp32/pull/3898/commits/4d6d1ccde092ac84d79f1fb87292e7889ff9a333). No, I am not using encrypted flash.  

viewing qwith oktete, i have the first byte as 0xe9, but viewing with hexdump it is 0x07, followed by 0xE9. Is there a byte order conflict of some sort?

robertpoll, 2020-10-01:
If you're not using flash encryption then you don't need the modified `Updater.cpp` and turning off the magic byte checking won't help. The invalid magic byte error looks like it was because the firmware didn't come across properly. I'd suggest dumping out the first few bytes in your firmware. Alternatively you could switch the bytes in the bin file on your Linux box as a quick check to see if that works. Then you'll need to work out where the switch is happening.

naturalmechanics, 2020-10-01:
OK. I will follow that.

I have tried[ this one ](https://github.com/platformio/bintray-secure-ota/blob/d0899cd98403d9b2315f367b0b670190d4c6325c/src/SecureOTA.cpp)as well. The client is the GSM modem client. Not wifi. The rest is same. I am coming up to the part : 

```
(Update.begin(contentLength))	
...
size_t written = Update.writeStream(client);

```

After that is a call to : (Update.end()) - and this is where I am again getting the same error as mentioned above.

Eiyvor, 2020-11-09:
Try increasing the RX buffer, call this macro #define TINY_GSM_RX_BUFFER

Eiyvor, 2020-11-09:
https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/tree/master/examples/Arduino_GSM_OTA try this for OTA update

closed 2021-01-24 by stale[bot]