espressif/arduino-esp32 #4379: ESP32-S2 I2C Slave Mode

nedimkivanc, 2020-10-01:
I noticed that esp32 is not in the library for i2c slave mode, is there a work for it?

vigneshtg, 2020-10-01:
I'm also facing the problem of I2C with ESP32S2, when interfaced with sensor, nothing is displayed on Serial Monitor. Please help out.

nedimkivanc, 2020-10-01:
> I'm also facing the problem of I2C with ESP32S2, when interfaced with sensor, nothing is displayed on Serial Monitor. Please help out.

This works well if you are using the i2c as a master

vigneshtg, 2020-10-01:
> > I'm also facing the problem of I2C with ESP32S2, when interfaced with sensor, nothing is displayed on Serial Monitor. Please help out.
> 
> This works well if you are using the i2c as a master

I'm trying to connect ESP32-S2(ESP32-S2-Saola-1 Board) with MLX90614 via I2C interface and have connected SDA to PIN8 and SCL to PIN9 of ESP32-S2. But the problem is nothing is displaying on Serial Monitor. Now, how do I use I2C Master for interfacing ESP32-S2 with MLX90614?

nedimkivanc, 2020-10-01:
> I'm trying to connect ESP32-S2(ESP32-S2-Saola-1 Board) with MLX90614 via I2C interface and have connected SDA to PIN8 and SCL to PIN9 of ESP32-S2. But the problem is nothing is displaying on Serial [...]

VIN -> 3.3v
GND -> GND
SCL -> 22
SDA -> 21

You can make pin connections this way and it will be useful for you to use Adafruit_MLX90614 library.

atanisoft, 2020-10-01:
@nedimkivanc pins 8 and 9 are the default for the ESP32-S2. Pin 22 is not usable on the ESP32-S2 btw.

@vigneshtg I'd suggest keep the MLX90614 discussion in #4375 since you are not using the ESP32-S2 as an I2C Slave device.

nedimkivanc, 2020-10-01:
> @nedimkivanc pins 8 and 9 are the default for the ESP32-S2. Pin 22 is not usable on the ESP32-S2 btw.
> 
> @vigneshtg I'd suggest keep the MLX90614 discussion in #4375 since you are not using the ESP32-S2 as an I2C Slave device.

Thanks for the warning I noticed I typed the model wrong and fixed it

closed 2020-10-02 by nedimkivanc