espressif/arduino-esp32 #4380: Cant connect to specific router

Viriatto, 2020-10-01:
> ### Hardware:
> **Router:**                                                   Hitron CVE30360
> **Board:**							DOIT Devkit V1
> **Core Installation version:**			1.0.4
> **IDE name:**						Arduino IDE
> **Flash Frequency:**					80Mhz
> **PSRAM enabled:**                                     (Not sure)
> **Upload Speed:**					921600
> **Computer OS:**                                         Windows 10

### Description:
When in station mode, can connect normally to mobile hotspot without any errors or problems, but connecting to home router displays a ton of AUTH_EXPIRE, ASSOC_EXPIRE, NO_AP_FOUND and 4WAY_HANDSHAKE_TIMEOUT errors. Sometimes after dozens of these it ends up connecting for a couple of minutes/seconds before disconnecting again.

### What I have tried:

- Adding `Wifi.disconnect()` at the beggining of `setup()` - [Comment](https://github.com/espressif/arduino-esp32/issues/2144#issuecomment-450394742)
- Tried different firmware versions (10.4, 1.03 and 1.02)
- Changed router mode from 802.11 g/n to 802.11 b/g/n
- Rebooted router 
- Added these lines of code on setup as suggested in [this comment](https://github.com/espressif/arduino-esp32/issues/2144#issuecomment-545303384)
```
ESP_ERROR_CHECK(nvs_flash_erase());
nvs_flash_init();
ESP.restart();
``` 

### I have "trimmed" some of the code for easier readibility, if you think I should post the full code let me know
```cpp

#include <WiFi.h>
#include "nvs_flash.h"

static void WiFiEvent(WiFiEvent_t event, WiFiEventInfo_t info)
{
  if (event == SYSTEM_EVENT_STA_DISCONNECTED) {
     WiFi.reconnect();
  }
}

void connectWifi() {
  const char* WIFI_SSID = "SSID";
  const char* WIFI_PASSWORD = "PASSWORD";

  IPAddress GATEWAY(192, 168, 1, 1);  
  IPAddress STATIC_IP(192, 168, 1, 80);  
  IPAddress SUBNET(255, 255, 255, 0); 
  
  const char* CONSOLE_CONNECTED = "CONNECTED TO ";
  const char* CONSOLE_IP_ADDRESS = "IP ADDRESS: ";

  WiFi.config(STATIC_IP, GATEWAY, SUBNET);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  Serial.println("");

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  
  Serial.println("");
  Serial.print(CONSOLE_CONNECTED);
  Serial.println(WIFI_SSID);
  Serial.print(CONSOLE_IP_ADDRESS);
  Serial.println(WiFi.localIP());
}

/* setup function */
void setup(void) {
  Serial.begin(115200);
  Serial.println();

  ESP_ERROR_CHECK(nvs_flash_erase());
  nvs_flash_init();
  
  connectWifi();
  WiFi.onEvent(WiFiEvent);
}

void loop(void) {}
```

### Debug Messages when connecting to home router:
```
...[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
......[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
.[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
.
CONNECTED TO NOS4920
IP ADDRESS: 192.168.1.80
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 4 - ASSOC_EXPIRE
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND

```
### Debug Messages when connecting to mobile hotspot:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
.
CONNECTED TO HUAWEI
IP ADDRESS: 192.168.1.80

```

Thank you very much for your help

atanisoft, 2020-10-01:
@Viriatto Can you add this to your setup() function:
```
esp_log_level_set("wifi", ESP_LOG_VERBOSE);
esp_wifi_internal_set_log_level(WIFI_LOG_VERBOSE);
esp_wifi_internal_set_log_mod(WIFI_LOG_MODULE_ALL, WIFI_LOG_SUBMODULE_ALL, true);
```

You will likely need to add this include as well:
```
#include <esp_log.h>
```

Viriatto, 2020-10-01:
@atanisoft I am getting the following compiler error: `'WIFI_LOG_VERBOSE' was not declared in this scope`

atanisoft, 2020-10-01:
Sorry you will also need to include:
```
#include <esp_wifi.h>
```

Viriatto, 2020-10-01:
Don't be sorry I appreciate your help. I have those 3 included but I'm still getting the same compiler error

atanisoft, 2020-10-01:
Ok, stick with only `esp_log_level_set("wifi", ESP_LOG_VERBOSE);` and capture the diagnostic output. That may show why you are having connection issues. This will very likely need to be routed over to https://github.com/espressif/esp-idf/issues as well.

Viriatto, 2020-10-01:
Here is the output:

```
[D][esp32-hal-cpu.c:178] setCpuFrequencyMhz(): PLL: 320 / 4 = 80 Mhz, APB: 80000000 Hz
I (225) wifi: wifi driver task: 3ffce7a0, prio:23, stack:3584, core=0
I (1375) wifi: wifi firmware version: 7997e4b
I (1375) wifi: config NVS flash: enabled
I (1375) wifi: config nano formating: disabled
I (1376) wifi: Init dynamic tx buffer num: 32
I (1380) wifi: Init data frame dynamic rx buffer num: 32
I (1385) wifi: Init management frame dynamic rx buffer num: 32
I (1391) wifi: Init management short buffer num: 32
I (1396) wifi: Init static rx buffer size: 1600
I (1400) wifi: Init static rx buffer num: 16
I (1403) wifi: Init dynamic rx buffer num: 32
I (1675) wifi: mode : softAP (24:62:ab:ff:39:79)
I (1677) wifi: Total power save buffer number: 16
I (1677) wifi: Init max length of beacon: 752/752
I (1678) wifi: Init max length of beacon: 752/752
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
I (1695) wifi: mode : sta (24:62:ab:ff:39:78)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
..I (2851) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1
....I (5126) wifi: state: init -> auth (b0)
..I (6127) wifi: state: auth -> init (200)
I (6128) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
....[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
.I (8380) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (8381) wifi: state: init -> auth (b0)
..I (9382) wifi: state: auth -> init (200)
I (9383) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
I (9514) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (9515) wifi: state: init -> auth (b0)
I (9536) wifi: state: auth -> assoc (0)
..I (10537) wifi: state: assoc -> init (400)
I (10538) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 4 - ASSOC_EXPIRE

```

As for the issue being routed over to the different repository, do I have to do something on my end for it to happen?

Thank you

atanisoft, 2020-10-01:
Can you try adding `WiFi.setSleep(false);` in setup() before calling `WiFi.begin(...);`?

Viriatto, 2020-10-01:
It seems like it is connecting with much ease now, not getting the AUTH_EXPIRE spam, but it is still often disconnecting after a couple of seconds/minutes:

```
.I (2168) wifi: state: auth -> init (200)
I (2169) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
.I (2311) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
.I (3094) wifi: state: init -> auth (b0)
I (3101) wifi: state: auth -> assoc (0)
I (3131) wifi: state: assoc -> run (10)
...I (4437) wifi: connected with NOS4920, channel 9, BW20
I (4441) wifi: pm start, type: 0

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
.
CONNECTED TO NOS4920
IP ADDRESS: 192.168.1.80
I (7383) wifi: state: run -> init (fc0)
I (7383) wifi: pm stop, total sleep time: 0 us / 2941615 us

I (7383) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
I (7520) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (7520) wifi: state: init -> auth (b0)
I (7525) wifi: state: auth -> assoc (0)
I (7563) wifi: state: assoc -> run (10)
I (10888) wifi: connected with NOS4920, channel 9, BW20
I (10892) wifi: pm start, type: 0

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
I (13831) wifi: state: run -> init (fc0)
I (13832) wifi: pm stop, total sleep time: 0 us / 2939340 us

I (13832) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
I (13969) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (13970) wifi: state: init -> auth (b0)
I (14970) wifi: state: auth -> init (200)
I (14971) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
I (15114) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (15114) wifi: state: init -> auth (b0)
I (15127) wifi: state: auth -> assoc (0)
I (15152) wifi: state: assoc -> run (10)
I (15456) wifi: connected with NOS4920, channel 9, BW20
I (15510) wifi: pm start, type: 0

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
I (18420) wifi: state: run -> init (fc0)
I (18421) wifi: pm stop, total sleep time: 0 us / 2911000 us

I (18421) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
I (18558) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (18558) wifi: state: init -> auth (b0)
I (18568) wifi: state: auth -> assoc (0)
I (18617) wifi: state: assoc -> run (10)
I (19285) wifi: connected with NOS4920, channel 9, BW20
I (19290) wifi: pm start, type: 0

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
I (22229) wifi: state: run -> init (fc0)
I (22229) wifi: pm stop, total sleep time: 0 us / 2939624 us

I (22230) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
I (22367) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (22367) wifi: state: init -> auth (b0)
I (22375) wifi: state: auth -> assoc (0)
I (22404) wifi: state: assoc -> run (10)
I (22963) wifi: connected with NOS4920, channel 9, BW20
I (22968) wifi: pm start, type: 0

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
I (25915) wifi: state: run -> init (fc0)
I (25916) wifi: pm stop, total sleep time: 0 us / 2948046 us

I (25916) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT
I (26053) wifi: n:9 0, o:9 0, ap:255 255, sta:9 0, prof:1
I (26053) wifi: state: init -> auth (b0)
I (26076) wifi: state: auth -> assoc (0)
I (26115) wifi: state: assoc -> run (10)
I (26494) wifi: connected with NOS4920, channel 9, BW20
I (26498) wifi: pm start, type: 0

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.80, MASK: 255.255.255.0, GW: 192.168.1.1

```

atanisoft, 2020-10-01:
That could be the AP disconnecting the client and not related to the ESP32 itself.

Viriatto, 2020-10-01:
Yes, because it connects just fine to mobile hotspot, just gets these errors when I try this specific router, any idea why the router could be disconnecting it?

DougArmstrong, 2020-10-01:
I had a similar issue connecting using an ESP32 to run a high powered Switching Charge Controller.  Scope leads near the CPU were inducing enough EMI to cause the radio to drop connections.  Any thing like this possible in your design? 

I did a little googling and found a couple articles where folks were chasing this issue, might not be an ESP32 thing.  Might read a few of these and see if there is anything that looks useful.  Hope this helps. https://forum.mikrotik.com/viewtopic.php?p=541785

Viriatto, 2020-10-01:
In this case the mobile hotspot actually works better than my router. Right now it is definitely working better, it fails to connect a couple of times but then connects and holds connection for some good time. Other times I boot it and it just gets stuck in an endless loop.

@DougArmstrong All Im using my board for is to control a small servo, but I did see someone in that link that fixed his problem by changing router password to all lowercase, I will try that later, thank you!

Viriatto, 2020-10-01:
Tried changing password to lowercase and unfortunately it didn't help

TD-er, 2020-12-01:
Maybe I am missing it in the comments, but what brand access point is causing the issues here?
For example Mikrotik does have a setting to tell the router what "distance" to expect.
Wireless -> WiFi interfaces -> wlan1
Then in Advanced mode: Installation: Any/Indoor/Outdoor

This setting has to do with regulations.
Any and Indoor  are the least strict ones.

There are other settings available which may be useful.
Like:
- Disconnect timeout
- Noise Floor Threshold
- Distance

The Distance (ACK-Timeout) setting may need some explanation:
- Dynamic - causes AP to detect and use the smallest timeout that works with all connected clients.
- Indoor - uses the default ACK timeout value that the hardware chip manufacturer has set.
- Number - uses the input value in formula: ACK-timeout = ((distance * 1000) + 299) / 300 us;

closed 2021-02-13 by stale[bot]

rajveer01, 2022-09-25:
I do not know the cause of the problem, but when I removed all the connections to my espe32 except the USB, It worked for me.