espressif/arduino-esp32 #4383: linker error with std::vector::push_back and static const uint8_t (undefined reference)

siochs, 2020-10-02:
### Description:
I have two static uint8_t elements defined in header file:
```c++
// include/SomeClass.hpp
#include <stdint.h>
#include <vector>

class SomeClass
{
private:
  static const uint8_t HEX_ONE = 0x01;
  static const uint8_t HEX_TWO = 0x02;
public: 
  std::vector<uint8_t> getVector();
};
```
I am trying to push back both elements in a std::vector:
```c++
// src/SomeClass.cpp
#include "SomeClass.hpp"

std::vector<uint8_t> SomeClass::getVector()
{
    std::vector<uint8_t> vector;
    // this does not link
    vector.push_back(HEX_ONE); // linker error: undefined reference to `SomeClass::HEX_ONE'
    vector.push_back(HEX_TWO); // linker error: undefined reference to `SomeClass::HEX_TWO'
    

    // this does link (pushing back only one element)
    /*
    vector.push_back(HEX_ONE); 
    */
   
    // this does link (casting one element to same type)
    /*
    vector.push_back((uint8_t)HEX_ONE);
    vector.push_back(HEX_TWO);
    */

    // this does link (casting both elements to same type)
    /*
    vector.push_back((uint8_t)HEX_ONE);
    vector.push_back((uint8_t)HEX_TWO);
    */
    return vector;
}
```

### Sketch: 
```cpp
#include <Arduino.h>
#include "SomeClass.hpp"

void setup() 
{
  SomeClass someClass;
  auto vector = someClass.getVector();
  for(;;);
}

void loop() {
}
```

### Debug Messages:
The linker errors look the following:
```
.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\src\SomeClass.cpp.o:(.literal._ZN9SomeClass9getVectorEv+0x0): undefined reference to `SomeClass::HEX_ONE'
.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32doit-devkit-v1\src\SomeClass.cpp.o:(.literal._ZN9SomeClass9getVectorEv+0x4): undefined reference to `SomeClass::HEX_TWO'
```
Build messages look the following:
```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-xtensa32 @ 3.80200.200512
Unpacking  [####################################]  100%
Tool Manager: toolchain-xtensa32 @ 3.80200.200512 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 0.0.0+sha.475208e 
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 3.80200.200512 (8.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
Building in release mode
...
```
### Hardware:
Board:							ESP32 Devkit C
Core Installation version:			Comes with PlatformIO Core 5.0.1, Home 3.3.0
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                         Windows 10

### Configuration:
This is a C++17 setup. It compiles and links in Visual Studio 2019. I am convinced, this is not related to PlatformIO but somehow with the Arduino-Framework and the C++-Compiler.
```ini
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
monitor_speed = 115200
platform_packages = 
  ; see https://community.platformio.org/t/possible-to-use-c-17/12272/2  
  toolchain-xtensa32 @ 3.80200.200512
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.0
build_flags =
  -std=c++17
build_unflags =
  -std=gnu++11
lib_deps = ../../src
```

Clickau, 2020-10-02:
This isn't related to the Arduino Framework, the error happens with any c++ compiler. Static variables must be declared in the .cpp file. See here: https://stackoverflow.com/questions/9282354/static-variable-link-error

igrr, 2020-10-02:
Hi @siochs, indeed you have met a difference between GCC and the Visual C++ compiler. GCC behavior here is not specific to the ESP32 or Xtensa architecture, it works the same way if you compile this program (replacing `setup` with `main`) with an x86_64 GCC.

The program indeed seems to be missing the definitions of HEX_ONE and HEX_TWO, if I read the C++17 standard correctly (12.2.3.2 paragraph 3, emphasis mine):

> If a non-volatile non-inline const static data member is of integral or enumeration type, its declaration
in the class definition can specify a brace-or-equal-initializer in which every initializer-clause that is an
assignment-expression is a constant expression (8.6). **The member shall still be defined in a namespace scope
if it is odr-used (6.2) in the program and the namespace scope definition shall not contain an initializer**

I'm not sure why Visual Studio successfully links this code, it might be related to the optimization settings. If you compile the above code with -O2 optimization flag with GCC, compilation will also be successful, because the `vector.push_back` calls will be optimized out (they don't have side effects).

siochs, 2020-10-05:
Thank you guys I got the point. However I had a really odd feeling after I just commented out one line and there was no linker error anymore, but this might be related to optimization settings. I changed the typing to `static constexpr uint8_t HEX_ ...`  in the header and that solved it. 
Issue can be closed.

closed 2020-10-08 by everslick