espressif/arduino-esp32 #4388: "WeMos" D1 R32, Wifi and DS18B20

dtmix, 2020-10-04:
Dear All,

I am stuck with an interesting and frustrating issue and appreciate some guidance/help.

I am trying to get DS18B20 sensor working **with** Wifi. 

At present, to just get DS18B20 readings  _without_ WiFI, is straightforward - just use the OneWire.h and DallasTemperature.h libraries and connect the data cable to Pin 4. VCC is to 5V and Gnd is Gnd. 

However, with these libraries AND using Pin 4, I just cannot get Wifi to start up on the "WeMos" D1 R32 dev board.

On the other hand, I cannot get DS18B20 readings if I try to use other pins (such as Pin 36). 
The libraries just don't detect the device on other Pins except Pin 4 (i tried, 36, 35, 39 and even 27). 

Am I missing something? The board is the one that is quite easily available, it looks like this:
https://www.hackster.io/uncle-yong/wemos-r32-with-arduino-startup-guide-7bc841

Thanks and warm regards to all!

lbernstone, 2020-10-04:
Sounds like a power issue.  Try a different USB port or get a 3V3 power supply that you can connect to Vin & bypass the LDO.

dtmix, 2020-10-05:
Hi Ibernstone,

Thanks!! Okay I'll use a breadboard with independent power and keep everyone posted. 

Will have to be when I get off work later tonight though ..  :-) 

thanks for the suggestion!

dtmix, 2020-10-05:
THANK YOU Ibernstone, Yes, you are right!! :-) 

I am pleased to update that by using a breadboard power supply, I am able to use both Pins 4 and Pins 27 to get DS18B20 readings AND connect to WiFi. We needed to power up the DS18B20 independently. 

The case is closed. again thanks!

closed 2020-10-05 by dtmix

lbernstone, 2020-10-05:
Just FYI, pins 34-39 are input only, so will not work for a bidirectional sensor like DS18B20.

dtmix, 2020-10-06:
Again thanks! Much appreciated and have a great day!