espressif/arduino-esp32 #4391: Using IDF components? (Ableton Link)

DatanoiseTV, 2020-10-05:
I'd like to use the IDF component for Ableton Link support in a Arduino ESP32 project.
Unfortunately when I install the lib using pio lib install, it doesn't set the right include paths and source paths.
What's the right way to do it?

lbernstone, 2020-10-05:
See if adding `#include <errno.h>` after this line helps it find those definitions.
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/asio/esp_exception.h#L23

DatanoiseTV, 2020-10-06:
I will try. Now I am first trying to build it on plain IDF 4.1, but it's throwing errors that it cannot find boost headers or include `<memory>`. Retrying with IDF 4.0

@lbernstone Do you have a suggestion / could you help packaging it for Arduino?

DatanoiseTV, 2020-10-06:
@lbernstone Same with 4.0
```
/Users/DatanoiseTV/dev/idf-4.0.0/esp-idf/components/asio/asio/asio/include/asio/associated_allocator.hpp:19:10: fatal error: memory: No such file or directory
 #include <memory>
          ^~~~~~~~
```

closed 2020-12-19 by stale[bot]