espressif/arduino-esp32 #4394: BLE set a timeout for client.connect()

nathan-casabieille, 2020-10-07:
Hello, 

I am currently programming an esp32 to retrieve data from all surrounding bluetooth devices.

First, it does a scan, then it create a linked list and connect to all the detected devices one by one, retrieving each time a data.

Everything works perfectly for tens of minutes, but randomly, at one point the program always hangs while I execute the command `pClient->connect(pAddress);`  while trying to connect to one of the detected devices.

Knowing that this is a blocking function, how can I execute it, and move on to the next one if after 30 seconds it is still blocked?

Any answer would help me greatly, thank you very much!

closed 2020-11-26 by nathan-casabieille

Blue186, 2024-04-07:
hello , have you solved this problem？I have also encountered this problem recently. I would greatly appreciate it if Can you give me some advice.

vivsamInstr, 2024-12-31:
Hey Blue,
I too got struck with this, could not find a fix but found a workaround: Find out the condition before which the connect hangs and reinitialize the client and go through the complete seq of establishing the conn.
In my case, it used to hang after every 30 successful connections. So I am re-initializing the connection at 25th and am able to move forward in my project. Hope this helps till is a proper/easy fix on all libs