espressif/arduino-esp32 #4398: WiFi Library consumes majority of flash

aeakin1, 2020-10-08:
Using an example sketch that Connects to wifi and blinks the onboard LED, Wifi.h takes up 47.6% of the flash.

`Flash: [=====     ]  47.6% (used 624550 bytes from 1310720 bytes)`
The net of this is it makes it unusable in a larger project.

PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

`/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char *ssid = "your-ssid";
const char *password = "your-password";

#define ONBOARD_LED 2

void setup() {
  
  pinMode(ONBOARD_LED,OUTPUT);

  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  delay(1000);
  digitalWrite(ONBOARD_LED, HIGH);
  delay(100);
  digitalWrite(ONBOARD_LED, LOW);
}
`
```

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform.io build output
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (2.0.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> 1.0
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
Linking .pio\build\esp32doit-devkit-v1\firmware.elf
Retrieving maximum program size .pio\build\esp32doit-devkit-v1\firmware.elf
Building .pio\build\esp32doit-devkit-v1\firmware.bin
Checking size .pio\build\esp32doit-devkit-v1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  11.8% (used 38696 bytes from 327680 bytes)
Flash: [=====     ]  47.6% (used 624550 bytes from 1310720 bytes)
esptool.py v2.6

```

jesse-schein, 2020-10-08:
Yes the arduino library has grown in size for sure. But you must realize the flash memory is used to store a lot of support code to allow the arduino code to work for convenience. the ESP-IDF experience which the arduino code is based on can eliminate the extra unnessesary code if you wish.

lbernstone, 2020-10-08:
esp32 modules can be purchased with up to 16MB flash if you need more space. To change the partition layout see https://docs.platformio.org/en/latest/platforms/espressif32.html#partition-tables.

aeakin1, 2020-10-08:
The identical code for ESP8266 (addressing the ESP8266WiFi Library) is much smaller

`Processing nodemcuv2 (platform: espressif8266; board: nodemcuv2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (2.6.2) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 45 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio\build\nodemcuv2\src\main.cpp.o
Generating LD script .pio\build\nodemcuv2\ld\local.eagle.app.v6.common.ld
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\BearSSLHelpers.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\CertStoreBearSSL.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFi.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFiAP.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFiGratuitous.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\WiFiClientSecureAxTLS.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\WiFiServer.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\WiFiServerSecureAxTLS.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
Compiling .pio\build\nodemcuv2\lib15a\ESP8266WiFi\WiFiUdp.cpp.o
Archiving .pio\build\nodemcuv2\libFrameworkArduinoVariant.a
Compiling .pio\build\nodemcuv2\FrameworkArduino\Crypto.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp-frag.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp-version.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Esp.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FS.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FSnoop.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\FunctionalInterrupt.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\HardwareSerial.cpp.o
Archiving .pio\build\nodemcuv2\lib15a\libESP8266WiFi.a
Compiling .pio\build\nodemcuv2\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\MD5Builder.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Print.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Schedule.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\StackThunk.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\StreamString.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\TypeConversion.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\Updater.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\WMath.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\WString.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\abi.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\base64.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\cbuf.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\cont.S.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\cont_util.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_app_entry_noextra4k.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_eboot_command.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\core_esp8266_features.cpp.o
[... 26 lines cut ...]
Compiling .pio\build\nodemcuv2\FrameworkArduino\sntp-lwip2.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\spiffs\spiffs_cache.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\spiffs\spiffs_check.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\spiffs\spiffs_gc.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\spiffs\spiffs_hydrogen.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\spiffs\spiffs_nucleus.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\spiffs_api.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\sqrt32.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\time.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\uart.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\umm_malloc\umm_info.c.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\umm_malloc\umm_integrity.c.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\umm_malloc\umm_local.c.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\umm_malloc\umm_malloc.cpp.o
Compiling .pio\build\nodemcuv2\FrameworkArduino\umm_malloc\umm_poison.c.o
Archiving .pio\build\nodemcuv2\libFrameworkArduino.a
Linking .pio\build\nodemcuv2\firmware.elf
[... cut ...]
Creating BIN file ".pio\build\nodemcuv2\firmware.bin" using "C:\Users\Andy\.platformio\packages\framework-arduinoespressif8266\bootloaders\eboot\eboot.elf" and ".pio\build\nodemcuv2\firmware.elf"
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  33.0% (used 27036 bytes from 81920 bytes)
Flash: [===       ]  25.5% (used 266196 bytes from 1044464 bytes)
========================================================================================= [SUCCESS] Took 10.94 seconds =========================================================================================`

aeakin1, 2020-10-08:
Let me add 1 additional data point.  

Simply including the library does not cause a large increase in memory.  For example, if I comment out the line:
`  //WiFi.begin("network-name", "pass-to-network");`
 Memory Utilization is 
RAM:   [=         ]   5.1% (used 16588 bytes from 327680 bytes)
Flash: [==        ]  16.9% (used 221173 bytes from 1310720 bytes)

The memory usage only spikes upon calling Wifi.begin()

atanisoft, 2020-10-09:
@aeakin1 as noted above, once you link in the wifi code it brings along the driver and supporting bits. It is not tiny like on the esp8266 and won't be.

There is no issue here.

aeakin1, 2020-10-09:
@atanisoft  What is the cause of this being a heavyweight?  If I were to address the ESP-IDF calls directly, would I potentially be able to make something lighter weight?

atanisoft, 2020-10-09:
@aeakin1 ESP-IDF wouldn't change this much at all. It would be the same since the WiFi library in arduino-esp32 is a very thin wrapper around the ESP-IDF code.

closed 2020-12-25 by stale[bot]