espressif/arduino-esp32 #4401: Hostname does not work in AP mode and with smartphones

Mark-81, 2020-10-09:
### Hardware:
Board:							Olimex ESP32-POE
Core Installation/update date:			11/jul/2017
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                          Ubuntu

### Description:
I'm able to connect the ESP32 in STA mode to an existing WiFi network, starting an HTTP webserver specifying the hostname and any device (PC or smartphone) can reach the pages using `myHostname.local`.

Instead, if I set the ESP32 in AP mode (see code below) a desktop PC can still reach the pages using `myHostname.local` but I cannot using a smartphone (tried a couple). Of course using the IP address always works fine.

I'm not sure what are the differences here between a PC and a smartphone, but most important: is there something else I should do in the code below?

### Sketch: I report only the most relevant parts:
```cpp
#include <WebServer.h>
#include <ESPmDNS.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <WiFiAP.h>

WebServer _server;

void setupAP()
{
    WiFi.softAP("mySSID", NULL);
    WiFi.softAPsetHostname("myHostname");
    MDNS.begin("myHostname");
    _server.on("/", handleRoot);
    _server.begin();
    MDNS.addService("http", "tcp", 80);
}

void handleRoot()
{
    _server.send(200, "text/plain", "Hello world!");
}
```

lasselukkari, 2020-10-09:
Are you using Android? If so, see: https://bugs.chromium.org/p/chromium/issues/detail?id=405925

Mark-81, 2020-10-09:
Il 09/10/20 11:43, Lasse Lukkari ha
      scritto:
    
    
      
      Are you using Android? If so, see: https://bugs.chromium.org/p/chromium/issues/detail?id=405925
    
    
    Yes, I am. But I also tried Firefox instead of Chrome. 
    Anyway, reading the link you provided it seems after 6 years there
    is not a solution yet?

lasselukkari, 2020-10-09:
I removed my original reply because I noticed you said it works in STA mode with your phone and made the conclusion that you must be using iPhone. Android does not support bonjour no matter what connection mode you are using with your ESP32. An yes, that has been the situation for long time now.

Mark-81, 2020-10-09:
To be sure I tried another time right now. I confirm that when my ESP32 is connected to my local WiFi network, I can open the html pages using `http://myHostname` from my **Android** phone, using either Chrome or Firefox. So the problem relies on AP mode only.

sansillusion, 2020-10-10:
Some router will do the .lan or .local themselves. I have an OpenWRT router
and it does work on my Android using the .lan but of course if connected in
ap mode on the ESP32 there is no working dns server or bonjour service to
answer.

closed 2020-10-10 by Mark-81