espressif/arduino-esp32 #4402: How to solve TransfeGuru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1) using Arduino IDE

bioshock2k, 2020-10-09:
Hi,

I have been developing a lora receiver with SD card incorporated using the TTGO lora v2.0 Oled board.
Everything is fine but randomly, after a few hours it displays this error and doesnt reboot by itself, like it should.
Anyone knows what to do?

atanisoft, 2020-10-09:
It appears that you have an ISR in your code (which you didn't share) that is using functions that are not ISR safe (ie: Serial).

bioshock2k, 2020-10-09:
It is probably a function that im calling, thanks i will explore!
Noob question, how did you saw that in the information i gave you? Looked around and couldn´t find any guide to help me understant de stack results.

atanisoft, 2020-10-09:
It is determined from the backtrace details, it has "Interrupt wdt timeout" which is typically an ISR misbehaving. Then in the decoded stacktrace are references to UART (esp32-hal-uart.c) which are known to have issues inside ISRs.

Which LORA library are you using as it is likely there is a bug in that library or how you are trying to use it.

bioshock2k, 2020-10-09:
LoRa.h library.
I might as well put the code in:
void setup() {
pinMode(25,OUTPUT); //BLINK

  Serial.begin(9600);
  delay(1000);
  
   //btStop();
   
   esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
 
  //----------LORA Initialization-------------
   Serial.println("LoRa Receiver Test");
  
  //SPI LoRa pins
  SPI.begin(SCK, MISO, MOSI, SS);
  //setup LoRa transceiver module
  LoRa.setPins(SS, RST, DIO0);

  if (!LoRa.begin(BAND)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  Serial.println("LoRa Initializing OK!");
 
  
  //-------------------SD Card-------------------
  
  sd_spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);

   if (!SD.begin(SD_CS, sd_spi))
        Serial.println("SD Card: mounting failed.");
   else 
        Serial.println("SD Card: mounted.");
        
  //----------WIFI Connection------------
  delay(4000);   //Delay needed before calling the WiFi.begin

   wf.all();                             // WIFI configuration
  
  // Init and get the time
  printLocalTime();
  //-----------------------------
  
  LoRa.onReceive(onReceive);
  LoRa_rxMode();
  
}

void loop() {
  
  if (received){
    printLocalTime();
    write_SD();
    
    if (WiFi.status() == WL_CONNECTED) {   //Check WiFi connection status
      
      HTTPClient http;

      T =message.substring(7,12); //index, lenght
          Serial.print(" Temperature is: ");
          Serial.print(T);
      H =message.substring(15,21); //index, lenght
          Serial.print(" Humidity is: ");
          Serial.println(H);
        
      http.begin("http://things.ubidots.com/api/v1.6/devices/cerveira?token=BBFF-o2K0PwFGIGRCEWYkPLy0jZ7og4vd62");  //Specify destination for HTTP request
      http.addHeader("Content-Type", "application/json");             //Specify content-type header
    
      String http_mess="{\"temperature\":"+T+",\"humidity\":"+H+"}";

      int httpResponseCode = http.POST(http_mess);   //Send the actual POST request

      if (httpResponseCode>0) {
        String response = http.getString();                       //Get the response to the request
        Serial.println(httpResponseCode);   //Print return code
        Serial.println(response);           //Print request answer
      } 
      else {
        Serial.print("Error on sending POST: ");
        Serial.println(httpResponseCode);
      }
      http.end();  //Free resources
      
    }
    else {
      Serial.println("Error in WiFi connection");
      ESP.restart();
    }
     received,received_isr=0;
  }
   
}

//--------------------------Functions-----------------------

void write_SD(){
    File test = SD.open(LOG_PATH, FILE_APPEND);
    if (!test) {
        Serial.println("SD Card: writing file failed.");
    } else {
        Serial.printf("SD Card: appending data to %s.\n", LOG_PATH);
        test.print(message);
        test.print(",Date:");
        test.print(&timeinfo, "%A %B %d %Y %H:%M:%S");
        test.printf("\n");
        test.close();
     }
}

void LoRa_txMode(){
  LoRa.idle();                          // set standby mode
  LoRa.enableInvertIQ();                // active invert I and Q signals
}
void LoRa_rxMode(){
  LoRa.disableInvertIQ();               // normal mode
  LoRa.receive();                       // set receive mode
}
void LoRa_sendMessage(String message) {
  LoRa_txMode();                        // set tx mode
  LoRa.beginPacket();                   // start packet
  LoRa.print(message);                  // add payload
  LoRa.endPacket(true);                 // finish packet and send it
}

void onReceive(int packetSize) {
  message = "";

  while (LoRa.available()) {
    message += (char)LoRa.read();
  }

  Serial.print("Gateway Receive: ");
  Serial.println(message);
  received=1;
}

void onTxDone() {
  Serial.println("TxDone");
  LoRa_rxMode();
}

boolean runEvery(unsigned long interval)
{
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    return true;
  }
  return false;
}

void printLocalTime(){
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");

bioshock2k, 2020-10-09:
void setup() {
pinMode(25,OUTPUT); //BLINK

  Serial.begin(9600);
  delay(1000);
  
   //btStop();
   
   esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
 
  //----------LORA Initialization-------------
   Serial.println("LoRa Receiver Test");
  
  //SPI LoRa pins
  SPI.begin(SCK, MISO, MOSI, SS);
  //setup LoRa transceiver module
  LoRa.setPins(SS, RST, DIO0);

  if (!LoRa.begin(BAND)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  Serial.println("LoRa Initializing OK!");
 
  
  //-------------------SD Card-------------------
  
  sd_spi.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);

   if (!SD.begin(SD_CS, sd_spi))
        Serial.println("SD Card: mounting failed.");
   else 
        Serial.println("SD Card: mounted.");
        
  //----------WIFI Connection------------
  delay(4000);   //Delay needed before calling the WiFi.begin

   wf.all();                             // WIFI configuration
  
  // Init and get the time
  printLocalTime();
  //-----------------------------
  
  LoRa.onReceive(onReceive);
  LoRa_rxMode();
  
}

void loop() {
  
  if (received){
    printLocalTime();
    write_SD();
    
    if (WiFi.status() == WL_CONNECTED) {   //Check WiFi connection status
      
      HTTPClient http;

      T =message.substring(7,12); //index, lenght
          Serial.print(" Temperature is: ");
          Serial.print(T);
      H =message.substring(15,21); //index, lenght
          Serial.print(" Humidity is: ");
          Serial.println(H);
        
      http.begin("http://things.ubidots.com/api/v1.6/devices/XXXXXX?token=XXXXXX");  //Specify destination for HTTP request
      http.addHeader("Content-Type", "application/json");             //Specify content-type header
    
      String http_mess="{\"temperature\":"+T+",\"humidity\":"+H+"}";

      int httpResponseCode = http.POST(http_mess);   //Send the actual POST request

      if (httpResponseCode>0) {
        String response = http.getString();                       //Get the response to the request
        Serial.println(httpResponseCode);   //Print return code
        Serial.println(response);           //Print request answer
      } 
      else {
        Serial.print("Error on sending POST: ");
        Serial.println(httpResponseCode);
      }
      http.end();  //Free resources
      
    }
    else {
      Serial.println("Error in WiFi connection");
      ESP.restart();
    }
     received,received_isr=0;
  }
   
}

//--------------------------Librarys-----------------------

void write_SD(){
    File test = SD.open(LOG_PATH, FILE_APPEND);
    if (!test) {
        Serial.println("SD Card: writing file failed.");
    } else {
        Serial.printf("SD Card: appending data to %s.\n", LOG_PATH);
        test.print(message);
        test.print(",Date:");
        test.print(&timeinfo, "%A %B %d %Y %H:%M:%S");
        test.printf("\n");
        test.close();
     }
}

void LoRa_txMode(){
  LoRa.idle();                          // set standby mode
  LoRa.enableInvertIQ();                // active invert I and Q signals
}
void LoRa_rxMode(){
  LoRa.disableInvertIQ();               // normal mode
  LoRa.receive();                       // set receive mode
}
void LoRa_sendMessage(String message) {
  LoRa_txMode();                        // set tx mode
  LoRa.beginPacket();                   // start packet
  LoRa.print(message);                  // add payload
  LoRa.endPacket(true);                 // finish packet and send it
}

void onReceive(int packetSize) {
  message = "";

  while (LoRa.available()) {
    message += (char)LoRa.read();
  }

  Serial.print("Gateway Receive: ");
  Serial.println(message);
  received=1;
}

void onTxDone() {
  Serial.println("TxDone");
  LoRa_rxMode();
}

boolean runEvery(unsigned long interval)
{
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval)
  {
    previousMillis = currentMillis;
    return true;
  }
  return false;
}

void printLocalTime(){
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");

closed 2020-12-25 by stale[bot]