espressif/arduino-esp32 #4403: ESP32-S2 I2C not working with any I2C based device except DS1307 RTC device

vigneshtg, 2020-10-09:
Hello,
I've checked DS3231 RTC Timer and DS1307 RTC  with ESP32-S2. It's working fine. But except DS3231, DS1307 other I2C based devices are not working with ESP32-S2. The program is successfully uploaded but Serial Monitor is blank and showing nothing. Please help me out.
Thank you.

DougArmstrong, 2020-10-09:
This sounds like it might not be a CPU related issue.  Couple of obvious questions just be clarify:
1) When you say the Serial monitor is blank, are you saying you are using an I2C bus monitor and there is not I2C traffic?  Either way? Do you see traffic from the DS3231?  Are you referring to the serial UART traffic for debugging in the development tool?

2) What pins are you using?

3) If you are not using an I2C bus monitor, it might be worth putting something on the hardware and seeing if data is actually being sent.

4) Are the I2C addresses correct for the devices?

Hope something here helps.
Good luck.

vigneshtg, 2020-10-09:
I'm using SDA - Pin 8 and SCL - Pin 9 on ESP32-S2. Also using Wire library for I2C interface. But the same thing working fine with ESP32 but not with ESP32-S2. Further I checked with I2C scanner and I2C address is correct. How it can be solved now?

DougArmstrong, 2020-10-09:
I am not tech support for Expressif or the development team, just another user like yourself.  You need to provide a great deal more detailed information.  Can you provide screen shots, and explain in detail what you think is the issue?  For example perhaps pull up resistors are needed? Without a great deal more information it will be difficult for anyone to assist, certainly the dev team won't be interested unless others are having the same issue.

When you say the Serial monitor is blank, are you saying you are using an I2C bus monitor and there is not I2C traffic? 

Is the ESP32 sending commands and not getting responses?  Do you see bidirectional traffic?

Do you see bidirectional traffic from the DS3231 on your I2C monitor? 

Are you referring to the serial UART traffic for debugging in the development tool when you say the screen is blank?

Do you have a scope to look at these signals directly?

Just a couple suggestions... Good luck.

vigneshtg, 2020-10-09:
I'm trying to interface MLX90614 Sensor with ESp32-S2 and using Sparkfun MLX90614 library: 
https://github.com/sparkfun/SparkFun_MLX90614_Arduino_Library
 I'm using the demo code which is available within this library. But when I connected and uploaded the code to ESP32-S2, the serial Monitor is blank and it's not printing or displaying temperature values on Serial Monitor.

vigneshtg, 2020-10-15:
Any help?

lbernstone, 2020-10-15:
Follow the instructions in the [Wire docs](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/doc/i2c_debugging.md) and post your results here.

vigneshtg, 2020-10-16:
Sure. I'll post the results once done. Thanks!

closed 2020-12-31 by stale[bot]

SuGlider, 2021-09-14:
Arduino Core 2.0.0 and PR #5664 may solve this issue.
Please test it and let me know.
Thanks.