espressif/arduino-esp32 #4404: rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT) when BLE starts

cvrdx, 2020-10-09:
Hello,
I have a ESP32-WROOM-32UE on a custom PCB board with thick traces and decoupling capacitors by the module, with battery connector and USB.
When I setup the BLE server, in the setup() function, the MCU sometimes crashes and then gets stuck in a loop where it would crash at the same point every time.
This mostly occurs when the board is running on battery (even at a good charge level of 3.8V-4V), but sometimes with USB connection.

This is the following message printed when the reset occurs. I am also printing the "Reset Reason".

```
Starting BLE...ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

Wake: OTHER - unknown wake source
E (3669) sleep: Incorrect wakeup source (2) to disable.
CPU0 reset reason:
POWERON_RESET
Vbat power on reset
CPU1 reset reason:
EXT_CPU_RESET
for APP CPU, reseted by PRO CPU
```

Below you can see the ble function that I used. In one of my tests when trying to solve the problem, I reduced the advertising power to its minimum, ESP_PWR_LNL_N14:

```
void setup_ble_server(char suffix[2], BLEServerCallbacks* pCallbacks) {
  dev_name_suffix[0] = suffix[0];
  dev_name_suffix[1] = suffix[1];
  // Create the BLE Device
  dev_name_prefix += dev_name_suffix[0];
  dev_name_prefix += dev_name_suffix[1];
  BLEDevice::init(dev_name_prefix);
  BLEDevice::setMTU(300);
  BLEDevice::setPower(ESP_PWR_LVL_N14);
  esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_N14);

  // Create the BLEUART Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(pCallbacks);
  // Create the BLE Service
  BLEService *pService = pServer->createService(SERVICE_UUID);
  // Create a BLE Characteristic
  pCharacteristic1 = pService->createCharacteristic(
                      CHARACTERISTIC_UUID1,
                      BLECharacteristic::PROPERTY_READ   |
                      BLECharacteristic::PROPERTY_NOTIFY 
                    );
  pCharacteristic2 = pService->createCharacteristic(
                      CHARACTERISTIC_UUID2,
                      BLECharacteristic::PROPERTY_WRITE
                    );
  // https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.descriptor.gatt.client_characteristic_configuration.xml
  // Create a BLE Descriptor
  pCharacteristic1->addDescriptor(new BLE2902());
  pCharacteristic2->addDescriptor(new BLE2902());
  // Start the service
  pService->start();

  // Create the BLEOTA Server
  // OTAServer = BLEDevice::createServer();
  // OTAServer->setCallbacks(OTACallbacks);
  // Create the BLE Service
  BLEService *OTAService = pServer->createService(OTA_SERVICE_UUID);
  // Create a BLE Characteristic
  OTACharacteristic = OTAService->createCharacteristic(
                      OTA_CHARACTERISTIC_UUID,
                      BLECharacteristic::PROPERTY_WRITE |
                      BLECharacteristic::PROPERTY_READ  |
                      BLECharacteristic::PROPERTY_NOTIFY
                    );
  OTACharacteristic->setCallbacks(new MyOTACallbacks() );
  // Start the service
  OTAService->start();

  BLEAdvertisementData advData;
  BLEAdvertisementData advDataResp;
  advData.setName(dev_name_prefix);
  advData.setPartialServices(pService->getUUID());
  advDataResp.setCompleteServices(OTAService->getUUID());
  // advData.setCompleteServices(pService->getUUID());
  // advData.setCompleteServices(OTAService->getUUID());

  // Start advertising
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  // pAdvertising->addServiceUUID(SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setScanFilter(false,false);
  pAdvertising->setMinPreferred(0x0);  // set value to 0x00 to not advertise this parameter
  pAdvertising->setAdvertisementData(advData);
  pAdvertising->setScanResponseData(advDataResp);
  BLEDevice::startAdvertising();
  Serial.println("Waiting a client connection to notify...");
```

I am using the stable espressif32 version. 
Any idea where the problem is? The reseat reason does not point to anything in particular, as when I press the RESET button, the same message is printed.

When I look at the 3.3V voltage at the oscilloscope, I can see that it looks good, with a stable 3.3V throughout. 
When I look at the RESET Signal, I can see that it goes from HIGH to LOW for about 200ms, when the reset occus.

Any idea where the problem might be coming from?

[image]

lbernstone, 2020-10-09:
During initialization, the esp32 requires up to 400mA (@3V3).  There is not really any way to reduce this.  If you are not able to provide enough, it will reset.  It should show you a brownout error when this happens, but it is possible the drawdown is simply so fast that it crashes before the error threshold is reached.  Does it ever fail if you use bench power?

cvrdx, 2020-10-15:
I know that the problem is not caused by the power supply, because If I hook a bare ESP32 on a breakout PCB ([from Seeed](https://www.seeedstudio.com/ESP32-Breakout-Kit-p-2783.html)) and I power it from my PCB, it works reliably, there is no reset problem when the BLE starts.

That could point to a problem on my PCB with a level on a GPIO triggering a reset - which pins could do that?
I can find information on the [espressif Github](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection) and also different information on [Random Nerds Tutorial](https://randomnerdtutorials.com/esp32-pinout-reference-gpios/
) , but none of them conclusive...

chegewara, 2020-10-15:
Power supply is the only reasonable explanation. If this is custom PCB then we cant guess what is problem with that design.
You can start topic on esp32 forum or try to get help from espressif, but issue tracker on github is not the right place.

libasoles, 2020-10-16:
I'm having the same problem. 
See the error:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```

People in forums use to point to power but I'm sure it's not that.

This is happening to me when I add the ESP32 BLE Arduino library to my project and run this line:

`BLEDevice::init("");`

However, If I try the Arduino IDE examples for the library it works fine. 

So, it's not the power. It's an obscure combination of my code and the library or something. I can't say. Or maybe a conflict between libraries (though I'm only using FastLED apart from BLE)

chegewara, 2020-10-16:
@libasoles What in those logs is telling you its the same problem?
I see here completely different logs, which is telling me its not the same problem.

libasoles, 2020-10-16:
Maybe my log is shorter, but I see it's the same than the original post. A message telling the kit reboots.

[image]

chegewara, 2020-10-16:
And i see completely different logs:
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

VS
> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)

libasoles, 2020-10-16:
I see. Thank you, and sorry for the confusion.

coriolan-v, 2020-12-27:
It looks like this was a power issue after all: I have replaced my 600mA LDO with a 1A LDO, and soldered 2 massives 47uF capacitors right next to the 3.3V input of the ESP32, and I've never seen the problem again.
The likely means that the esp32-wroom-32ue draws more current thaw what stated in its datasheet (and more than the ESP32-WROOM-32D that I was using before, that worked with the same LDO and same FW, and without these capacitors)

closed 2021-06-23 by cvrdx

ericb013, 2023-01-12:
Just incase this will help anyone. I too was having this issue on an older MacBook Pro. Could not figure this out until someone on this post mentioned power. I ended up restarting my Mac and the issue went away. Hope this helps save someone a hassle.

EmanuelJimenez23, 2024-11-02:
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
E (143) quad_psram: PSf5�ID read error: 0xffffffff, PSRAM chip not found or not supported
Accelerometer Test

No se detectó el ADXL345... ¡Revisa las conexiones!

karo2022, 2025-03-01:
[image] Que hago si me sale este error?

ThuZarKhaing, 2025-03-20:
[image]

I have a problem with this. Can someone give me advice?