espressif/arduino-esp32 #4405: error: expected ')' before numeric constant  #define IPADDR_NONE         ((u32_t)0xffffffffUL)

MarkEvens, 2020-10-10:
ESP-IDF 4.0.1 with Arduino as components

suddenly i got above title error and don't understand why i am changing think in component files.

directory for this file is : $HOME/esp/esp-idf/components/lwip/lwip/src/include/lwip/ip4_addr.h:63:37

turkysha, 2020-10-10:
You are missing ')' in that file ( in ip4_addr.h).

MarkEvens, 2020-10-12:
@turkysha 
#define IPADDR_NONE         ((u32_t)0xffffffffUL)

this is the line 63 of ip4_addr.h what do you think that i have to change

MarkEvens, 2020-10-12:
my problem solved by below issue
https://github.com/Microsoft/azure-iot-developer-kit/issues/169

closed 2020-10-12 by MarkEvens

ameerhazo, 2020-11-29:
Hello @MarkEvens, I've stumbled upon the same error while trying to build an ESP-IDF component (coap) in my esp32 arduino/ESP-IDF mix project. Could you tell me how you resolved the issue? I'm asking this because the solution you linked is done on a different board. Thanks for the help.

moefear85, 2022-02-11:
I face this too, although solved it by rearranging the order/locations of includes, since it appears different components are redefining the same symbols but in different ways. It might also be related to mixing C with C++ components. Usually this means extra code to safeguard has to be embedded. Unfortunately although now I'm following the same steps as I usually do, it's still not working. Does anybody know the exact systematic way to resolve this? In this case I believe the conflict is between Arduino and LwIP, as each defines IPADDR_NONE differently.

awaters-meltin, 2022-06-07:
In case anyone else encounters this - the fix for me was to include the Arduino header first:
```
#include <WiFiUdp.h> // before lwip RE: https://github.com/espressif/arduino-esp32/issues/4405
#include <lwip/sockets.h>
```

Magnatto, 2022-08-22:
> In case anyone else encounters this - the fix for me was to include the Arduino header first:
> 
> ```
> #include <WiFiUdp.h> // before lwip RE: https://github.com/espressif/arduino-esp32/issues/4405
> #include <lwip/sockets.h>
> ```

Thanks! Worked for me