espressif/arduino-esp32 #4407: WiFi.softAP triggers watchdog timer and restarts ESP32. I cant start anything with wifi on my esp32

turkysha, 2020-10-10:
### Hardware:
Board:							ESP32 DevKit-C
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Im just trying to start simple webserver. I cant even get softAP to work. My WiFi.softAP(ssid) function just failes. I have very simple program just to start softAP and watchdog timer is somehow triggered and my ESP32 restarts.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <WiFi.h>
const char *ssid = "ESP32-WiFi";
const char *password = "12345678";

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println("Configuring access point...");
  WiFi.softAP(ssid);
  delay(1);
  Serial.print("IP address: ");
  Serial.print(WiFi.softAPIP());
  Serial.println("Server started");
}

void loop() {
 
}
```

### Debug Messages:
```
ts Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

Configuring access point...
E (11051) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11051) task_wdt:  - IDLE0 (CPU 0)
E (11051) task_wdt: Tasks currently running:
E (11051) task_wdt: CPU 0: wifi
E (11051) task_wdt: CPU 1: IDLE1
E (11051) task_wdt: Aborting.
abort() was called at PC 0x400d463f on core 0

Backtrace: 0x4008e8b0:0x3ffbe170 0x4008eae1:0x3ffbe190 0x400d463f:0x3ffbe1b0 0x40086ecd:0x3ffbe1d0 0x400dea79:0x3ffb52f0 0x400deacd:0x3ffb5310 0x400def35:0x3ffb5330 0x400dc56d:0x3ffb53d0 0x400dc812:0x3ffb5450 0x400dd10e:0x3ffb54a0 0x400ddc3e:0x3ffb54c0 0x400de018:0x3ffb54e0 0x400d5243:0x3ffb55a0 0x400d54c2:0x3ffb55d0 0x400eeddc:0x3ffb5600 0x400eeecc:0x3ffb5630 0x400ef1d6:0x3ffb5660 0x400eb5ae:0x3ffb5690 0x400919f7:0x3ffb56b0 0x4008aff9:0x3ffb56f0

Rebooting...

```

lbernstone, 2020-10-10:
Put a delay(50); in loop, or you could even put some code...

turkysha, 2020-10-10:
@lbernstone That did not solved the issue. I get the same output. Thanks for so fast reply!

turkysha, 2020-10-11:
@lbernstone I tried everything and cant get it working. I tried code on fresh new esp32 from package and it was the same error, so its not the hardware, its something in software. I cant even tell why is it happening. Few months ago It worked perfectly.

felmue, 2020-10-11:
Hi @turkysha 

I have the WiFiAccessPoint example running fine on a ESP32_LyraT board. So I was wondering - in your case could it be the power supply not delivering enough 'juice' when WiFi gets initialised? Also, do you happen to know / remember what has changed from a few months ago when it has worked for you?

Thanks
Felix

turkysha, 2020-10-11:
@felmue Hi,
on my regular project I have USB host shield connected to esp32 via SPI, so maybe it is power. I would try now to connect it to my second esp32 that has newer be used besides I tried to connect it few days before but without delay in loop(). Nothing has changed in this months. In May I made example with webserver which could turn the led off and on and it was working. Few days ago I opened that same project which I didnt open in months and tried to run it and I got this error.

lbernstone, 2020-10-11:
Sounds like something is damaged in your IDE. Turn on verbose debugging during compilation and see if it is using libraries from locations outside of arduino-esp32.  Try to remove/install  esp32 in boards manager.  Otherwise, you probably need to wipe your .arduino15 directory and remove/install Arduino.

turkysha, 2020-10-11:
@lbernstone I reinstalled everything and its still not working. I deleted Arduino15 folder, deleted arduino folder from documents and installed everything new. 
I got this in my arudino compiling output:

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Korisnik\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.42.0_x86__mdqgnx93n4wtt\libraries\WiFi
Using library WiFi at version 1.0 in folder: C:\Users\Korisnik\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi 

I think he is using correct library here? I can post you whole log.

felmue, 2020-10-12:
Hi @turkysha 

my apologies if you already tried that. Does a very simple project which doesn't use WiFi (for instance just turn on / off a GPIO) run properly on your ESP32 board?

Thanks
Felix

turkysha, 2020-10-12:
@felmue Hi,
don't worry, its oke. Projects without using WiFi module are working. I tried now bluetooth example and it is also failing with same exception. Could it be something with my windows? Hello world example and blink example are working. 
Thanks

felmue, 2020-10-12:
Hi @turkysha 

please note that I am only a developer / user trying to help.

That said, I wonder if `WiFi.softAP()` is returning an error causing the following statements to fail. You could try to add these lines after `Serial.begin()` to enable error logging in the libraries:
```
  Serial.setDebugOutput(true);
  esp_log_level_set("*", ESP_LOG_ERROR);
```
in addition it might be helpful to check the return value of `WiFi.softAP()` being `true` or `false`.

Thanks
Felix

turkysha, 2020-10-13:
@felmue 
I added that and It did not help. I get same outup like I posted in issue in "Debug Message". I can't check what WiFi.softAP() is returning because somewhere in that function the watchdog gets triggered and the function fails. I am going to comapre what is done in my working example and WiFi.softAP() function and maybe figure out what is different and change that.

turkysha, 2020-10-14:
@felmue @lbernstone 
I found the issue but I can't get it wokring yet. Issue is that WiFi.softAP() is missing ESP-NETIF initialization. When I run my working example without esp_netif_init() I get the same behavior like in this broken example. I made everything the same in my broken example like in my wokring but I can't get esp_netif_init() to compile. So I checked everything in WiFi.softAP() and everything is correct except that esp_netif_init() is missing. When I put esp_netif_init() my compiler says that he can't find the function even that all my #include's are same as in my working example. So I tried to #include "esp_netif.h" but my compiler can't find that file. So I need to somehow call esp_netif_init() but I can't get it to work. Have You any solutions for this? And I'm 100% sure that not initializing ESP-NETIF is causing the problem because only thing that is different from my working example and this broken is not calling that function and when I remove that function from my working example I get same behavior like in my borken example. Can you please help me?

lbernstone, 2020-10-14:
That function does not exist in 1.0.4.  The function is tcpip_adapter_init

atanisoft, 2020-10-14:
@turkysha are you trying to use arduino-esp32 1.0.4 with IDF v4.1+?

turkysha, 2020-10-14:
@atanisoft yes, IDF v4.1

atanisoft, 2020-10-14:
arudino-esp32 will not work with IDF v4.1 without a number of changes that won't likely be made.

turkysha, 2020-10-14:
@atanisoft oh I didn't know that. To what version should I downgrade?

atanisoft, 2020-10-14:
idf v3.3 if you want to use arduino-esp32 1.0.4. Or you can move up to IDF v4.2 or IDF v4.3 (master) and use the esp32s2 branch of arduino-esp32 and it will work.

felmue, 2020-10-14:
Hi @atanisoft 

thank you for chiming in. I didn't know that either.

@turkysha : good luck!

Thanks
Felix

lbernstone, 2020-10-14:
Please note how correctly supplying information at the top of your issue would have saved a lot of time.

turkysha, 2020-10-14:
@lbernstone 
Yes, I said only arduin-esp32 version not esp-idf but nobody asked so I didn't even think that would be a problem. And having working code and not expecting developer to know which versions are exactly compatible would aslo saved a lot of time.

turkysha, 2020-10-14:
@lbernstone @felmue @atanisoft 
Thank You all for all your help! I tested with eps-idf 3.3 and arduino-esp32 1.0.4 and it is finally working now!

closed 2020-10-14 by turkysha

waecoo, 2020-12-28:
I'm having the same issue with Arduino IDE 1.8.13. I've installed esp32 using Boards Manager and tried 1.0.4 and other versions.

How do I check which version of eps-idf I'm using?

Should I follow this installation instructions instead of Boards Manager?
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md

Thank you!

KrikkaKripto, 2023-03-09:
@turkysha maybe you could explain a bit better what your solution was exactly also?