espressif/arduino-esp32 #4409: Can't turn off 'startup' logs

mkeveney, 2020-10-12:
I need to use the main serial pins on my board (AI-Thinker ESP32-CAM) to communicate
with another device.

On power-up, my serial lines emit a bit of debugging output before 'setup()' 
gets control:

    rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
    configsip: 0, SPIWP:0xee
    clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
    mode:DIO, clock div:2
    load:0x3fff0018,len:4
    load:0x3fff001c,len:1044
    load:0x40078000,len:8896
    load:0x40080400,len:5828
    entry 0x400806ac
    --This is the first serial output from my 'setup()' function
    .
    .
    .

How do I turn this logging off?

I'm using PlatformIO, in case that matters.  I've tried setting this in my platformio.ini file:

    build_flags = -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_NONE

But this gives no change.

lbernstone, 2020-10-12:
Pull gpio15 low at boot.
https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection#other-pins

mkeveney, 2020-10-12:
Good to know, but not an option in my case.  I need to use the SD card too, which is hardwired to gpio 15 on this board.

https://raw.githubusercontent.com/SeeedDocument/forum_doc/master/reg/ESP32_CAM_V1.6.pdf

There must be some way to recompile the boot loader with this logging turned off.  Can anyone point me to it?

MarkyAD, 2020-10-12:
If you're not using i2c, assign gpio26 and 27 to Serial1 and use that instead of Serial.

MarkyAD, 2020-10-12:
Of course they're 3.3v TTL level uart pins, so whatever serial device you've got has to be happy with that or you'll have to use a level converter to/from USB or RS232 levels. Sounds like extra pain for a workaround.

Maybe gobbling up the boot messages until the serial buffer is empty might be an easier option.

mkeveney, 2020-10-12:
Thanks, but that won't work with this board either.  Basically, all the pins are used up by the built-in camera and SD card.  Pinout is described here:
https://randomnerdtutorials.com/esp32-cam-ai-thinker-pinout/
...with a few good tidbits of info in the comment section.

MarkyAD, 2020-10-12:
Indeed it does show the i2c pins being used by the cam. It certainly doesn't have a lot of free pins.. bummer.

closed 2020-12-25 by stale[bot]