espressif/arduino-esp32 #4413: Degradation of audio PWM output during SD Card read

RafigRzayev, 2020-10-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE

### Description:
I am trying to play audio wav. file stored on SD card via PWM. I had two approaches to do it:
1 - Read part of data from SD card via SPI to a buffer. Then, loop through the data and send it to PWM via  ledcWrite() and repeat the whole process until audio file is finished. In this case audio quality is pretty good. Only issue is that it leads to "pause/delays" during audio output since it takes quite some time to read chuck of data from SD card and save it in the buffer. Afterwards, I switched to SD_MMC library to decrease the delays. Delay time decreased 2 times but it was not sufficient. Also, I decreased sampling rate and resolution of original audio file to 24kHz and 8 bit, but again delays were noticable. Decreasing buffer size to small values in order to decrease SD read time leads to bad audio quality because of high frequency read delays. This lead me to second approach.
2 - In this approach there are two buffers, one for loading data and one for playing. In a loop, I use ledcWrite() to update the PWM signal according to the audio sample. PWM signal should not be updated too fast to keep up with the sampling rate, so there is a period of time where we have to wait until the next PWM update. In this free period, I am reading data from SD card. This approach removes delays completely, which is good. But on the other hand, it led to degradation of audio quality. 

So, in the first approach, we are reading a big block of data and playing it afterwards. It leads to good quality audio with delays in between, which is normal. In the second approach, we are sending one byte of data, and in the time period until the next send operation, we are reading from SD card. This removes delays, but somehow degrades audio quality. 

Logically second approach should produce good quality audio, but I can't understand why it gets degraded. What I have tried so far: 
- Reading from SD card on one core, playing on other
- Tried both SD and SD_MMC libraries
- Tried DAC instead of PWM
- Tried various power supplies
- Observed PWM waveforms on oscilloscope: both approach 1 and 2 seem the same while former has good quality and second is degraded
- Tried Low-Pass / High-Pass RC filters on amplifier input

None of these gave any result. I want to emphasize that the approach 1 produces decent audio output, so there are no problems related to speakers/amplifiers etc. Only thing that changes audio quality is adding the line for reading from SD card. 

And another important observation: Delay values used to synchronize PWM output update rate with sampling rate are currently written manually. I tried to automate it via using micros() to calculate time consumed by code for each sample, but it looks like using micros() inside the PWM loop also leads to same degradation. I have a feeling that once some piece of code is added after ledcwrite() except for delay() function, it leads to a problem. 

In order to use the code, you should use SD card, amplifier, speaker and 8 bit, 24kHz audio file inside SD card. Here is dropbox link to the one I am using:
https://www.dropbox.com/s/cx7tcc77kne122b/sound_1.wav?dl=0

Also, if you just want to hear the difference, here are the samples that I have recorded from the project:
Approach 1 audio sample: https://www.dropbox.com/s/upxmxhjfv0rvxga/good_quality_with_read_delays.mp3?dl=0
Approach 2 audio sample: https://www.dropbox.com/s/1hhp74zpy3bno76/degraded_quality_no_delays.mp3?dl=0
In the code below, you can select which approach you want to play in the loop() section. 

I would appreciate your assistance. 

### Sketch:  
```cpp

/*
 * SPI Connection:
 * SD Card      ESP32
 * 
 * CS           5
 * CLK          18
 * MISO         19
 * MOSI         23
 * GND          GND
 * Vdd          Vdd  
*/

#include "FS.h"
#include "SD.h"
#include "SPI.h"

#define PWM_PIN 17
#define PWM_CHANNEL 0
#define PWM_FREQUENCY 24000
#define PWM_RESOLUTION 8
uint8_t duty_cycle = 0;

File audio_file;

/*                                        Case 1:
   We have single data buffer. Firstly we load data from SD card to this buffer.
   Afterwards, we play the loaded part. In this scenario audio quality is great
   Only issue is that reading chunk of data from SD card takes sufficient amount of time,
   and this causes long delays between two consecutive audio chunks. As a result,
   following code produces high quality audio which stops every second to load data.
   I have also used SD_MMC library instead of SD via SPI to increase read the read speed.
   Speed increased by a factor of 2, but it was not nearly enough to remove interruptions
   from the audio output.
*/

// Sampling rate of audio to be played is 24kHz, so this buffer corresponds to one second of data
#define BUFFER_SIZE 24000
// We load data into this buffer, and afterwards send it to PWM from here.
uint8_t buf[BUFFER_SIZE];

void load_and_play_separately() {
  // Load part of audio
  audio_file.read(buf, BUFFER_SIZE);

  // Play part of audio
  for (int i = 0; i < BUFFER_SIZE; ++i) {
    duty_cycle = 255 - buf[i];
    ledcWrite(PWM_CHANNEL, duty_cycle);
    /* Delay amount has been calculated manually:
       1/f = 1/24000 = 41 us. -> time to wait before updating PWM output signal
       code consumes ~7 us for each sample,
       so we add up 34us to synchronize with sampling rate. */
    delayMicroseconds(34);
  }
}

/*                                        Case 2:
   PWM needs delays anyway, otherwise data would be updated too fast. So idea is to
   load data from SD card during these delays. We shorten delay by the amount
   it takes to read one byte of data from SD card. This way we will no longer have
   to pause audio to load data. Loading and playing data simultaneously is achieved
   by using two buffers: one to receive data from SD card, another to send data to PWM.
   Once buffers have been fully processed, they will be swapped, and audio will continue
   normally. Using this method removes interruptions in audio completely, but audio quality
   degrades.
*/

// Buffers for loading and playing
uint8_t buf_1[BUFFER_SIZE];
uint8_t buf_2[BUFFER_SIZE];
uint8_t* play_buffer = buf_1;
uint8_t* load_buffer = buf_2;

void load_and_play_simultaneously() {
  // Play & Load
  for (int i = 0; i < BUFFER_SIZE; ++i) {
    duty_cycle = 255 - play_buffer[i];
    ledcWrite(PWM_CHANNEL, duty_cycle);
    load_buffer[i] = audio_file.read(); // Line that causes degradation.
    // delay is calculated same way as in load_and_play_separately()
    delayMicroseconds(25);
  }
  
  // Swap buffers
  uint8_t* temp = play_buffer;
  play_buffer = load_buffer;
  load_buffer = temp;
}

void setup() {
  // Enable UART for logging
  Serial.begin(115200);

  // Initialize SD module
  if (!SD.begin()) {
    Serial.println("Card Mount Failed");
    return;
  }

  // Open audio file
  audio_file = SD.open("/SOUND_1.WAV");
  if (!audio_file) {
    Serial.println("Failed to open file for reading");
    return;
  }

  // Initialize PWM module
  ledcSetup(PWM_CHANNEL, PWM_FREQUENCY, PWM_RESOLUTION);
  ledcAttachPin(PWM_PIN, PWM_CHANNEL);
}

void loop() {
  // Use one of below:
  
  //load_and_play_separately();
  
  load_and_play_simultaneously();
}

atanisoft, 2020-10-13:
loading from an SD card byte by byte will be a lot slower than reading a block at a time (512 or 4096 bytes) as it needs an SPI transaction per byte which will add up to a lot more than your 25usec delay between samples.

I'd suggest looking at I2S rather than LEDC to generate the audio output and pre-load/buffer the audio samples so the calls to SD will not cause delays unnecessarily.

RafigRzayev, 2020-10-13:
Thank you very much for fast reply. 

I measured the time it takes to read one byte from SD card watched it through serial port. It seemed pretty stable to me. The current 25us delay is actually calculated by taking into consideration SPI read. 41us is the required delay by the audio sampling rate. 41us - 25us is the time used by SPI read and ledwrite. Please correct me if I misunderstood something.

RafigRzayev, 2020-10-13:
I measured it following way: 
```
    auto t1 = micros();
    duty_cycle = 255 - play_buffer[i];
    ledcWrite(PWM_CHANNEL, duty_cycle);
    load_buffer[i] = audio_file.read(); // Line that causes degradation.
    auto t2 = micros();
    Serial.println(String(t2 - t1)); //-> This is 16us
    // delay is calculated same way as in load_and_play_separately()
    delayMicroseconds(25);
```

RafigRzayev, 2020-10-13:
If you could check the audio samples in the links, I believe it will give more insight. Because it is only voice that has been changed(degraded). Audio speed and etc. are normal.

atanisoft, 2020-10-13:
very likely LEDC is not using a 24kHz frequency, `ledcSetup` should be returning the actual frequency used. As noted before, using I2S to generate the sound output would ensure the output is as expected from the input data.

RafigRzayev, 2020-10-13:
It is returning 24k, I checked it. ledcwrite can be used, the example for approach 1 clearly produces good sound.

RafigRzayev, 2020-10-14:
I was able to solve the problem in Approach 1 by drastically decreasing SD card read time according to the following post: https://github.com/espressif/arduino-esp32/issues/1611 

Now audio quality is pretty good and without delays. 

I will share user-friendly PWM Audio source code in following days here. 

P.S. It is still unknown why approach 2 degrades.

RafigRzayev, 2020-10-26:
The code for playing PWM audio can be found below. Speaker should be connected to PWM and ground pins, plus an amplifier in between to increase the volume.  PWM audio depends on manual time delays, which is not reliable. Code is for demo purposes only, it doesn't cover all of wave format details. 

As @atanisoft said, I2S is much better, because it handles the time synchronization. I deciced to abandon PWM and continue with I2S.

```cpp
/*
   Connect the SD card to the following pins:

   SD Card | ESP32
      D2       12
      D3       13
      CMD      15
      VSS      GND
      VDD      3.3V
      CLK      14
      VSS      GND
      D0       2  (add 1K pull up after flashing)
      D1       4
*/

#include "FS.h"
#include "SD_MMC.h" // Preferred to SD_SPI due to faster read/write speed

// PWM parameters
#define PWM_PIN 17
#define PWM_CHANNEL 0
#define PWM_RESOLUTION 8

// Play .wav file on SD card via PWM
void play(const char* song_name) {
  // Check song name
  if (song_name == NULL) {
    Serial.println("Song name empty");
    return;
  }

  // Init SD interface
  if (!SD_MMC.begin()) {
    Serial.println("SD_MMC initialization error");
    return;
  }

  // Open file to determine file size and format
  File file = SD_MMC.open(song_name);
  if (!file) {
    Serial.println("Failed to open file");
    return;
  }

  // Skip RIFF header to reach the file size section
  for (int i = 0; i < 4; ++i) file.read();

  // Extract file size
  uint8_t FILE_SIZE[4];
  file.read(FILE_SIZE, 4);
  uint32_t file_size = FILE_SIZE[3] << 24 | FILE_SIZE[2] << 16 | FILE_SIZE[1] << 8 | FILE_SIZE[0];

  // Skip forward to reach audio format
  for (int i = 0; i < 12; ++i) file.read();

  // Extract audio format
  uint8_t AUDIO_FORMAT[2];
  file.read(AUDIO_FORMAT, 2);
  if (AUDIO_FORMAT[1] << 8 | AUDIO_FORMAT[0] != 1) {
    Serial.println("Format not supported");
    return;
  }

  // Extract number of channels
  uint8_t NUM_CHANNELS[2];
  file.read(NUM_CHANNELS, 2);
  if (NUM_CHANNELS[1] << 8 | NUM_CHANNELS[0] != 1) {
    Serial.println("Only mono is supported");
    return;
  }

  // Extract sampling rate
  uint8_t SAMPLE_RATE[4];
  file.read(SAMPLE_RATE, 4);
  uint32_t sample_rate = SAMPLE_RATE[3] << 24 | SAMPLE_RATE[2] << 16 | SAMPLE_RATE[1] << 8 | SAMPLE_RATE[0];

  // Skip forward to reach next section
  for (int i = 0; i < 6; ++i) file.read();

  // Extract resolution
  uint8_t BITS_PER_SAMPLE[2];
  file.read(BITS_PER_SAMPLE, 2);
  if (BITS_PER_SAMPLE[1] << 8 | BITS_PER_SAMPLE[0] != 8) {
    Serial.println("Only 8 bit resolution is supported");
    return;
  }

  // At this point we have verified that wav file is mono and 8 bit.
  // Sampling rate and file size is known as well
  file.close();

  // For example, if sampl. freq. is 24kHz, time between two consecutive samples is 1/24000 ≈  41 microseconds
  uint32_t PWM_PERIOD = 1000000 / sample_rate; // in us

  // Open file for reading data & playing
  String fullpath = String("/sdcard") + String(song_name);
  // Here we open file with different method to increase the read speed in play loop
  FILE* audio_file = fopen(fullpath.c_str(), "rb"); // rb stands for read in binary mode
  if (audio_file == NULL) {
    Serial.println("Error opening file");
    return;
  }

  // Initiate PWM
  ledcSetup(PWM_CHANNEL, sample_rate, PWM_RESOLUTION);
  ledcAttachPin(PWM_PIN, PWM_CHANNEL);
  uint8_t duty_cycle = 0;

  // Create buffer for storing audio data
  const int BUFFER_SIZE = 24000;// Decrease buffer size for short samples
  uint8_t* buf = (uint8_t*)malloc(BUFFER_SIZE);

  // Calculate how much time is used for PWM operations on average
  uint32_t average_time_used_by_code = 0;
  for (int i = 0; i < 100; ++i) {
    auto t1 = micros();
    duty_cycle = 255 - buf[i];
    ledcWrite(PWM_CHANNEL, duty_cycle);
    auto t2 = micros();
    average_time_used_by_code += t2 - t1;
  }
  average_time_used_by_code /= 100;

  // Calculate time delay between two consecutive PWM updates in order to keep up with sampling rate
  const int time_to_wait = PWM_PERIOD - average_time_used_by_code;
  if (time_to_wait <= 0) {
    Serial.println("Unable to keep up with audio frequency");
    return;
  }

  uint32_t counter = 0;
  while (counter + BUFFER_SIZE < file_size) {
    // Read part of audio. SD_MMC file.read() is not used because it is much slower compared
    read(fileno(audio_file), buf, BUFFER_SIZE);

    // Play part of audio
    for (int i = 0; i < BUFFER_SIZE; ++i) {
      duty_cycle = 255 - buf[i];
      ledcWrite(PWM_CHANNEL, duty_cycle);
      delayMicroseconds(time_to_wait);
    }

    counter += BUFFER_SIZE;
  }

  // Release memory
  free(buf);
  // Disable PWM output
  ledcDetachPin(PWM_PIN);
  // Close file
  fclose(audio_file);
  // Exit SD interface
  SD_MMC.end();
}

void setup() {
  Serial.begin(115200);
  delay(1000);
  // any 8 bit wave audio file
  play("/sound4.wav");
}

void loop() {

}`

closed 2020-10-26 by RafigRzayev

JustinWeidmann, 2021-02-10:
Do you mind sharing the I2S version of this code?

RafigRzayev, 2021-02-10:
@JustinWeidmann Sure, I will post this week

RafigRzayev, 2021-02-13:
@JustinWeidmann 

Here is the code https://github.com/RafigRzayev/WAV_Player_ESP32. 

I had to reorganize the original I2S code in order to share here, and will only be able to test this version next week. But anyway I expect it to work fine. 

The code is not perfect, but I tested many times, didn't have any issues yet. 

Works on MAX98357 with 16 bit mono WAV files. 

Following link was great reference for me during development: 
http://soundfile.sapp.org/doc/WaveFormat/

And this - https://github.com/schreibfaul1/ESP32-audioI2S - is a nice library for I2S audio on ESP32. You can check the code or use it directly.

RafigRzayev, 2021-03-10:
@JustinWeidmann I tested the code, everything works. Did it help you?