espressif/arduino-esp32 #4414: pinMode not working for pins>=40 for ESP32S2

HX2003, 2020-10-14:
It appears that digitalPinIsValid needs pin number to be <40, other functions also have this problem.

HX2003, 2020-10-14:
> Are you using the esp32s2 branch, if so please share code that is failing.

Yes, I'm using the latest esp32s2 branch.

```
void setup() {
  pinMode(41, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(41, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                       // wait for a second
  digitalWrite(41, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                       // wait for a second
}
```

chegewara, 2020-10-14:
https://github.com/espressif/arduino-esp32/blob/esp32s2/cores/esp32/esp32-hal-gpio.h#L71

felmue, 2020-10-14:
Hi guys

exactly, `esp32-hal-gpio.h` (on `esp32s2` branch) has not yet been adapted to the S2 version.

BTW: For a test I've modified that line defining `digitalPinIsValid()` and with that the LED blinks.

Thanks
Felix

me-no-dev, 2020-10-14:
fixed :)

felmue, 2020-10-14:
Hi @me-no-dev 

that was quick. Thank you - much appreciated.

Cheers
Felix

lbernstone, 2020-10-14:
Bit of a spelling error in there :scream:

vigneshtg, 2020-10-15:
Hello, 
I'm also using ESP32-S2 interfaced with MLX90614 (I2C interface). But I'm facing problem that the program is uploaded successfully and when I open Serial Monitor, it is showing nothing.  Also I've used SDA - Pin8  and SCL - Pin9. Is that right?

Further, I'm using Sparkfun MLX90614 library:
https://github.com/sparkfun/SparkFun_MLX90614_Arduino_Library

Please help me out.
Thank you.

atanisoft, 2020-10-15:
@vigneshtg that is entirely unrelated to this issue. You have also filed an issue for your issue as well.

vigneshtg, 2020-10-16:
Okay, I'll move the issue.

closed 2020-12-31 by stale[bot]