espressif/arduino-esp32 #4417: PSRAM detection issue

Sanketa54, 2020-10-15:
Hi, 
I'm interfacing external PSRAM of 64Mb with ESP32D2WD chip.
Tried following things :
1. ENABLED Psram from BOARD settings. --- "psram not supported "msg 
2. Used arduino / idf commands to check the PSRAM total/free memory size.
3. Tried by ENABLING CONFIG_SPIRAM_BOOT_INIT in sdkconfig file. ----  SHowing no memory
4. Tried by changing pin configuration in sdkconfig file for psram.

Please help me wiht the solution

lbernstone, 2020-10-15:
Please follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md).  You need to provide more information if you want help with your problem.

Sanketa54, 2020-10-16:
> Please follow the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md). You need to provide more information if you want help with your problem.

I have updated .. Please check if there's any solution

me-no-dev, 2020-10-16:
your debug messages show that PSRAM was enabled 👍 
in any case, list what PSRAM chip you are using and how you have it connected to the ESP

Sanketa54, 2020-10-16:
> your debug messages show that PSRAM was enabled 👍
> in any case, list what PSRAM chip you are using and how you have it connected to the ESP

Its showing ENABLE when I changed the sdkconfig.h i.e CONFIG_SPIRAM_BOOT_INIT is Enabled
I think its hould be enabled inspite of any changes in sdkconfig.h

Sanketa54, 2020-10-19:
> your debug messages show that PSRAM was enabled 👍
> in any case, list what PSRAM chip you are using and how you have it connected to the ESP
@me-no-dev  : PSRAM :  APS6404L-SQR-ZR  (64Mb)

closed 2021-01-02 by stale[bot]