espressif/arduino-esp32 #4419: Problems with I2C

x10102, 2020-10-15:
I have a 128x64 SSD1306 display and no matter what I tried, it didn't seem to work with the ESP32. It always displayed a few lines correctly and the rest was just noise. When looking at the SCL pin with an oscilloscope, I noticed that the frequency is 581KHz instead of the correct 400KHz.
Any idea what causes this and how can I fix it?

### Hardware:
Board:							ESP32 development board
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					80MHz
PSRAM enabled:                                     No
Upload Speed:				        256000
Computer OS:                                         Win10

felmue, 2020-10-15:
Hi @x10102 

On my board ESP32_LyraT I measure about 360 kHz I2C clock frequency and a 128x32 OLED display works just fine.

That said, I found that the frequency can be set when the display is created:
```
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, OLED_RESET, 400000U, 100000U);
```
- second last parameter is clkDuring (default 400000U)
- last parameter is clkAfter (default 100000U)

With my board changing `400000U` to `450000U` results in a clock very close to 400kHz.

 In your case you could try to lower the value of the second last parameter.

Cheers
Felix

x10102, 2020-10-16:
> Hi @x10102
>
> On my board ESP32_LyraT I measure about 360 kHz I2C clock frequency and a 128x32 OLED display works just fine.
>
> That said, I found that the frequency can be set when the display is [...]

Thank you, The frequency is almost spot on now. The results with the display are still the same though

x10102, 2020-10-16:
Nevermind. I just found out that the display was labeled as SSD1306 when it actually uses the SSH1106. Sorry for such a stupid issue :D

closed 2020-10-16 by x10102