espressif/arduino-esp32 #4421: Error when reinitializing WiFi after deinitialization with WiFi.mode( WIFI_MODE_NULL )

GiovanniCmpaner, 2020-10-17:
Using branch esp32s2 

Right now, when using `WiFi.mode( WIFI_MODE_NULL )` and then initializing it again with `WiFi.mode( WIFI_MODE_STA )` generates the message:
`[E][WiFiGeneric.cpp:1240] mode(): Could not set mode! 12289`
With error code 12289 = ESP_ERR_WIFI_NOT_INIT

Minimal sketch:
```
#include <Arduino.h>
#include <WiFi.h>

void setup()
{
    Serial.begin( 115200 );
    Serial.setDebugOutput( true );

    WiFi.mode( WIFI_MODE_STA );
    WiFi.mode( WIFI_MODE_NULL );
    WiFi.mode( WIFI_MODE_STA );
}

void loop(){ }
```
Output:
```
[   942][V][WiFiGeneric.cpp:314] _arduino_event_cb(): STA Started
[   943][D][WiFiGeneric.cpp:967] _eventCallback(): Arduino Event: 2 - STA_START
[   946][V][WiFiGeneric.cpp:319] _arduino_event_cb(): STA Stopped
[   950][D][WiFiGeneric.cpp:967] _eventCallback(): Arduino Event: 3 - STA_STOP
[   960][W][WiFiGeneric.cpp:1268] getMode(): WiFi not started
[   963][E][WiFiGeneric.cpp:1224] mode(): Could not set mode! 12289
```

This assignment:
https://github.com/espressif/arduino-esp32/blob/baaff8480ebbdcf5c8c9214283f538a3fcdba29d/libraries/WiFi/src/WiFiGeneric.cpp#L568
Should be 
`lowLevelInitDone = !(esp_wifi_deinit() == ESP_OK);`

lbernstone, 2020-10-17:
Fix https://github.com/espressif/arduino-esp32/commit/80e9e42c3b7a8ad646cd8803c9f4f84e35c9ebb1 has not been merged into the branch yet, but will be.

GiovanniCmpaner, 2020-10-18:
Thats good to know, thanks!

closed 2021-01-02 by stale[bot]

GiovanniCmpaner, 2021-02-20:
Even after fix [80e9e42](https://github.com/espressif/arduino-esp32/commit/80e9e42c3b7a8ad646cd8803c9f4f84e35c9ebb1), I'm still getting
```
[ 10769][V][WiFiGeneric.cpp:315] _arduino_event_cb(): AP Started
...
[ 10800][E][WiFiGeneric.cpp:983] mode(): Could not set mode! 12289
```
My initial proposition would be a possible solution.