espressif/arduino-esp32 #4423: Connection issue with ESP32 SoftAP with Android 10 devices - why?

ringmybell, 2020-10-18:
I have setup an ESP32 to act as a softAP serving up a webpage to connected clients. I am using websockets and ESPAsync libraries. Works great...... most of the time. 

When connecting newer Android Phones the Wifi connects to the ESP32, then drops off within a second, reconnects, then drops off in a second, reconnects.... and so on. My older Android phones connect no problem as do my windows laptops. 

**Works**
Samsung Galaxy J5 Prime, Android 6.0.1
Sony Xperia Z1 Compact D5503, Android 5.1.1
Toshiba Laptop, running windows 10
Dell G3 Laptop, , running windows 10

**Does not work**
Motorola Moto G8 XT2045-6, Android version 10
Xiaomi 9T pro M1903F11G, Android 10

So seems to be an issue with Android 10, the Arduino code seems to work fine.  Does anyone else have this issue, any ideas on why this happening? 

Happens if I upload sketch through Arduino and PlatformIO, using latest ESP32 from Github and latest libraries.. 

framework-arduinoespressif32 3.10004.201016 (1.0.4)
ArduinoJson 6.16.1
AsyncTCP 1.1.1
ESP Async WebServer 1.2.3
WebSockets 2.3.1

Bmooij, 2020-10-19:
In later Android versions you should bind your app to the WiFi connection.

Because later Android versions detect that the ESP32 SoftAP doesn't provide access to the internet, it routes the requests over the cellular connection

Bmooij, 2020-10-19:
By disabling the cellular data on your Android 10 phone and connect it to the WiFi access point of the ESP32, it should work fine.

Or if you have access to the app call: [bindProcessToNetwork](https://developer.android.com/reference/android/net/ConnectivityManager#bindProcessToNetwork(android.net.Network))

ringmybell, 2020-10-19:
Thankyou for your reply, much appreciated.

Good link, interesting information I did not know about previously. But I am not using any custom apps in my process, and I am not sure which app I would need to modify on the phone, assuming you can modify the source code? 

I am trying to get the softAP to act as server and pass webpage over websocket to client when url in webpage is entered. But I can not even get to this point as the wifi drops out. Sorry if I have misunderstood what you have said. 

Since this post I learnt about Android debug tool and Verbose Wifi. This shows the access point has been labelled NETWORK_SELECTION_PERMANENTLY_DISABLED. I have searched google but can not find out how I can override this. 

As you note it seems to be because the softAP has no internet connection and it is trying to move back to a mobile connection. However, I have the same phones connected to an internet router, I have pulled the WAN ethernet cable (removed internet connection) but do not get the same response from the phone, so maybe the problem is on the ESP32 side. 

I have tried all of the following, nothing works. On mobile phone side.....
> Disconnecting Mobile 4G
> Disconnecting Bluetooth
> Turning on flight mode
> Every combination of the above
> Restarting Wifi
> Revert to factory default for Wifi settings
> Revert to factory setting for Apps
> Downloaded latest Android 10 updates
> Used device MAC instead of randomised MAC

On ESP32 side.....
> Removed password from softAP

Seems this might be an issue with Websockets or ESP32 wifi libary? Does it need to issue some command to the connected device? 

Thanks

Bmooij, 2020-10-19:
Can you try to compile your code with the CORE_DEBUG_LEVEL to ARDUHAL_LOG_LEVEL_VERBOSE?
Maybe the Serial output of the ESP32 can provide some information.

ringmybell, 2020-10-20:
I think I did the verbose log correct

When using Android 10.....
22:13:06.747 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 16 - AP_STADISCONNECTED
22:13:06.908 -> dhcps: send_offer>>udp_sendto result 0
22:13:06.948 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 18 - AP_PROBEREQRECVED
22:13:09.687 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
22:13:11.128 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 16 - AP_STADISCONNECTED
22:13:11.327 -> dhcps: send_offer>>udp_sendto result 0
22:13:11.368 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 18 - AP_PROBEREQRECVED
222:13:19.767 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 17 - AP_STAIPASSIGNED
and then wifi drops off

When I use another Android phone
22:10:49.348 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 16 - AP_STADISCONNECTED
22:10:49.548 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 18 - AP_PROBEREQRECVED
22:10:51.668 -> [1] Connection from 192.168.1.186
and stays connected

ringmybell, 2020-10-22:
Is anyone from espressif going to help trouble shoot this? Are they active on this forum?  

The softAP does not seem to work with Android 10 devices which are now prevalent in the market, the same Android 10 devices can connect to a Raspberry Pi access point, or a WiFi router with the Modem pulled out, they dont exhibit the same behavior, so it does look like there is something not exactly right with the Arduino code with the more modern Android devices.

atanisoft, 2020-10-22:
It works fine on my Android 10 phone when using the SoftAP. 

Most likely it is due to the Android system seeking the internet and not finding the dns entries required. Since you haven't included any code we can't really help you solve your code issue.

Bmooij, 2020-10-22:
Which ESP32 are you using? And which pins are you using on the ESP32?

ringmybell, 2020-10-22:
Are you tweaking some kind of developer setting on the Android 10 phone? Or is just connects straight up? 

Even the most basic type of sketch wont work for me, for example this will not work https://shawnhymel.com/1882/how-to-create-a-web-server-with-websockets-using-an-esp32-in-arduino/

I am using ESP32 Pico Kit https://www.digikey.com/en/products/detail/espressif-systems/ESP32-PICO-KIT/9381703

I have tried two different ESP32 pico, same result. 

As noted, earlier Android phones and windows laptops are okay. I have just retried the two Android 10 phones I have an neither work on the ESP32 softAP. The exact same phones WILL connect to a Raspberry Pi acting as a softAP or router AP without ethernet connections. 

This is my platformIO INO file

```
[env:pico32]
platform = espressif32
board = pico32
framework = arduino
monitor_speed=9600
lib_deps = 
    ArduinoJson                       
    AsyncTCP                            
    ESP Async WebServer                 
    WebSockets      
```

Bmooij, 2020-10-22:
Can you try a very simple example?

```ini
[env:pico32]
platform = espressif32
board = pico32
framework = arduino
monitor_speed=9600
```

```cpp
#include <WiFi.h>

void setup() {
  Serial.begin(9600);
  
  WiFi.softAP("ESP32", "123456789");
  Serial.println();
  Serial.println("AP running");
  Serial.print("My IP address: ");
  Serial.println(WiFi.softAPIP());
}

void loop() {}
```

ringmybell, 2020-10-22:
That works.

The same phone connects and stays connected, it does not keep dropping of and trying to reconnect.

ringmybell, 2020-10-23:
Okay, I dont understand......

If I change the ssid to anything other than "ESP32" with using a password I am getting the same drop out behavior. In the WiFi debug on my mobile ti says "Check password and try again"  Huh? For example I just change ESP32 to ESP33...

```
#include <Arduino.h>
#include <WiFi.h>

void setup() {
  Serial.begin(9600);
  WiFi.softAP("ESP33", "123456789");

  Serial.println();
  Serial.println("AP running");
  Serial.print("My IP address: ");
  Serial.println(WiFi.softAPIP());
}

void loop() {}
```

However, the code works if I remove the password as follows.....

```
#include <Arduino.h>
#include <WiFi.h>

void setup() {
  Serial.begin(9600);
  WiFi.softAP("ESP33");

  Serial.println();
  Serial.println("AP running");
  Serial.print("My IP address: ");
  Serial.println(WiFi.softAPIP());
}

void loop() {}
```

That seems like very odd behavior ????

ringmybell, 2020-10-28:
Anyone have any idea on what is going on?

ringmybell, 2020-11-17:
Do any of the developers from espressif look at this forum?  As you can see above there is some strange issue happening with the ESP32 Arduino code.

The Android 10 phone works fine with Raspberry Pi acting as a softAP without internet connection, as does connecting to a wireless router with internet disconnected. So I dont think the issue is necessarily with the mobile phone and maybe an issue working with the new Android phones.

lbernstone, 2020-11-17:
Have you tested with the [development version 1.0.5](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md)?

CoenieK, 2021-10-15:
> I have setup an ESP32 to act as a softAP serving up a webpage to connected clients. I am using websockets and ESPAsync libraries. Works great...... most of the time.
>
> When connecting newer Android [...]

Hay there, I'm currently having the same issue you did. Non-Android 10 phone connects to my ESP32 AP without any issues. Android 10 phone (Redmi Note 7) almost instantly gets disconnected.

Have you found a solution?

Kind regards.

VojtechBartoska, 2021-10-15:
Hello guys, @ringmybell and @CoenieK. I'm adding this issue on the Roadmap so we will take a look and come up with resolution.

leeebo, 2021-11-15:
@ringmybell Thanks for your detailed test result, seems like the same problem in my project... I will ask some Wi-Fi guys for help

gabriel-mendes-dev, 2021-12-01:
Hello, for anyone who is having this problem yet, I was facing it too. To work around it, I configured my smartphone to ask for static IP to ESP32 soft AP (in this case, I was using default 192.168.4.0 network so I put IP 192.168.4.x, gateway 192.168.4.1, network preffix 24, DNS 0.0.0.0). So it seems like the issue is somehow related to DHCP.

leeebo, 2021-12-03:
Hi @all, in this case, please disable Wi-Fi AMPDU RX to work normally. Because some device/OS will try to setup multiple `Block Ack session` in this mode,[ this will cause problems to ESP32 SoftAP](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html?highlight=ampdu#wi-fi-qos).

You can disable it through menuconfig:
`(Top) → Component config → Wi-Fi → WiFi AMPDU RX`.

writetosumeet, 2022-01-07:
Hi - I am facing same issue. Android 11 not able to hold on to ESP32 SoftAP Wifi connection.
Among other things, tried:
1) disabling Wifi AMPDU RX as per @leeebo  
2) static IP as per @gabriel-mendes-dev.
No improvement. Phone disconnects after 2-3 minutes.
Mobile data on/off has no impact.
This is a major issue holding my product development!
Can anyone help please?
Can we befool Android in thinking that esp32 has internet access by setting a DNS server?

sansillusion, 2022-01-07:
If it's a Samsung, try setting the eap32 IP to 4.3.2.1
This is what wled does to get Samsung phones to connect to it's AP.

youngda, 2022-03-06:
Same problem, version 1.0.6 saved my life.

Billy3195257, 2022-03-21:
> You can disable it through menuconfig: `(Top) → Component config → Wi-Fi → WiFi AMPDU RX`.
You're the best!
This solves my problem with uploading a large file (OTA Update) from Android phone.

davek61, 2022-03-21:
I discovered that all of my Android 10 connection drop out problems were actually solved by changing my code to this 👍
 
void setup() {
  Serial.begin(115200);
  Serial.println("Configuring access point...");

  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid);//no password requested since no second argument in the brackets Only SSID.
  Serial.println("Wait 100 ms for AP_START...");
  delay(100);//seems like this delay is quite important.  
  
  Serial.println("Set softAPConfig");
  IPAddress Ip(192, 168, 1, 22);
  IPAddress NMask(255, 255, 255, 0);
  WiFi.softAPConfig(Ip, Ip, NMask);
  
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);

VojtechBartoska, 2022-04-07:
Hello folks, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? 

Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

VojtechBartoska, 2022-04-21:
any results from testing this? Thanks

VojtechBartoska, 2022-05-04:
Anyone gave this a try?

CoenieK, 2022-05-05:
Hi @VojtechBartoska , thanks for following up on this issue. I'm starting a new project within the next two weeks and would really like to use the ESP32 - will give an update as soon as I have some boiler plate code running.

VojtechBartoska, 2022-05-26:
@CoenieK Can you confirm this issue?

CoenieK, 2022-05-29:
> @CoenieK Can you confirm this issue?

Hi again, I'm sorry. Really want to test it properly, but other projects currently have higher priority. I made a basic connection with ESP32 as AP (using latest IDF) and the connection to my Android 12 phone seems to be stable. It is on my stack to test this thoroughly - but it will take some more time. :-/

ukrsms, 2022-06-15:
Faced the problem in my project as well.
In my case, Android 10 is keeping the connection, but the problem is being observed when using Xiaomi Note 5 Pro (Android 9  MIUI 12), however, Xiaomi Note 5 (Android 9  MIUI 11) works well without ant connection dropping.
For my case the [advice ](https://github.com/espressif/arduino-esp32/issues/4423#issuecomment-985171055) given by @leeebo solved the issue

GrandeRipper, 2022-07-01:
I also have the problem described. Everything worked fine in an older version of my program (around 2020). No idea which Android version or ESP32 Framework version was up to date at the time.
Now I wanted to start a similar project. All possible devices can be connected perfectly. Only my Xiaomi Mi 7 with Android 10 MiUI 12.0.3 causes problems. ESP Framework version is now 2.0.3. A workaround seems to have been found by disabling AMPDU RX as mentioned above. Maybe the problem also comes from the customized Xiaomi firmware...? Xiaomi smartphones have been mentioned here several times.
Here is the code snippet:
`  WiFi.mode(WIFI_AP);
  //WiFi.begin(ssid, password);
  WiFi.softAP(ssid, password, 4, 0, 8);

  delay(500);//seems like this delay is quite important or not...?
  //ANDROID 10 WORKAROUND==================================================
  //set new WiFi configurations
  WiFi.disconnect();
  Serial.println("reconfig WiFi...");
  /*Stop wifi to change config parameters*/
  esp_wifi_stop(); //stop WIFI
  esp_wifi_deinit(); //"De init"
  /*Disabling AMPDU RX is necessary for Android 10 support*/
  wifi_init_config_t my_config = WIFI_INIT_CONFIG_DEFAULT();   //We use the default config ...
  my_config.ampdu_rx_enable = 0;                               //... and modify only what we want.
  Serial.println("WiFi: Disable AMPDU...");
  esp_wifi_init(&my_config); //set the new config = "Disable AMPDU"
  esp_wifi_start(); //Restart WiFi
  delay(500);
  //ANDROID 10 WORKAROUND==================================================
  
  Serial.println("Set softAPConfig");
  IPAddress Ip(192, 168, 1, 22);
  IPAddress NMask(255, 255, 255, 0);
  WiFi.softAPConfig(Ip, Ip, NMask);`

harissutanrafiq, 2022-12-29:
> I also have the problem described. Everything worked fine in an older version of my program (around 2020). No idea which Android version or ESP32 Framework version was up to date at the time. Now I wanted to start a similar project. All possible devices can be connected perfectly. Only my Xiaomi Mi 7 with Android 10 MiUI 12.0.3 causes problems. ESP Framework version is now 2.0.3. A workaround seems to have been found by disabling AMPDU RX as mentioned above. Maybe the problem also comes from the customized Xiaomi firmware...? Xiaomi smartphones have been mentioned here several times. Here is the code snippet for Arduino IDE:
> 
> ```
>   #include <WiFi.h>
>   #include <esp_wifi.h>
>   #include <WebServer.h>
>   
>   WiFi.mode(WIFI_AP);
>   //WiFi.begin(ssid, password);
>   WiFi.softAP(ssid, password, 4, 0, 8);
> 
>   delay(500);//seems like this delay is quite important or not...?
>   //ANDROID 10 WORKAROUND==================================================
>   //set new WiFi configurations
>   WiFi.disconnect();
>   Serial.println("reconfig WiFi...");
>   /*Stop wifi to change config parameters*/
>   esp_wifi_stop(); //stop WIFI
>   esp_wifi_deinit(); //"De init"
>   /*Disabling AMPDU RX is necessary for Android 10 support*/
>   wifi_init_config_t my_config = WIFI_INIT_CONFIG_DEFAULT();   //We use the default config ...
>   my_config.ampdu_rx_enable = 0;                               //... and modify only what we want.
>   Serial.println("WiFi: Disable AMPDU...");
>   esp_wifi_init(&my_config); //set the new config = "Disable AMPDU"
>   esp_wifi_start(); //Restart WiFi
>   delay(500);
>   //ANDROID 10 WORKAROUND==================================================
>   
>   Serial.println("Set softAPConfig");
>   IPAddress Ip(192, 168, 1, 22);
>   IPAddress NMask(255, 255, 255, 0);
>   WiFi.softAPConfig(Ip, Ip, NMask);
> ```

SOLVED

when i set password using hardcode not using variable PASS_WIFI its working fine on all device

```
char SSID_WIFI[30];
  char PASS_WIFI[30];

    // Connect to Wi-Fi network with SSID and password
  #ifdef DEBUG_MODE
  PORT.print("Setting AP (Access Point)");
  #endif
  // Remove the password parameter, if you want the AP (Access Point) to be open
  String S_SSID = readFile("/config/config_SSID.txt");
  String S_PASS = readFile("/config/config_PASS.txt");

  S_SSID.toCharArray(SSID_WIFI, 30);
  S_PASS.toCharArray(PASS_WIFI, 30);

  if (strlen(SSID_WIFI) < 1) {
    sprintf(SSID_WIFI , "%s %s", "SS-JWS-WIFI-",__DATE__);
  }

  if ( strlen(PASS_WIFI) < 1) {
    sprintf(PASS_WIFI, "%s", S_PASS);
  }

  WiFi.mode(WIFI_AP);
  WiFi.softAP("MY-SSID", "12345678",4,0,8); // <----- hardcode password
  WiFi.disconnect(true);

  esp_wifi_stop(); //stop WIFI
  esp_wifi_deinit(); //"De init"
  /*Disabling AMPDU RX is necessary for Android 10 support*/
  wifi_init_config_t my_config = WIFI_INIT_CONFIG_DEFAULT();   //We use the default config ...
  my_config.ampdu_rx_enable = 0;                               //... and modify only what we want.
  esp_wifi_init(&my_config); //set the new config = "Disable AMPDU"
  esp_wifi_start(); //Restart WiFi
  
  IPAddress local_IP(192, 168, 4, 1);
  IPAddress gateway(192, 168, 1, 1);
  IPAddress subnet(255, 255, 0, 0);
  
   WiFi.softAPConfig(local_IP, gateway, subnet);
  
  PORT.print("PASS_WIFI: ");
  PORT.println(PASS_WIFI);
  wifiPassword = PASS_WIFI;

  IPAddress IP = WiFi.softAPIP();
  PORT.print("AP IP address: ");
  PORT.println(IP);
```

piyush9999, 2023-03-02:
In android I turn on my Developer options and turn on the USB debugging, mobile data active, wifi throttling and it work all great for me.

Endorfin35, 2023-10-06:
I had the same problem.

My previous code was like this:
WiFi.softAPConfig(AP_IP,  AP_GATE,  AP_MASK);

I saw in the debug output that the dhcp starting address was 0.0.0.0

The addition I made:
WiFi.softAPConfig(AP_IP,  AP_GATE,  AP_MASK,  AP_DHCP);

When I added the dhcp parameter as an IP block in accordance with the IP address, the problem was solved.

bockesj2, 2024-02-11:
I had a problem with my Android tablet dropping the connection after about a minute of being connected to the ESP32 (configured as an AP) and I'd like to pass on what I found out.
I connected my laptop to the ESP32 and no problem - it would stay connected.  So I knew it was an Android issue.
There are no settings on my Android tablet to say "Stay connected no matter what" but on my device, a brief message comes up saying that no internet is available on that device, I found out that if I quickly click that brief message it gives me an option to "stay connected".  So far, it seems like this was the key, the tablet stays connected.
Once again it seems that a device is trying to be too smart by dropping the Esp32 connection for a connection with an internet connection. Note that this tablet does not have a "data" connection so it is not dropping it, in this case, for the "data" connection, as was mentioned in other posts.

faizannazir, 2024-03-30:
I have same kind off issue i am using wifi manager first Android 10 and 11 connect without any issue when i update the wifi creditionals and then reset to remove wifi and if again i try to connect it not connecting.

2nd time it work if reset esp32 and within second  click on connect on mobile

BijeshSomarajan, 2024-06-17:
I too had the same issue that  the android device keeps reconnecting while the ESP_32 AP/Wifi is selected.
It seems that the devise is checking for internet access. 
This can be solved it by - Disable DHCP for the AP/Wifi and set the DNS IP as 0.0.0.0
 so that Internet access is not constantly checked.

rftafas, 2025-01-06:
From the multiple reports here, it seems to be a smartphone issue (it keeps looking after internet) or something that ended solved along the way. This thread also has several suggestions to be tested. 

Because it is using a now outdated version of Arduino for ESP32 (2.x.x), if the issue remains, before re-opening it, please, update the test set-up.

closed 2025-01-06 by rftafas