espressif/arduino-esp32 #4427: esp32s2 branch in platformio

ccssw, 2020-10-19:
## Hardware:

Board: ESP2-WROVER-B
Core Installation version: Unknown
IDE name: Platform.io
Flash Frequency: 40Mhz
PSRAM enabled: yes
Upload Speed: 115200
Computer OS: Ubuntu

## Description:

I have an ESP2-WROVER-B and I want to use the PSRAM. How can I use the branch esp32s2 in platformio? Can you give me any ```platformio.ini``` example for an ESP2-WROVER-B?

atanisoft, 2020-10-19:
Add this to your platformio.ini:
```
[env]
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#esp32s2
```

ccssw, 2020-10-19:
Thank you for your answer but it doesn't work for me. My platformio.ini is:

```
[env:esp32s2]
board_build.partitions = /$PROJECT_DIR/custompart.csv
platform = espressif32
board = esp-wrover-kit
framework = arduino
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#esp32s2

;upload_port = /dev/cu.SLAB_USBtoUART
upload_port = /dev/ttyUSB0
upload_speed = 921600

;monitor_port = /dev/cu.SLAB_USBtoUART
monitor_port = /dev/ttyUSB0
monitor_speed = 115200

build_flags =
;-Wl,-Map,output.map ;generate memory map in output.map
;Debug level: 0 None, 1 Error, 2 Warn, 3 Info, 4 Debug, 5 Verbose
    -DCORE_DEBUG_LEVEL=3
    -DMQTT_MAX_PACKET_SIZE=512 	
    -DBOARD_HAS_PSRAM 
    -mfix-esp32-psram-cache-issue
	-DSPIRAM_CACHE_WORKAROUND=1
	-DSPIRAM_SUPPORT=1
	-DSPIRAM_BOOT_INIT=1
	-DSPIRAM_TYPE_AUTO=1
	-DSPIRAM_SIZE=-1
	-DSPIRAM_SPEED_40M=1
	-DSPIRAM_MEMTEST=1
	-DSPIRAM_MALLOC_RESERVE_INTERNAL=32768
	-DSPIRAM_MALLOC_ALWAYSINTERNAL=16384
	-DSPIRAM_USE=1
	-DSPIRAM_USE_MEMMAP=1
    ; -DSPIRAM_USE_CAPS_ALLOC=1
    ; -DSPIRAM_USE_MALLOC=1
	-DSUPPORT_STATIC_ALLOCATION=1
	-DPICO_PSRAM_CS_IO=10
	; -DSPIRAM_OCCUPY_VSPI_HOST
```

And I got:
```
Processing esp32s2 (platform: espressif32; board: esp-wrover-kit; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
KeyError: 'arduino-esp32':
  File "/home/x/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 159:
    env.LoadPioPlatform()
  File "/home/x/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/home/x/.platformio/penv/lib/python3.8/site-packages/platformio/builder/tools/pioplatform.py", line 75:
    type_ = p.get_package_type(pkg.metadata.name)
  File "/home/x/.platformio/penv/lib/python3.8/site-packages/platformio/platform/base.py", line 179:
    return self.packages[name].get("type")
=========================================================================== [FAILED] Took 0.37 seconds ===========================================================================
The terminal process "platformio 'run'" terminated with exit code: 1.
```

atanisoft, 2020-10-19:
add the snippet exact as I posted it and not in your `[env:esp32s2]` tag. It should work, if not I'd suggest post over on the PIO community forum as this is the same as how they handle arduino and esp-idf in a single project (except they stick to the v4.0 branch of arduino-esp32)

ccssw, 2020-10-23:
Changing
`
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#esp32s2
`
for
`
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#baaff8480ebbdcf5c8c9214283f538a3fcdba29d
`

I got this error during compilation:
`
xtensa-esp32-elf-g++: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'
`

lbernstone, 2020-10-23:
You must uninstall any older versions of arduino-esp32.  It is using the wrong version of the gcc toolchain.

ccssw, 2020-10-26:
Thank you for your answer @lbernstone I uninstalled everything and still got the same error.

closed 2021-01-09 by stale[bot]