espressif/arduino-esp32 #4428: mDNS memory leak

Bmooij, 2020-10-19:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Platform.io
Computer OS:                                        Windows 10

### Description:
When ending and starting the mDNS service, there is a memory leak overtime.

### Platformio.ini
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
build_flags =
    -DCORE_DEBUG_LEVEL=5
```

### Sketch: 
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <ESPmDNS.h>

const char* ssid = "..........";
const char* password = "..........";

void startMDNS() {
  if (!MDNS.begin("esp32")) {
      Serial.println("Error setting up MDNS responder!");
      while(1) {
          delay(1000);
      }
  }
  MDNS.setInstanceName("SomeRandomInstanceName");

  MDNS.addService("http", "tcp", 80);
  MDNS.addServiceTxt("http", "_tcp", "txt1", "value1");
  MDNS.addServiceTxt("http", "_tcp", "txt2", "value2");
  MDNS.addServiceTxt("http", "_tcp", "txt3", "value3");
  MDNS.addServiceTxt("http", "_tcp", "txt4", "value4");
}

void stopMDNS() {
  MDNS.end();
}

void setup(void)
{  
    Serial.begin(115200);

    // Connect to WiFi network
    WiFi.begin(ssid, password);
    Serial.println("");

    // Wait for connection
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());

    startMDNS();
[... 621 lines cut ...]
[D][main.cpp:78] loop(): Heap: 221004 => 227148 => 221212
[D][main.cpp:78] loop(): Heap: 221000 => 227148 => 221212
[D][main.cpp:78] loop(): Heap: 221004 => 226812 => 220804
[D][main.cpp:78] loop(): Heap: 221004 => 226952 => 221212
[D][main.cpp:78] loop(): Heap: 221004 => 226812 => 220680
[D][main.cpp:78] loop(): Heap: 221000 => 227148 => 221016
[D][main.cpp:78] loop(): Heap: 207956 => 226812 => 220684
[D][main.cpp:78] loop(): Heap: 221000 => 227148 => 221212
[D][main.cpp:78] loop(): Heap: 221004 => 227148 => 221212
[D][main.cpp:78] loop(): Heap: 221012 => 227148 => 221020
[D][main.cpp:78] loop(): Heap: 221016 => 227148 => 221208
[D][main.cpp:78] loop(): Heap: 221000 => 226812 => 221216
[D][main.cpp:78] loop(): Heap: 221000 => 227148 => 219464
[D][main.cpp:78] loop(): Heap: 221008 => 226952 => 221252
[D][main.cpp:78] loop(): Heap: 221004 => 227148 => 221212
[D][main.cpp:78] loop(): Heap: 221000 => 226812 => 220696
[D][main.cpp:78] loop(): Heap: 221016 => 227148 => 220960
[D][main.cpp:78] loop(): Heap: 221004 => 226812 => 220780
[D][main.cpp:78] loop(): Heap: 221016 => 226812 => 220676
[D][main.cpp:78] loop(): Heap: 220996 => 227148 => 221248
[D][main.cpp:78] loop(): Heap: 221000 => 227148 => 221024
[D][main.cpp:78] loop(): Heap: 221012 => 227148 => 221208
[D][main.cpp:78] loop(): Heap: 220996 => 227148 => 221020
[D][main.cpp:78] loop(): Heap: 221008 => 226812 => 218060
```

Bmooij, 2020-10-19:
Also tested with the latest master branch it still leaks memory
```ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
build_flags =
    -DCORE_DEBUG_LEVEL=5
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

me-no-dev, 2020-10-19:
and you stop and start the service why?

Bmooij, 2020-10-19:
Overtime the mDNS service stops on the esp32. With end and start the mDNS service is restarted

EricMc1289, 2020-12-02:
I recognized the same behaviour. I created a sample code to speed things a little bit up, but the really interesting part is, that the Heap reduces not continusly. As seen in the logs every couple seconds / minutes it doubles the used heap which is really strange. And after more than 70k used heap when it would jump to 140k the esp crashes. 
 
The question about why to start and stop mdns is easlily answered. I need to start mdns with a random hostname and then can take a look up if the preferred hostname is in use and if its not i restart the mdns service with the correct hostname to avoid collisions and have a failover system. If one esp32 crashes or is offline another one handles the incoming requests.

`
#include <WiFi.h>
#include <ESPmDNS.h>

int32_t totalHeap = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.begin("***","***");

    Serial.println("");

    // Wait for connection
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.print("Connected");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
    totalHeap = ESP.getFreeHeap();
}
int32_t lastHeap = 0;
void loop() {
   lastHeap = ESP.getFreeHeap();
   if (MDNS.begin("esp32")) {
        int32_t heapnow = ESP.getFreeHeap();
        Serial.println(String(lastHeap-totalHeap)+" | "+String(heapnow-lastHeap)+" | "+String(heapnow));
        delay(100);
        MDNS.end();
    } else {
      Serial.println("Error setting up MDNS responder!");
      delay(100);
    }
    
}
`

`
22:28:09.451 -> .ConnectedIP address: 10.0.1.73
22:28:09.451 -> 0 | -5412 | 274968
22:28:09.556 -> -200 | -5448 | 274732
22:28:09.656 -> -236 | -5288 | 274856
22:28:09.766 -> -308 | -5412 | 274660
22:28:09.874 -> -508 | -5360 | 274512
22:28:10.019 -> -656 | -5008 | 274716
22:28:10.123 -> -656 | -5008 | 274716
22:28:10.226 -> -796 | -5272 | 274312
22:28:10.332 -> -456 | -5696 | 274228
22:28:10.435 -> -744 | -5208 | 274428
22:28:10.544 -> -744 | -5212 | 274424
22:28:10.684 -> -944 | -5008 | 274428
22:28:10.792 -> -744 | -5408 | 274228
22:28:10.894 -> -1080 | -5408 | 273892
22:28:11.035 -> -944 | -5008 | 274428
22:28:11.137 -> -944 | -5012 | 274424
22:28:11.242 -> -944 | -5592 | 273844
22:28:11.345 -> -1520 | -5208 | 273652
22:28:11.456 -> -1520 | -5016 | 273844
22:28:11.599 -> -1520 | -5012 | 273848
22:28:11.701 -> -1520 | -5008 | 273852
22:28:11.811 -> -1660 | -5408 | 273312
22:28:11.917 -> -1656 | -5072 | 273652
22:28:12.018 -> -1520 | -5008 | 273852
22:28:12.152 -> -1660 | -5404 | 273316
[... 1408 lines cut ...]
22:30:50.352 -> -73892 | -5208 | 201280
22:30:50.460 -> -73892 | -5208 | 201280
22:30:50.566 -> -73892 | -5408 | 201080
22:30:50.709 -> -74228 | -4876 | 201276
22:30:50.814 -> -73892 | -5208 | 201280
22:30:50.914 -> -74092 | -5008 | 201280
22:30:51.024 -> -73892 | -5208 | 201280
22:30:51.133 -> -74092 | -5008 | 201280
22:30:51.245 -> -73892 | -5208 | 201280
22:30:51.348 -> -74092 | -5408 | 200880
22:30:51.490 -> -74092 | -5008 | 201280
22:30:51.596 -> -73892 | -5208 | 201280
22:30:51.700 -> -73892 | -5208 | 201280
22:30:51.811 -> -73892 | -5208 | 201280
22:30:51.916 -> -74092 | -5408 | 200880
22:30:52.018 -> -73892 | -5208 | 201280
22:30:52.162 -> -73892 | -5208 | 201280
22:30:52.266 -> -73892 | -5212 | 201276
22:30:52.366 -> -74092 | -5208 | 201080
22:30:52.472 -> -74232 | -5404 | 200744
22:30:52.581 -> -74228 | -5408 | 200744
22:30:52.718 -> -74228 | -5404 | 200748
22:30:52.822 -> -73892 | -5208 | 201280
22:30:52.933 -> -74092 | -5008 | 201280
`

me-no-dev, 2020-12-03:
MDNS has collision detection and avoidance :) it will change the name itself.

closed 2021-02-16 by stale[bot]

HamzaHajeir, 2023-09-16:
Any update so far regarding this issue?
I face strange memory leak and I highly suspect MDNS being the cause.