espressif/arduino-esp32 #4430: Is there an estimate to update ESP32 Arduino 1.0.4 (IDF 3.2) Release?

jaggil, 2020-10-20:
Hello,
Is there an estimate to update ESP32 Arduino 1.0.4 (IDF 3.2) Release to include 167 commits to master since this release?

lorol, 2020-10-20:
@jaggil A someone said ... February 30th :)
In general, never ask for estimates on community projects as your boss asks you.
It doesn't work this way.

bfaliszek, 2020-10-22:
IDF 4.2 is currently in beta1. Instead of releasing the arduino-esp32 update based on IDF 3.2, I would prefer to wait for the addition of support for the final IDF-4.2 version.

jaggil, 2020-10-26:
> @jaggil As someone said on similar question ... February 30th :)
> In general, never ask for estimates on community projects as your boss asks you.
> It doesn't work this way.

Hi @lorol,
In the first place, I feel that my query seems offensive to you, that is not my intention at all. Sorry.

And on the other hand, to take it into account in the future, I understand that the maintenance of this project is done by the espressif company, not by the community, right or not?
Best regards, and forgive again

lorol, 2020-10-27:
@jaggil you are right, this one is almost commercial but It is nice to pretend it is community in the Arduino spirit.
ESP care more about the money and selling chips :)  than releases here.
And BTW, what exactly will do for you a new release? Wil sound more serious and authentic, you think is more stable, or it will need fewer clicks to be downloaded from IDE? Just grab the head of the repo. That's it ...

me-no-dev, 2020-10-27:
there are many changes necessary in order to update. I would love to hit the release button, but the result needs to be good.
1.0.5 is coming and 2.0.0 close after that. I am almost done with the fixes for 1.0.5

jaggil, 2020-10-27:
Hi @lorol,
I think that if Espressif has opened this project, it is because they have many additional customers in the Arduino IDE.
We may not use a large amount of devices such as Sonoff, for example, but in business you do not have to underestimate any customer, and even less if they help you to debug your product

Regarding your second point, I think it is better that I do not answer you in order to keep the Arduino spirit much more friendly.

Thank you @me-no-dev  for your kind reply. Sorry if this is not the right place to ask this question, but I couldn't find a better place to inform me of it.
Greetings

lorol, 2020-10-27:
@jaggil 
If you are satisfied of the answer, can you please close the issue?

closed 2020-11-02 by jaggil