espressif/arduino-esp32 #4431: "Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled."

ishakaky, 2020-10-21:
Hey Ppl,

with below "simple" code I get a strange error on my ESP32-DevKitC-V4, Arduino-Version 1.8.13:
I cannot find anything in my code to resolve this. Can someone help please???

**Error:**
> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
> 12:17:39.460 -> Core 1 register dump:
> 12:17:39.460 -> PC      : 0x00000000  PS      : 0x00060430  A0      : 0x800d326e  A1      : 0x3ffb1ea0  
> 12:17:39.507 -> A2      : 0x3ffb1f10  A3      : 0x00000000  A4      : 0x3ffccf98  A5      : 0x0000000b  
> 12:17:39.507 -> A6      : 0x3ffccf44  A7      : 0x0000000b  A8      : 0x8018083e  A9      : 0x3ffb1e80  
> 12:17:39.507 -> A10     : 0x3ffba8d8  A11     : 0x00000000  A12     : 0x0000000b  A13     : 0x3ffccfa3  
> 12:17:39.507 -> A14     : 0x0000000b  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
> 12:17:39.507 -> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
> 12:17:39.507 -> 
> 12:17:39.507 -> Backtrace: 0x00000000:0x3ffb1ea0 0x400d326b:0x3ffb1ec0 0x400d32e5:0x3ffb1ee0 0x400d1aee:0x3ffb1f00 0x400d65a5:0x3ffb1fb0 0x4008b019:0x3ffb1fd0

**Exception Decoder:**

```
PC: 0x00000000
EXCVADDR: 0x00000000

Decoding stack results
0x400d326b: HTTPClient::disconnect(bool) at C:\Users\Ishak\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 359
0x400d32e5: HTTPClient::end() at C:\Users\Ishak\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\HTTPClient\src\HTTPClient.cpp line 347
0x400d1aee: loop() at C:\Users\Ishak\Documents\Arduino\MyWifi/MyWifi.ino line 38
0x400d65a5: loopTask(void*) at C:\Users\Ishak\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\main.cpp line 19
0x4008b019: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

**Code:**
```
#include <WiFi.h>
#include <AsyncTCP.h>
#include "ESPAsyncWebServer.h"
#include "HTTPClient.h"
#include <iostream>
#include <string>

const char* ssid = "YOURNETWORK";
const char* password = "33333333";
long lastSendTime = 0;
int interval = 100;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("Connecting");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to WiFi network with IP Address: ");
  Serial.println(WiFi.localIP());

}

void loop() {
  if ((millis() - lastSendTime) > interval) {
    Serial.println(".");
    if ((WiFi.status() == WL_CONNECTED)) {
      // GET Request example
      HTTPClient http;
      http.begin("http://192.168.4.1/hello");
      int httpcode = http.GET();
      if (httpcode > 0) {
        String payload = http.getString();
      }
      http.end();
      lastSendTime = millis();
    }
  }
}
```

Bmooij, 2020-10-21:
What is the result of `http.begin("http://192.168.4.1/hello");` ?

ishakaky, 2020-10-21:
The response is "Hello World". I can see the response by putting:
```
      if (httpcode > 0) {
        String payload = http.getString();
        Serial.println(payload);
      }
```
Serial Monitor:

> 13:03:32.072 -> Connected to WiFi network with IP Address: 192.168.4.2
> 13:03:32.072 -> .
> 13:03:32.119 -> Hello World
> 13:03:32.212 -> .
> 13:03:34.810 -> Hello World
> 13:03:34.938 -> .
> 13:03:34.938 -> Hello World
> 13:03:35.073 -> .
> 13:03:35.073 -> Hello World
> 13:03:35.166 -> .
> 13:03:35.214 -> Hello World
> 13:03:35.295 -> .
> 13:03:35.333 -> Hello World
> 13:03:35.428 -> .
> 13:03:35.428 -> Hello World
> 13:03:35.559 -> .
> 13:03:35.559 -> Hello World
> 13:03:35.559 -> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
> 13:03:35.559 -> Core 1 register dump:
> 13:03:35.559 -> PC      : 0x00000000  PS      : 0x00060430  A0      : 0x800d30f2  A1      : 0x3ffb1ea0  
> 13:03:35.608 -> A2      : 0x3ffb1f10  A3      : 0x00000000  A4      : 0x0000000b  A5      : 0x0000000b  
> 13:03:35.608 -> A6      : 0x3ffb3178  A7      : 0x0000000b  A8      : 0x80180706  A9      : 0x3ffb1e80  
> 13:03:35.608 -> A10     : 0x3ffba848  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x0000ff00  
> 13:03:35.608 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
> 13:03:35.608 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
> 13:03:35.608 -> 
> 13:03:35.608 -> Backtrace: 0x00000000:0x3ffb1ea0 0x400d30ef:0x3ffb1ec0 0x400d3169:0x3ffb1ee0 0x400d1aba:0x3ffb1f00 0x400d6305:0x3ffb1fb0 0x4008b019:0x3ffb1fd0
> 13:03:35.647 -> 
> 13:03:35.647 -> Rebooting...
> 13:03:35.647 -> ets Jun  8 2016 00:22:57

ishakaky, 2020-10-21:
I changed the code to:
```
void loop() {
  if ((millis() - lastSendTime) > interval) {
    if ((WiFi.status() == WL_CONNECTED)) {
      // GET Request example
      HTTPClient http;
      int begin = http.begin("http://192.168.4.1/hello");
      Serial.println(begin);
      int httpcode = http.GET();
      if (httpcode > 0) {
        String payload = http.getString();
        Serial.println(payload);
      }
      http.end();
      lastSendTime = millis();
    }
  }
}
```

> 13:06:20.856 -> 1
> 13:06:20.856 -> Hello World
> 13:06:20.973 -> 1
> 13:06:20.973 -> Hello World
> 13:06:21.067 -> 1
> 13:06:21.114 -> Hello World
> 13:06:21.196 -> 1
> 13:06:21.234 -> Hello World
> 13:06:21.328 -> 1
> 13:06:21.376 -> Hello World
> 13:06:21.376 -> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
> 13:06:21.376 -> Core 1 register dump:
> 13:06:21.376 -> PC      : 0x00000000  PS      : 0x00060430  A0      : 0x800d30f6  A1      : 0x3ffb1ea0  
> 13:06:21.376 -> A2      : 0x3ffb1f10  A3      : 0x00000000  A4      : 0x0000000b  A5      : 0x0000000b  
> 13:06:21.376 -> A6      : 0x3ffcd910  A7      : 0x0000000b  A8      : 0x80180772  A9      : 0x3ffb1e80  
> 13:06:21.376 -> A10     : 0x3ffba8fc  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x0000ff00  
> 13:06:21.376 -> A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000014  
> 13:06:21.423 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
> 13:06:21.423 -> 
> 13:06:21.423 -> Backtrace: 0x00000000:0x3ffb1ea0 0x400d30f3:0x3ffb1ec0 0x400d316d:0x3ffb1ee0 0x400d1abb:0x3ffb1f00 0x400d6371:0x3ffb1fb0 0x4008b019:0x3ffb1fd0
> 13:06:21.423 -> 
> 13:06:21.423 -> Rebooting...
> 13:06:21.423 -> ets Jun  8 2016 00:22:57
> 

**http.begin() returns 1
http.getString() returns "Hello World"**

MarkyAD, 2020-10-21:
I'd at least add another clause in checking httpcode.
```c
    if (httpcode > 0) {
      if (httpcode == HTTP_CODE_OK) {
        Serial.println("HTTP 200 OK");
        String payload = http.getString();
        Serial.println(payload);
      }
    }
```
It may also help to print a readable value of httpcode with http.errorToString(httpcode).

ishakaky, 2020-10-21:
> I'd at least add another clause in checking httpcode.
>
> ```c
>     if (httpcode > 0) {
>       if (httpcode == HTTP_CODE_OK) {
>         Serial.println("HTTP 200 OK");
>         String payload = [...]

I have checked this, same reaction :(

MarkyAD, 2020-10-21:
You have int interval = 100; thats 100ms between retries, maybe slow that down a bit?

ishakaky, 2020-10-22:
Hi MarkyAD,

I really need that interval for my application. I just started working with websockets, because I cannot solve this problem :( Somewhere I have read something about installing Version 1.0.1 to solve this..

MarkyAD, 2020-10-22:
You building a proxy scanner or netmapper? You'll need something that can do non-blocking i/o like an asyncronous web client.

markusTz, 2020-10-22:
This is my setting: Controlling some little robots with sensors over a master..

- Using **Wifi**
- Master is No.1.
- The Slaves are No.2 to No.6
- Communication is bidirectional from Master to Slaves.
- Both parts send **cyclic AND event messages**. I need both types!

[image]

So my first idea was using Http GET and POST, with described error message above :(
My current idea is using Websockets..

MarkyAD, 2020-10-22:
You're right, why bother with the overhead of http when plain tcp or udp will work just fine. All you need is a custom data structure to send back and forth. I've done something similar with those RF24 radio modules.

closed 2021-01-06 by stale[bot]