espressif/arduino-esp32 #4432: How to safely delete RTOS task

RafigRzayev, 2020-10-21:
Let's say we have a task created with xTaskCreate() which is doing some work. At some time later, our main loop receives a signal to kill this task. We have to use vTaskDelete() in order to remove the task. If our task has allocated memory, memory won't be freed automatically. Plus, as I have understood from the following link, even destructors won't  be called on task termination: https://freertos.org/FreeRTOS_Support_Forum_Archive/December_2017/freertos_C_RAII_in_FreeRTOS_f0d9a348j.html

What comes to my mind as a solution to the possible memory leak, is to store all pointers to be used inside the task outside from the task's scope. Once main loop receives a signal to kill the task, we firstly free these pointers, and afterwards call vTaskDelete(). Following code demonstrates this approach:
```cpp

uint8_t* my_task_memory = NULL;
TaskHandle_t my_task_handle = NULL;

void my_task(void* params) {
  for (;;) {
    my_task_memory = (uint8_t*)malloc(100);
    Serial.println("My Task doing something");
    delay(500);
    free(my_task_memory);
  }
}

void clear_my_task_memory() {
  if (my_task_memory != NULL) {
    free(my_task_memory);
  }
}

void setup() {
  Serial.begin(115200);
}

void loop() {
  // Create a task
  Serial.println("---------------------");
  Serial.println("My task started");
  xTaskCreate(my_task, "MyTask", 1000, NULL, 1, &my_task_handle);
  
  // Let task to run for some time
  Serial.println("Wait 3 seconds");
  delay(3000);
  
  // At this point we decide to delete the task;
  Serial.println("Killing the task");
  vTaskSuspend(my_task_handle);
  clear_my_task_memory();
  vTaskDelete(my_task_handle);
  
  // Check if heap size has changed
  Serial.println(String("HEAP SIZE: ") + ESP.getFreeHeap());
}
```

As visible from the logs, there are no memory leaks:
[image]

Now, if we modify our task code by removing the delay between the calls to malloc() and free():
```cpp
void my_task(void* params) {
  for (;;) {
    my_task_memory = (uint8_t*)malloc(100);
    //Serial.println("My Task doing something");
    //delay(500);
    free(my_task_memory);
  }
}
```
We will get following error:

[image]

If I am not mistaken, the reason is following: Initially, when we had a delay, vTaskDelete() was called during the delay since most of the time used by the task was in the delay state. After removing the delay, task time is mostly spent only on processing free/malloc functions. And the task is terminated while those functions are in the midst of their operation which leads to corrupted state. 

Which mechanism can be used in order to avoid such situation? How can I make malloc() and free() functions in this example to behave atomic?

In my project, I am using a lot of external libraries/functions inside the task (such as SD_MMC, WiFi and etc.). I need a way to make these function calls atomic, otherwise my software is constantly crashing during task kills. This problem lead to memory leaks, invalid states and etc. 

I would appreciate your assistance.

mkfrey, 2020-10-21:
You could use one of the available mechanisms like [notifies](https://www.freertos.org/xTaskNotify.html) to notify the task that it should delete itself. This gives you the ability to free the resources inside the task before making it delete itself.

RafigRzayev, 2020-10-26:
Thank you very much @mkfrey 

I am looking into it. I have also discovered following approaches that are relevant:

1) Wrap code of interest and guarantee that task switching won't happen until this code has not finished execution via:
- taskENTER_CRITICAL() / taskEXIT_CRITICAL()
- vTaskSuspendAll() / xTaskResumeAll()

2) Assign a delete function to a task. Once task will be deleted, the function will be called automatically
- vTaskSetThreadLocalStoragePointerAndDelCallback()

I had many issues during experimenting with these functions. I am planning to use notifications this week as well. 

I will hopefully share my experience with you within 1-2 weeks.

RafigRzayev, 2020-11-06:
Answer given in FreeRTOS forum. refer to https://forums.freertos.org/t/advice-on-designing-task-deletion/11065/8

closed 2020-11-06 by RafigRzayev