espressif/arduino-esp32 #4434: Filesystem functions do not work

mcmega, 2020-10-21:
Hello!
The functions for working with the file system stopped working.
In this example, I want to create an array of filenames in the /dl directory.
```
void createLangList() {
  DynamicJsonDocument doc(512);
  JsonArray langList = doc.to<JsonArray>();
    Dir dir = SPIFFS.openDir("/dl");
    while (dir.next()) {
      File entry = dir.openFile("r");
      const char *path = entry.name();
      char langName[3] = "";
      int pos = 0, slash_cnt = 0;
      for (int in_pos = 0; in_pos < strlen(path); in_pos++) {
        if (path[in_pos] == '/')
          slash_cnt++;
        else if (slash_cnt == 2) {
          langName[pos] = path[in_pos];
          pos++;
        }
        if (pos > 1)
          break;
      }
      langName[pos] = '\0';
      langList.add(langName);
      entry.close();
    }
  serializeJson(langList, configDev.dls);
}
```
The compiler throws an error.
`error: 'Dir' was not declared in this scope`

Code works for ESP8266, but don't work for ESP32

lorol, 2020-10-21:
@mcmega It is different between esp8266 and esp32
Get oriented here for example:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/SPIFFSEditor.cpp#L442

mcmega, 2020-10-22:
Thanks!
Everything worked out.
It would be nice to have documentation for ESP32 for Arduino.

closed 2020-10-22 by mcmega

lorol, 2020-10-22:
There are some guides meanwhile ... as as search at Google shows:
https://diyprojects.io/esp32-get-started-spiff-library-read-write-modify-files/