espressif/arduino-esp32 #4435: WiFiClient improperly treats zero data available for read as an error

MHotchin, 2020-10-22:
In "arduino-esp32/libraries/WiFi/src/WiFiClient.cpp".

If the client has run out of data that has been received, it will return an error, instead of returning 0 (indicating no data available right now).

Starting line 106:

```
    int read(uint8_t * dst, size_t len){
        if(!dst || !len || (_pos == _fill && !fillBuffer())){
            return -1;

```
Change to:

```
    int read(uint8_t * dst, size_t len){
        if(!dst || !len || (_pos == _fill && !fillBuffer())){
            return _failed ? -1: 0;
```

MHotchin, 2020-10-22:
This appears to be the cause of #4390 as well.

MHotchin, 2020-10-23:
Add to fix - function `'read()` needs to handle the zero data available case as well.  Found by using `readBytes()`, which calls the single byte version of `read()`.
```
int WiFiClient::read()
{
    uint8_t data = 0;
    int res = read(&data, 1);
    if(res < 0) {
        return res;
    }
    if (res == 0) {
        return -1;
    }
    return data;
}

```

MarkusAD, 2020-10-26:
@MHotchin you should submit your changes as a pull request so that they can be reviewed and merged into the current code base.

MHotchin, 2020-10-27:
PR  #4448 submitted.

commit 7e40de22 (2020-10-27): Fixes #4435 - WiFiClient improperly treats zero data available for read as an error (#4448)
closed 2020-10-27 by me-no-dev